blackmagic-debug/blackmagic #769: When Hosted disconnects from target, the target is left halted

sidprice, 2020-11-02:
Windows 10 x64
Target STM32F415RG using BMP native (firmware at master-head)
Hosted built from master-head
Execute the following commands in GDB to reproduce the problem of the target being left halted:
(gdb) target extended-remote :2000
Remote debugging using :2000
(gdb) mon s
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      STM32F40x M4
(gdb) att 1
Attaching to program: F:\DataRoot\ctxLink_tests\ctxLink_Blicky_STM32415RG\build\ctxLink_Blicky_STM32415RG.elf, Remote target
HAL_GetTick () at Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal.c:325
325       return uwTick;
(gdb) disc
Ending remote debugging.

The same commands when using gdb directly with the BMP native probe leave the target running afer disconnect.

UweBonnes, 2020-11-02:
(gdb) help disconnect
Disconnect from a target.
The target will wait for another debugger to connect.  Not available for
all targets.

After attach both serial and hosted for me keep the device halted, a.k. waiting for another debugger to connect. Use detach or quit.

sidprice, 2020-11-02:
OK ... however, with serial I see the target start running after "disc". Seems inconsistent somewhere.

UweBonnes, 2020-11-02:
Okay, I see inconsistancies now too.

UweBonnes, 2020-11-02:
With "disconnect" nothing is sent via gdb. So what happens afterwards is for me "implementation defined". If you want to have the device in a known state, use detach or quit and not disconnect.

closed 2020-11-02 by UweBonnes

sidprice, 2020-11-02:
@UweBonnes Thanks for checking that out and clarifying this, much appreciated.