blackmagic-debug/blackmagic #775: Patch : Add detection for gigadevice GD32F1/GD32F3

mean00, 2020-11-20:
Add better detection for GD32F103/GD32F303 chips (the ones i have at least).

Those are  MCUs compatible with STM32F103 made by Gigadevice. They also containe some enhancements.
They come with their own CPU ID

The F3 is a buffed up F1 with a M4 core, it is not comparable to the STM32F3

The patch is just enough so that the blackmagic reports STM32F1 clone instead of unknow
It also adds GIGADEVICE designer id

In case it is helpful somehow....

[gd32f1f3.diff.gz](https://github.com/blacksphere/blackmagic/files/5573503/gd32f1f3.diff.gz)

UweBonnes, 2020-11-20:
mean00 writes:
> Add better detection for GD32F103/GD32F303 chips (the ones i have at least).
>
> Those are MCUs compatible with STM32F103 made by Gigadevice. They also containe
> some enhancements.
> They come with their [...]
Can you send the dumps of "blackmagic -t" from blackmagic/hosted for
the devices you tested. I would like to have them as reference.

Thanks
-- 
Uwe Bonnes                bon@elektron.ikp.physik.tu-darmstadt.de

Institut fuer Kernphysik  Schlossgartenstrasse 9  64289 Darmstadt
--------- Tel. 06151 1623569 ------- Fax. 06151 1623305 ---------

UweBonnes, 2020-11-20:
I applied to https://github.com/UweBonnes/blackmagic/tree/fixes

Feedback welcome!
-- 
Uwe Bonnes                bon@elektron.ikp.physik.tu-darmstadt.de

Institut fuer Kernphysik  Schlossgartenstrasse 9  64289 Darmstadt
--------- Tel. 06151 1623569 ------- Fax. 06151 1623305 ---------

mean00, 2020-11-20:
Seems to work fine, just rebuilt from the fixes branch

**Before :** 
mon swdp_scan
Available Targets:
No. Att Driver
*** 1 Unknown ARM Cortex-M Designer 751 Partno   e M4

**After : GD32F303**

Available Targets:
No. Att Driver
 1      STM32F1  VL density M4

** After: GD32F103 **
Available Targets:
No. Att Driver
 1      STM32F1 (clone) medium density M3

I'll still do the probe, just need to remember how to rebuild the sw in hosted mode.

mean00, 2020-11-20:
the output of probe both for the GD32F103 and GD32F303
Thanks a lot !
l
[probe_f1.txt.gz](https://github.com/blacksphere/blackmagic/files/5574081/probe_f1.txt.gz)
[probe_f3.txt.gz](https://github.com/blacksphere/blackmagic/files/5574082/probe_f3.txt.gz)

UweBonnes, 2020-11-21:
I still would like the  output of hosted.You only gave the output of gdb.

mean00, 2020-11-21:
the probe_f1.txt.gz & f3 above contains
Running in Test Mode
Target voltage:  Volt
RESET_SEQ failed
AP   0: IDR=24770011 CFG=00000000 BASE=e00ff003 CSW=23000040
ROM: Table BASE=0xe00ff000 SYSMEM=0x00000001, designer 751 Partno 268
0 0xe000e000: Generic IP component - Cortex-M3 SCS (System Control Space) (PIDR = 0x04000bb000  DEVTYPE = 0x00 ARCHID = 0x0000)-> cortexm_probe
Halt via DHCSR: success 01030003 after 1ms
1 0xe0001000: Generic IP component - Cortex-M3 DWT (Data Watchpoint and Trace) (PIDR = 0x04003bb002  DEVTYPE = 0x00 ARCHID = 0x0000)
2 0xe0002000: Generic IP component - Cortex-M3 FBP (Flash Patch and Breakpoint) (PIDR = 0x04002bb003  DEVTYPE = 0x00 ARCHID = 0x0000)
3 0xe0000000: Generic IP component - Cortex-M3 ITM (Instrumentation Trace Module) (PIDR = 0x04003bb001  DEVTYPE = 0x00 ARCHID = 0x0000)
4 0xe0040000: Debug component - Cortex-M3 TPIU (Trace Port Interface Unit) (PIDR = 0x04003bb923  DEVTYPE = 0x11 ARCHID = 0x0000)
5 Entry 0xfff42002 -> Not present
ROM: Table END
***  1      STM32F1 (clone) medium density M3
Ram   Start: 0x20000000, length    0x5000
Flash Start: 0x08000000, length   0x20000, blocksize    0x400

is something missing ?

UweBonnes, 2020-11-27:
Can you warp into a pull request?

UweBonnes, 2020-12-09:
Handled with #784

closed 2020-12-09 by UweBonnes