blackmagic-debug/blackmagic #778: success make but fail make probe_host=var

stkenneth, 2020-11-25:
Hi, i have success building this project when I do "make", but always fail when I do "make PROBE_HOST=swlink" tried change it to hosted,stlink and fail. but without the PROBE_HOST comment, always success.

I am on windows 10 using minGW and Msys trying to build bmp for a blue pill.

closed 2020-11-25 by stkenneth

UweBonnes, 2020-11-25:
Did you resolve the problem? Otherwise post relevant output

stkenneth, 2020-11-25:
> Did you resolve the problem? Otherwise post relevant output

yes i think i just did, but still need to test if it is successful.

what i did is make clean && make PROBE_HOST=swlink, and now it returns no error. but will test and report in a while.

StonedDawg, 2020-11-26:
ok, so `make clean && make PROBE_HOST=swlink` solved the problem