blackmagic-debug/blackmagic #780: Double free error with J-Link

Maldus512, 2020-11-28:
Hello,
I'm trying to use the PC hosted application to interact with a J-Link SWD only debugger. I can confirm the debugger itself works when used with `openocd`. Unfortunately when trying the same with `blackmagic` the procedure fails:

```
$ blackmagic -v3 -t
INFO: Open USB 06cb:0078 class ff failed
BMP hosted v1.7.1-69-gbf548e9
 for ST-Link V2/3, CMSIS_DAP, JLINK and LIBFTDI/MPSSE
Using 1366:0101 000000123456 SEGGER
 J-Link
J-Link ARM-OB STM32 compiled Aug 22 2012 19:52:04
Emulator speed 48.0 MHz, Mindiv 4
Caps 88ea5833
HW: Type 0, Major 1, Minor 17, Rev 112
SWD active, JTAG available
Running in Test Mode
Target voltage:  3.300 Volt
DPIDR 0x00000000 (v0 rev0)
DEBUG Power-Up failed
free(): double free detected in tcache 2
Aborted (core dumped)

```

I'm using the latest built tool from the master branch. Am I missing something or is this a bug? The same installation works when connected to an ST-Link V2.

UweBonnes, 2020-11-28:
What is your target? Is there a target connected at all? I can reproduce on a Tiny Gecko Board with the target switched off.

Maldus512, 2020-11-28:
It happens with or without target, no difference. I'm trying to flash a blue pill (stm32f103).

UweBonnes, 2020-11-28:
In a similar situation I get:Energy Micro EFM32 compiled Dec 10 2012 15:59:12
Emulator speed 48.0 MHz, Mindiv 4
Caps aaea5833
HW: Type 0, Major 1, Minor 17, Rev 112
SWD active, JTAG not available
Running in Test Mode
Target voltage:  0.886 Volt
DPIDR 0x00000000 (v0 rev0)
DEBUG Power-Up failed
No target found

So no abort.

Your setup works successfull with openocd without any change in wiring? Please drop a -v31 log.

UweBonnes, 2020-11-28:
Compiling with ASAN show the double free.

Maldus512, 2020-11-28:
> Your setup works successfull with openocd without any change in wiring? Please drop a -v31 log.
Exactly. Output of `openocd`:

```
$ openocd -f openocd.cfg -c "program bin/stm32/bluepill.elf verify reset exit"
Open On-Chip Debugger 0.10.0+dev-01336-g68611efc-dirty (2020-07-21-12:17)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : J-Link ARM-OB STM32 compiled Aug 22 2012 19:52:04
Info : Hardware version: 7.00
Info : VTarget = 3.300 V
Info : clock speed 1000 kHz
Info : SWD DPIDR 0x1ba01477
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f1x.cpu on 3333
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000354 msp: 0x20000800
** Programming Started **
Info : device id = 0x20036410
Info : flash size = 64kbytes
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
shutdown command invoked
```

where `openocd.cfg` is:

```
source [find interface/jlink.cfg]
transport select swd

source [find target/stm32f1x.cfg]
```

Here is the `blackmagic -v31` log:
```
 Send (  1): 01 Rec (2/2)7000
 Rec (112/112)4a2d4c696e6b2041.524d2d4f42205354.4d333220636f6d70.696c656420417567
 Send (  1): c0 Rec (6/6)006cdc020400
 Send (  1): e8 Rec (4/4)3358ea88
 Send (  1): f0
 Rec (4/4)70110100
 Send (  2): c7fe Rec (4/4)01000000
 Send (  2): c7ff Rec (4/4)03000000
 Send (  1): dd
 Send (  1): 07 Rec (8/8)e40c000101000101
 Send (  2): c7ff Rec (4/4)03000000
 Send (  2): c701 Rec (4/4)01000000
 Send (  3): 05d007
 Send ( 38): cf008800ffffffff.ffffffffffffffff.ffffffffffffffff.ffffffff9ee7ffff.
             ffffffff0000 Rec (17/17)ffffffffffffff9e.e7ffffffffffff00.00
 Rec (1/1)00
 Send (  8): cf000b00fffea500. Rec (2/2)a501
 Rec (1/1)00
 Send ( 14): cf00230000000000.fe0000000000 Rec (5/5)0000000000
 Rec (1/1)00
 Send (  8): cf000b00fffe8d00. Rec (2/2)8d01
 Rec (1/1)00
 Send ( 14): cf00230000000000.fe0000000000 Rec (5/5)0000000000
 Rec (1/1)00
 Send (  8): cf000b00fffe8d00. Rec (2/2)8d01
 Rec (1/1)00
 Send ( 14): cf00230000000000.fe0000000000 Rec (5/5)0000000000
 Rec (1/1)00
Read  CTRL/STAT: 0x00000000
Write CTRL/STAT: 0x50000000
[... 2804 lines cut ...]
 Send (  8): cf000b00fffe8d00. Rec (2/2)8d01
 Rec (1/1)00
 Send ( 14): cf00230000000000.fe0000000000 Rec (5/5)0000000000
 Rec (1/1)00
Read  CTRL/STAT: 0x00000000
 Send (  8): cf000b00fffe8d00. Rec (2/2)8d01
 Rec (1/1)00
 Send ( 14): cf00230000000000.fe0000000000 Rec (5/5)0000000000
 Rec (1/1)00
Read  CTRL/STAT: 0x00000000
 Send (  8): cf000b00fffe8d00. Rec (2/2)8d01
 Rec (1/1)00
 Send ( 14): cf00230000000000.fe0000000000 Rec (5/5)0000000000
 Rec (1/1)00
Read  CTRL/STAT: 0x00000000
 Send (  8): cf000b00fffe8d00. Rec (2/2)8d01
 Rec (1/1)00
 Send ( 14): cf00230000000000.fe0000000000 Rec (5/5)0000000000
 Rec (1/1)00
Read  CTRL/STAT: 0x00000000
free(): double free detected in tcache 2
```

> Compiling with ASAN show the double free.
What does ASAN stand for?

UweBonnes, 2020-11-28:
ASAN is an access sanitizer for Linux, instrumenting malloc and friends to hep to find double free or leaking memory. Double free is one part of the problem. I have a fix for it, 

--- a/src/platforms/hosted/jlink_adiv5_swdp.c
+++ b/src/platforms/hosted/jlink_adiv5_swdp.c
@@ -186,8 +186,6 @@ int jlink_swdp_scan(bmp_info_t *info)
 
        jlink_adiv5_swdp_error(dp);
        adiv5_dp_init(dp);
-       if (!target_list)
-               free(dp);
        return target_list?1:0;
 }

and am just checking if that fix is right also for src/target/adiv5_swdp.c.

But main problem is, that  no valid SWD data is seen.

UweBonnes, 2020-11-28:
B.t.w, it helps others to scan your log if you do some work and replace repeated output  by an elipsis (...). CTRL/STAT of the DP always return 0 while we wait for some bits to appear.

Is the reset line connected. Can you check if the reset line is active unwanted while we try to attach?

Maldus512, 2020-11-28:
I'm not using a reset line, just GND, 3v3, SWDIO and SWCLK

UweBonnes, 2020-11-28:
My fixes branch should fix the crash and similar problems.The main problem is, that on your setup no communication to the target succeeds. jlink swd low access  had an error to always reads ACK[1:2] as 0.  So maybe in your setup above the target signalling a protocoll error (ACK == 0x7) was masked.  Can you test with my fixes branch and post the line between "Write CTRL/STAT: 0x50000000" and the next "Read". And I will try on a standalone JLINk next week,

Maldus512, 2020-11-28:
Indeed your fix addresses the `double free` crash, and now the command fails gracefully by signalling that no target is found.
The output is significantly shorter this time, and I cannot find the lines you mention.

```
$ blackmagic -v31 -t
INFO: Open USB 06cb:0078 class ff failed
BMP hosted v1.6.1-656-g89b692b
 for ST-Link V2/3, CMSIS_DAP, JLINK and LIBFTDI/MPSSE
Using 1366:0101 000000123456 SEGGER
 J-Link
 Send (  1): 01 Rec (2/2)7000
 Rec (112/112)4a2d4c696e6b2041.524d2d4f42205354.4d333220636f6d70.696c656420417567
J-Link ARM-OB STM32 compiled Aug 22 2012 19:52:04
 Send (  1): c0 Rec (6/6)006cdc020400
Emulator speed 48.0 MHz, Mindiv 4
 Send (  1): e8 Rec (4/4)3358ea88
Caps 88ea5833
 Send (  1): f0
 Rec (4/4)70110100
HW: Type 0, Major 1, Minor 17, Rev 112
 Send (  2): c7fe Rec (4/4)01000000
 Send (  2): c7ff Rec (4/4)03000000
SWD active, JTAG available
 Send (  1): dd
Running in Test Mode
 Send (  1): 07 Rec (8/8)e40c000101000101
Target voltage:  3.300 Volt
 Send (  2): c7ff Rec (4/4)03000000
 Send (  2): c701 Rec (4/4)01000000
 Send ( 38): cf008800ffffffff.ffffffffffffffff.ffffffffffffffff.ffffffff9ee7ffff.
             ffffffff0000 Rec (17/17)ffffffffffffff9e.e7ffffffffffff00.00
 Rec (1/1)00
 Send (  8): cf000b00fff0a500. Rec (2/2)a507
 Rec (1/1)00
Protocol 7
 Send ( 42): cf00980000000000.00ffffffffffffff.ffff00000000e000.f7b77552ffffffff.
             ffffff00a5000000.f000 Rec (19/19)ffffffffffffffff.ffffffff00a5ffff.ffff1f
 Rec (1/1)00
 Send (  8): cf000b00fff08d00. Rec (2/2)8d07
 Rec (1/1)00
Protocol 7
 Send ( 42): cf00980000000000.00ffffffffffffff.ffff00000000e000.f7b77552ffffffff.
             ffffff00a57f0000.f000 Rec (19/19)ffffffffffffffff.ffffffff00a5ffff.ffff1f
 Rec (1/1)00
Invalid DP idcode 00000000
No target found
```

UweBonnes, 2020-11-28:
The device signals "protocol error" when asked for the DP idcode. I do not know why. If you can look with a scope or logic analyser, that would be fine. Otherwise I will try  reproduce with a stand-alone jlink hopefully next week and come back.

Maldus512, 2020-11-28:
I am in no rush, at this point I'm just curious.
I do have access to an oscilloscope at work, so I could take a quick glance at the problem next week as well; do you have a clue about what should I look for exactly?

UweBonnes, 2020-11-28:
Let me first have a try with e standalone jlink next week. I will come back...

UweBonnes, 2020-11-29:
Fixed in git head.

closed 2020-11-29 by UweBonnes

UweBonnes, 2020-11-30:
I can connect with a standalone Segger via SWD with success.

Maldus512, 2020-11-30:
I had the chance to try a second JLink debugger and it worked successfully for me as well. To further clarify: previously I was using a small, SWD only (and possibly not Segger-produced, I can't remember where I got it from) JLink compatible debugger. 

[image]

Here are the two debuggers. The one on the left works without issues with `blackmagic`, while the one on the right doesn't. I would consider it broken if it wasn't for the `openocd` route working successfully. It is marked as "JLINK-OB-V1.2".

At this point do whatever you want with this information: I initially used this debugger because I was stuck home and had nothing else; now I can work with more reliable tools. Thank you for assisting me up to this point.

UweBonnes, 2020-11-30:
Having a problem unsolved is unsatisfying but without a way to reproduce I can not solve. The firmware seems to be the lastest :https://www.segger.com/products/debug-probes/j-link/models/other-j-links/st-link-on-board/#firmware-version-history-and-limitations
I have a Disco board reflashed with probably that firmware and did not see the error. Will have to look again.

Best for now is to open another issue, now with the real issue: Some jlink , perhaps clone, does not see the target while openocd does. If more people find that error too, it may be real.