blackmagic-debug/blackmagic #787: Loss of synchronization between BMP hosted and the BMP probe (`Timeout on read RESP` error)

stoyan-shopov, 2020-12-03:
Here is a description of the problem.

Assume that a remote packet has been received by function `gdb_getpacket()` in file `gdb_packet.c`. Assume that this is a memory read request packet - a `!HM` packet. Function `gdb_getpacket()` has its packet argument set to the static `pbuf` array. It passes this pointer to function `remotePacketProcess()`, which then passes it to `remotePacketProcessHL()`, which eventually passes it to `adiv5_mem_read()`, and then `firmware_mem_read()`, and then to function `extract()` to use as a temporary buffer. Assume that 4 bytes of memory are being read, and the first byte is a `$` sign. Now, when control is eventually returned to function `gdb_getpacket()`, the loop condition is checked - `} while (packet[0] != '$');` the loop is terminated, and at this point the hosted BMP and the BMP are out of synchronization. The firmare fails to detect the next remote packet, and does not respond, which leads to the `Timeout on read RESP` error from BMP hosted.

This situation can probably be remedied in `gdb_getpacket()`, e.g., by setting the first byte of `pbuf` to zero.

stoyan-shopov, 2020-12-03:
Strangely, I am only seeing this on an stlinkv3, I don't get this problem with latest master built for stlink. I will investigate more

UweBonnes, 2020-12-03:
It happend before I activated caches, so it should not be a missing barrier (famous last words...)

stoyan-shopov, 2020-12-03:
I think I found the culprit. It is this line in file remote.c:
```
/* Re-use packet buffer. Align to DWORD! */
void *src = (void *)(((uint32_t)packet + 7) & ~7);
```
In my setup, for the stlink host, the `pbuf` variable is located on address `0x200001dc`, which is not DWORD aligned, and will be reused from 4 bytes since its start. For the stlinkv3 host, it is on address `0x20000228` which is already DWORD aligned, and so will be reused from the start.

I am not sure how this issue should be safely addressed, so that the `pbuf` buffer does not get overrun, I am not familiar how many bytes can be written to this buffer from the `remotePacketProcessHL()` function.

UweBonnes, 2020-12-03:
What about
--- a/src/remote.c
+++ b/src/remote.c
@@ -381,6 +381,7 @@ void remotePacketProcessHL(uint8_t i, char *packet)
                packet += 8;
                uint32_t count = remotehston(8, packet);
                packet += 8;
+               *src = 0;
                adiv5_mem_read(&remote_ap, src, address, count);
                if (remote_ap.dp->fault == 0) {
                        _respond_buf(REMOTE_RESP_OK, src, count);
l

stoyan-shopov, 2020-12-03:
This will not work, because `adiv5_mem_read()` modifies `* src`, and in your code the zero will be overwritten.
The code `* src = 0;` just before leaving this function should work, but it looks like a botch to me.

Maybe it is best to take corrective actions in `gdb_getpacket()` and document what is being done. I am not very familiar with the blackmagic source code, but it looks like an acceptable solution to me to modify `gdb_getpacket()`.
I will make a pull request

UweBonnes, 2020-12-03:
memset(src, 0, 1)?

UweBonnes, 2020-12-03:
I do not have all hardware here for the test...

UweBonnes, 2020-12-03:
My proposal are nonsense! I suggest using 	void *src = (void *)(((uint32_t)packet + 8) & ~7); instead of 	void *src = (void *)(((uint32_t)packet + 7) & ~7);
That way the first byte is never reused.

stoyan-shopov, 2020-12-03:
Yes, that should work, but I don't know how readable it is for the casual reader. Please also see the proposed solution in my pull request, I think it is simpler, and also explicitly states what the problem is.

I also can't test right now, but judging from my debugging yesterday, this should solve the issue. I can test probably later this week, this is nothing urgent anyway, but we should be aware of the problem.

UweBonnes, 2020-12-09:
Handled in git head by #790

closed 2020-12-09 by UweBonnes