blackmagic-debug/blackmagic #789: what

mttbx, 2020-12-03:


UweBonnes, 2020-12-03:
I try to contact Esden about that. It it his design that he warped from Garteths design

sidprice, 2020-12-03:
The "Power Flag" in Kicad tells the electrical rules checker that the net it is attached to is a power net.

mttbx, 2020-12-03:
Thank you! Waiting for the answer to the second question.

esden, 2020-12-04:
The DMG1023UV-7 is a true switch Mosfet circuit. When it is off it prevents the target from being powered as well as the probe being powered by the target. Google for true switch circuits. For example: https://www.homemade-circuits.com/bidirectional-switch/ You can build it with N-MOSFET and P-MOSFET but I found that the P-MOSFET version worked better.

mttbx, 2020-12-05:
Thank you for your guidance!

closed 2020-12-05 by mttbx