blackmagic-debug/blackmagic #792: STM32F103C8T6 SUPPORT

jhonToni, 2020-12-04:
Is possible install blackmagic firmware on a STM32F103C8T6 MCU ?

UweBonnes, 2020-12-04:
Read src/platforms/stlink/Flashsize_F103

jhonToni, 2020-12-04:
I have executed this commands:
1) git clone https://github.com/blacksphere/blackmagic.git
2) cd blackmagic
3) git submodule init
4)  git submodule update
5) make PROBE_HOST=stlink
Now I have the two files:
1) `blackmagic_dfu.bin` loaded into 0x8000000
 2) `blackmagic.bin` loaded into 0x8002000
(I  have flashed the device with  a usb to serial adapter. The device boot 0 pin  is High)

After the flash I setted the boot 0  to LOW.

When I connect the device to my Windows 10 pc  the device is not recognized.
Why ?

UweBonnes, 2020-12-04:
"not recognized" is not exact. Does no USB device appear in the device manager? Than you probably have a clone F103 or much less probable a F103x8 with bad flash above 64 k. With reset, bootloader jumps to 0x08002000 and program crashed with bad access above 64k. Otherwise, what devices appears with replug?

Flashing with scripts/stm32mem.py checks flash above and would show bad flash, but needs python, some pyton module and probably libusb and probably "zadig" .

pvonmoradi, 2020-12-27:
@jhonToni It's better to use [swlink platform](https://github.com/blacksphere/blackmagic/tree/master/src/platforms/swlink) for the bluepill as it re-purposes the 4 broken-out sw pins as bmp sw pins.

UweBonnes, 2021-04-03:
Seems resolved, not more reaction.

closed 2021-04-03 by UweBonnes