blackmagic-debug/blackmagic #796: Can I debug two device with only one probe?

mttbx, 2020-12-07:
Can I debug two device with only one probe?
1. one use swd, the other use jtag
2. both use jtag

mttbx, 2020-12-08:
I study this question a little while, and find out I can do both. If I am wrong, please tell me.

mttbx, 2020-12-08:
I find that target_list_free is called in jtag_scan and adiv5_swdp_scan. Why should we need to target_list_free before? I mean we can only delete all jtag or swd targets.

UweBonnes, 2020-12-08:
At the moment, jtag and swd share probe pins, so either jtag or swd is possible, but not both at the same time. Even on a jtag chain with several cortex-m target, the gdb server can only talk to one device at a time. To changes, the gdb server needs to get reworked probably a lot.

mttbx, 2020-12-08:
Yes, I just realize that. Thank you for your reply!

mttbx, 2020-12-09:
Here's one more question. If I have two chips, one use jtag and the other use swd. I only debug one chip at a time and jtag and swd share pins. Would they interfer with each other? I mean if I debug one chip with swd, will the other chip which use jtag protocol just ignore the swd signal?

UweBonnes, 2020-12-09:
It could work, Please try and report

mttbx, 2020-12-09:
I did a serial of experiments. Here's the conclusion.
```
1. two swd stm32, can't debug, it seems f103cb didn't support multi-drop.
2. one stm32(program jtag), jtag_scan/swdp_scan can both debug it.
3. two stm32(one program jtag, the other program swd), swdp_scan can't debug, jtag_scan can debug one of them.
4. one riscv(program jtag), one stm32(program swd), can debug them.
5. two stm32(program jtag), （non daisy chain), jtag_scan can't debug, but it can pause both of them.
6. two stm32(program jtag), （daisy chain), can debug them.

It seems stm32 will respond to swd signal even when the program use jtag.
```

closed 2020-12-09 by mttbx

UweBonnes, 2020-12-09:
1: DP on most STM32  is verson 1. Only Version2 can support multi-drop, but that is not requires.  BMP does not yet support multi drop
2: Normal setup
3. Both devices see the "switch to swd" sequence, so both react on SWD and collisions will happen 
4: Probably STM32 TDI/TDO was not connected. As riscv has no SWD, the switch sequence will only seen by the STM32 and either the stm32 answers on TMS or the risc on TDI and so there are no collisions
5. Collisions on TDI will happen.
6 . Indented setup, but not yet tested by me. Thanks for the test!