blackmagic-debug/blackmagic #798: need permission to gdb in linux

mttbx, 2020-12-08:
This can give us access without the need to use permission.
```
SUBSYSTEM=="tty", ACTION=="add", ATTRS{interface}=="Black Magic GDB Server", SYMLINK+="ttyG", MODE="0666", TAG+="uaccess", GROUP="plugdev"
SUBSYSTEM=="tty", ACTION=="add", ATTRS{interface}=="Black Magic UART Port", SYMLINK+="ttyU", MODE="0666", TAG+="uaccess", GROUP="plugdev"
SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ATTR{idVendor}=="1d50", ATTR{idProduct}=="6017", MODE="0666", TAG+="uaccess", GROUP="plugdev"
SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ATTR{idVendor}=="1d50", ATTR{idProduct}=="6018", MODE="0666", TAG+="uaccess", GROUP="plugdev"
```

eclig, 2020-12-08:
Is this an issue with the Black Magic?  Note that there is a [relevant entry in the FAQ](https://github.com/blacksphere/blackmagic/wiki/Frequently-Asked-Questions#its-annoying-to-look-up-an-always-changing-device-name-on-linux).  It could though be more prominently documented in the [Getting Started page](https://github.com/blacksphere/blackmagic/wiki/Getting-Started).

UweBonnes, 2020-12-08:
There are two issues:
* Device naming, where you find  information in the [relevant entry in the FAQ](https://github.com/blacksphere/blackmagic/wiki/Frequently-Asked-Questions#its-annoying-to-look-up-an-always-changing-device-name-on-linux).
* Device permission: You need access permissions for the serial device and when running hosted also to the USB device
> Is this an issue with the Black Magic? Note that there is a [relevant entry in the [...]

closed 2020-12-08 by mttbx