blackmagic-debug/blackmagic #809: Build of current master head fails on Windows

sidprice, 2020-12-12:
Just rebased my blackmagic repo on the current master head and did "make clean" followed by "make", this results in linker errors:"

`c:/program files (x86)/gnu arm embedded toolchain/9 2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: platform.o: in function `platform_init':
F:\DataRoot\Projects\blackmagic\src/platforms/native/platform.c:100 undefined reference to `rcc_clock_setup_pll'
c:/program files (x86)/gnu arm embedded toolchain/9 2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: F:\DataRoot\Projects\blackmagic\src/platforms/native/platform.c:166: undefined reference to `rcc_hse_configs'
collect2.exe: error: ld returned 1 exit status
make: *** [Makefile:93: blackmagic.elf] Error 1`

UweBonnes, 2020-12-12:
Did you upgrade the submodule?

sidprice, 2020-12-12:
Yes, the submodule commit is the same as blacksphere.

libopencm3 @ 8435287

sidprice, 2020-12-12:
Wait ... I need to build libopencm3 I think I forgot that. Let me try that.

sidprice, 2020-12-12:
Yes, that was my problem. Sorry for the "noise"

closed 2020-12-12 by sidprice