blackmagic-debug/blackmagic #816: reading method for gdb usb interface

mttbx, 2020-12-19:
why gdb use polling usb method for stm32f1 but not for stm32f4?

UweBonnes, 2020-12-19:
What does the git log tell about that?

stoyan-shopov, 2020-12-19:
@mttbx If you are asking about the code in `platforms/stm32/gdb_if.c`, my understanding is as follows. I may be wrong at some points, though.

The hardware usb core for F1 is quite different than the usb core in F4 (F7 uses the same libopencm3 driver as F4, with some tweaks).

For F1, the usb packet memory for OUT usb packets (directed from the usb host to the BMP probe) is sliced into fixed memory areas, and when a packet is received in the probe for some OUT endpoint, this does not prevent receiving data for other OUT endpoints. The key thing to note is that there is a separate, dedicated memory area for each OUT endpoint.

This is different in F4 (and F7) devices. They are more high-performance devices, with more memory. The usb core memory there is organized as a large fifo memory which accepts OUT usb packets, without regard to exactly which endpoint they  are directed. There is no separate memory area for each OUT endpoint, but instead the usb packet memory is available to receive packets directed for any OUT endpoint. My guess is that this is being done for efficiency reasons. This means, however, that once a packet arrives - it must be read out of the usb packet memory fifo before any other OUT packet can be read. That packet reading is what the `gdb_usb_out_cb()` callback is used for. OUT packets arrive in a FIFO (first in, first out) manner and share a common memory area. This means that if, for example, 3 packets arrive for OUT endoints 1, 2, and 3 in the usb core, in this order - then if you want to read packet for endpoint number 3, you have no choice, but to first read packet for endpoint 1, then packet for endpoint 2, and only then you can read the packet for endpoint number 3. That is how the libopencm3 driver is organized - it calls the callback, when an OUT packet arrives, the callback is expected to read the data. Then, when control returns to libopencm3, the driver there discards any data that the callback did not read, because if it does not discard this data, it will be impossible to receive any subsequent OUT packets, and the usb communication will be blocked.

I hope this is clear.

Actually, for the `stlinkv3` hardware, there are some tweaks to the usb driver that can allow for usb packets to be handled more like in the F1 case (which simplifies the firmware), but at the cost of additional memory use in the microcontroller. I guess it was an architecture decision for the libopencm3 not to use such an approach, because it uses more memory.

mttbx, 2020-12-21:
@stoyan-shopov Nice! You explained a lot. However we can still receive data in the callback for stm32f1 like stm32f4, so that the usb driver can prepare for the next OUT packet. Even though gdb host would wait our probe to send ack, so until we read, no more OUT packets would be sent.

closed 2020-12-21 by mttbx