blackmagic-debug/blackmagic #819: STM32G070 option bytes support

fabalthazar, 2020-12-22:
Hi,
I am working on STM32G070 in my daily job and I notice the option bytes are not implemented in BMP yet.
STM32G0 support is implemented in stm32l4.c.
I actually bricked a Nucleo-G070RB by applying by mistake (before analysing BMP firmware code) STM32L4 option bytes to STM32G070.
I did a "monitor option write help" (yes it has no sense but did not throw any error) then "monitor option erase".

I have done a significant work on option bytes analysis (including PCROP_RDP option bit designed to clear the flash R/W protected pages, not much documented by the manufacturer, even in the Reference Manual Rev 5).
I would like to contribute to the project in that perspective. So my questions are:

- Is there a special meaning for STM32Gx support residing in stm32l4.c? Memory space?
- Is there a special objection if I suggest a contribution with a separate stm32g0.c file? Given that there is already some if() existing in stm32l4.c to differentiate chips and STM32G0x support could grow as:
    - STM32G0[B,C]x are arriving with more than 128 kBytes of Flash;
    - STM32G05x is missing (more RAM than G03x);
    - Other contributions possible...

Is my help welcome?
What do you suggest to me? In stm32l4.c or a separate file?

UweBonnes, 2020-12-22:
Hello Baltazar(?)

fabalthazar writes:
> Hi,
> I am working on STM32G070 in my daily job and I notice the option bytes are not
> implemented in BMP yet.
> STM32G0 support is implemented in stm32l4.c.
> I actually bricked a Nucleo-G070RB by [...]

It is a mixture of code reuse, lazyness and not endless flash space (on the F103xB).

>   • Is there a special objection if I suggest a contribution with a separate
>     stm32g0.c file?

Before things get out of hand and sprinkled with to many #ifs or if
things get cleaner, make your own file.
> Given that there is already some if() existing in stm32l4.c
>     to differentiate chips and STM32G0x support could grow as:
>       □ STM32G0[B,C]x are arriving with more than 128 kBytes of Flash;
>       □ STM32G05x is missing (more RAM than G03x);
Now knowing with probe that we have a G0 by the romtable ids will help
too.

>       □ Other contributions possible...
>
> Is my help welcome?

Very much.
Device contribution is mostly concentrated in few spaces with few chances for
collateral damage.

> What do you suggest to me? In stm32l4.c or a separate file?

I thinks separate files are good, considered that 32WB, 32WL and L5 also
have the L4 flash IP. Maybe some stm32lp_common.c is needed just like
lpc_common. #707 and #458 have similar issues.

B.t.w, you know that for developping algos, hosted is best used. No
need for constant reflash, only reflash if (you think) you are done.

Useful discussions sometims also on the discord channel.

Seasons greetings

fabalthazar, 2021-02-16:
Hello Uwe,
Thanks for your explanations and guidelines. Sorry for the late reply.

Well, my needs at work on STM32G070 pushed me to contribute a little bit further than just option bytes.
As you said, STM32L4 is already sprinkled with some #ifs and #switches, and considered that #707  and #458  issues have still not found a proper way of merging, I wrote a separate stm32g0.c file (kind of sandbox).
https://github.com/fabalthazar/blackmagic/blob/target-STM32G0/src/target/stm32g0.c

As is, it can legitimately be considered as bloatware as it adds ~2k on the final binary (with L4 implem kept)!
So here is a summary of the features dispersed all along the file.
In case some of these could be of interest upstream, I will be happy to discuss.
In case not, I will let the branch online for future reference if needed.

- Support for STM32G05x/G06x;
- Support for STM32G0Bx/G0Cx (dual-bank). Chips not released yet! Aligned on rev 5 reference manuals. To test;
- Really enable debug in low power modes (DBG peripheral clock enabled). Current L4 implementation may not work (it does not on G0 chips);
- Watchdogs (IWDG and WWDG) are stopped when the core is halted. I found it to be the only way to program a binary file with hosted blackmagic where the debug registers cannot be written (I'm wrong?). Already appeared in #117 ;
- Lock the Flash back after a flash operation to leave the device in a state as close as possible to its previous state;
- Dual-bank flash erasure on coming STM32G0Bx/G0Cx devices seems to be different from STM32L4 API as far as I understand the reference manuals. G0 pages count above 256. To test once chips are released;
- Properly clear EOP flag after a flash erasure/write as recommended;
- EMPTY bit cleared after a successful programming to allow rebooting in Main Flash memory without a necessary power off. Particularly useful in case of virgin programming in production where subsequent actions are done automatically, for instance;
 Option bytes:
- G0-specific default values. I did not reproduce an "option erase" command on G0 with current STM32L4 values so I cannot confirm whether it is the reason for my bricked Nucleo board;
- Option bytes support for G0 dual-bank devices;
- Flexible write command "monitor option write _addr_ _val_ [_addr_ _val_]" to allow writing a subset of option bytes;
- Really erase all option bytes with "erase" command by taking care of clearing PCROP in two steps if needed (semi-auto two steps as "auto rescan and attach" is not supported in BMP);
- RDP Level 2 is protected/allowed by a "irreversible (enable|disable)" command to still make it possible if one really wants it.

With hindsight, STM32 target drivers often share the same base. And redondant char strings may occupy non negligible ROM space (I have no idea how much).
Anyway. When F103 will be short in memory, the refactoring will not be easy (if ever possible)... :)

BTW, I have gone through Discord too.

Greetings
Fabrice

UweBonnes, 2021-02-16:
Well, if the F103 runs out of space, a lot of things can be done. Switching to F723 is one thing, savings are also possible in many ways. And exclusion of new functionality is a possibility too, so a G0 file is welcome. And we still are some way off the flash limit.
Otherwise I assume a lot of problems you solve above still apply to the L4 file. So having one single file where a fix to G0 also fixes the other parts  has many profits. As we now know what part we probe, target and command structure can be different when needed. B.t.w, I was wrong with the assumption of #if sprinkled code. There in NULL #if  in stm32l4.c

Let me have  a look at your branch.

#707 has not seen any user feedback and I have not yet worked with it, so I kept it pending. Just going to rebase...

fabalthazar, 2021-02-16:
My fault: I meant if() and while() and not #if. To differentiate chips particularities.
The only actual problem this G0 file could solve on L4 is the debug in low power mode where the DBG clock is not enabled so the registers are not written. Others (EMPTY bit, WDG stopped, PCROP clear...) can be considered as making life easy and predictable.

If I had a STM32L55 at hand, I would have tested...

UweBonnes, 2021-02-16:
Still no link...

fabalthazar, 2021-02-17:
No link to my branch? Curious. I put it in my previous comment.
This one: [stm32g0.c](https://github.com/fabalthazar/blackmagic/blob/target-STM32G0/src/target/stm32g0.c)

fabalthazar, 2021-02-18:
This branch proposes an integration alonside current L4 driver and builds well: [target-STM32G0-with-L4](https://github.com/fabalthazar/blackmagic/commits/target-STM32G0-with-L4).
Previous [target-STM32G0](https://github.com/fabalthazar/blackmagic/commits/target-STM32G0) branch is isolated, does not compile.

UweBonnes, 2021-02-18:
Argh, wrong PR.

PR #841 (2021-03-29): Comprehensive STM32G0 driver
Fixes #819.

This PR provides a comprehensive STM32G0 target driver in a separate file. G0 has some divergences from L4/G4.
And may leave cleaner room in stm32l4.c for #707 and #458. Maybe...

Refer to this message for content details:
https://github.com/blacksphere/blackmagic/issues/819#issuecomment-780142079

Waiting for Dual-bank G0B/C release (in the ST's pipe) might be reasonable to fully test it.
closed 2021-03-29 by UweBonnes