blackmagic-debug/blackmagic #822: why does bm write to reset vector when it should write to ram

chriskuku, 2020-12-25:
This little assembler program loads the address of ram base+4 to r1
and is supposed to store a value (0x188) into that location.

It stores it also 

```
$ ./GDB_bmp 
GNU gdb (GNU Tools for Arm Embedded Processors 8-2018-q4-major) 8.2.50.20181213-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-apple-darwin10 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
Available Targets:
No. Att Driver
 1      STM32F40x M4
0x5b33b9be in ?? ()
(gdb) load
Loading section .text, size 0x450 lma 0x0
Start address 0x440, load size 1104
Transfer rate: 119 KB/sec, 552 bytes/write.
(gdb) s
36	      ldr r0,=ram+4
(gdb) x /10x 0
0x0 <Bottom>:	0x40000000	0x00000440	0x0000044c	0x0000044c
0x10 <Bottom+16>:	0x0000044c	0x0000044c	0x0000044c	0x00000000
0x20 <Bottom+32>:	0x00000000	0x00000000
(gdb) s
37	      str r1, [r0]
(gdb) i r
r0             0x20000004          536870916
r1             0x188               392
r2             0x0                 0
r3             0x0                 0
r4             0x0                 0
r5             0x0                 0
r6             0x0                 0
r7             0x0                 0
r8             0x0                 0
r9             0x0                 0
r10            0x0                 0
r11            0x0                 0
r12            0x0                 0
sp             0x8fec498           0x8fec498
lr             0xfffffff9          0xfffffff9
pc             0x446               0x446 <Reset+6>
xpsr           0x1000003           16777219
fpscr          0x0                 0
msp            0x8fec498           0x8fec498
psp            0x0                 0x0 <Bottom>
primask        0x0                 0 '\000'
basepri        0x0                 0 '\000'
faultmask      0x0                 0 '\000'
control        0x0                 0 '\000'
(gdb) s
Loop () at simple.s:38
38	Loop:    b Loop
(gdb) x /10x 0
0x0 <Bottom>:	0x40000000	0x00000188	0x0000044c	0x0000044c
0x10 <Bottom+16>:	0x0000044c	0x0000044c	0x0000044c	0x00000000
0x20 <Bottom+32>:	0x00000000	0x00000000
(gdb) 
$ ./GDB_bmp 
GNU gdb (GNU Tools for Arm Embedded Processors 8-2018-q4-major) 8.2.50.20181213-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-apple-darwin10 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
Available Targets:
No. Att Driver
 1      STM32F40x M4
0x5b33b9be in ?? ()
(gdb) load
Loading section .text, size 0x450 lma 0x0
Start address 0x440, load size 1104
Transfer rate: 119 KB/sec, 552 bytes/write.
(gdb) s
36	      ldr r0,=ram+4
(gdb) x /10x 0
0x0 <Bottom>:	0x40000000	0x00000440	0x0000044c	0x0000044c
0x10 <Bottom+16>:	0x0000044c	0x0000044c	0x0000044c	0x00000000
0x20 <Bottom+32>:	0x00000000	0x00000000
(gdb) s
37	      str r1, [r0]
(gdb) i r
r0             0x20000004          536870916
r1             0x188               392
r2             0x0                 0
r3             0x0                 0
r4             0x0                 0
r5             0x0                 0
r6             0x0                 0
r7             0x0                 0
r8             0x0                 0
r9             0x0                 0
r10            0x0                 0
r11            0x0                 0
r12            0x0                 0
sp             0x8fec498           0x8fec498
lr             0xfffffff9          0xfffffff9
pc             0x446               0x446 <Reset+6>
xpsr           0x1000003           16777219
fpscr          0x0                 0
msp            0x8fec498           0x8fec498
psp            0x0                 0x0 <Bottom>
primask        0x0                 0 '\000'
basepri        0x0                 0 '\000'
faultmask      0x0                 0 '\000'
control        0x0                 0 '\000'
(gdb) s
Loop () at simple.s:38
38	Loop:    b Loop
(gdb) x /10x 0
0x0 <Bottom>:	0x40000000	0x00000188	0x0000044c	0x0000044c
0x10 <Bottom+16>:	0x0000044c	0x0000044c	0x0000044c	0x00000000
0x20 <Bottom+32>:	0x00000000	0x00000000
(gdb) 
(gdb) x /10x 0x20000000
0x20000000:	0x40000000	0x00000188	0x0000044c	0x0000044c
0x20000010:	0x0000044c	0x0000044c	0x0000044c	0x00000000
0x20000020:	0x00000000	0x00000000
(gdb) 
```
Why does it store in both regions? It causes all kinds of havoc when I press Reset (since the Reset vector is overwritten).
Maybe I'm misunderstandig something but I cannot get any senseful usage from blackmagic debugging right now. 

Firmware v1.7.1-69

Help appreciated.

chriskuku, 2020-12-25:
Wanted to edit my post to correct some things but it doesn't seem to work. Forgot to post the program itself:

 ```
      .syntax unified
        .cpu cortex-m4
        .thumb
        .global Reset,Bottom,Loop
        .section vectors
        .equ ram,0x20000000
        .equ dict,0x188
        .equ stack,0x40000000
        .text
        .org 0
        .align 2
Bottom:
        .long stack             /* 0x00000000 The initial stack pointer (defined by the linker-script) */
                                /* Address:   Exception Vector Description: */
        .long Reset             /* 0x00000004 The startup-code, the code that runs on power-on or RESET */
        .long NMI_Handler       /* 0x00000008 Non-Masktable Interrupt, this can not be stopped, preempted or prevented */
        .long HardFault_Handler /* 0x0000000c Hard Fault, all classes of Fault */
        .long MemManage_Handler /* 0x00000010 Memory Management, MPU mismatch, including Access Violation and No Match */
        .long BusFault_Handler  /* 0x00000014 Bus Fault, Pre-Fetch- Memory Access Fault, other address/memory related Fault */
        .long UsageFault_Handler /* 0x00000018 Usage Fault, i.e. Undefined Instructions, Illegal State Transitions */
        .long 0                 /* 0x0000001c */
        .long 0                 /* 0x00000020 */
        .long 0                 /* 0x00000024 */
        .long 0                 /* 0x00000028 */
        .long SVC_Handler       /* 0x0000002c Supervisor Call */
        .long DebugMon_Handler  /* 0x00000030 Debug Monitor */
        .long 0                 /* 0x00000034 */
        .long PendSV_Handler    /* 0x00000038 Pending Service, pending requests for system service */
        .long SysTick_Handler   /* 0x0000003c System Tick Timer (this may not exist on all implementations) */

        .skip 0x400

       /* you can place your startup code here if you wish. */
Reset:
      ldr r1,=dict
      ldr r0,=ram+4
      str r1, [r0]
Loop:    b Loop

NMI_Handler:
HardFault_Handler:
MemManage_Handler:
BusFault_Handler:
UsageFault_Handler:
SVC_Handler:
DebugMon_Handler:
PendSV_Handler:
SysTick_Handler:
       .end
```

chriskuku, 2020-12-25:
I'm uploading some files.

```
memmap:

MEMORY
{
   rom(RX)   : ORIGIN = 0x00000000, LENGTH = 0x4000
   ram(WAIL) : ORIGIN = 0x20000000, LENGTH = 0x4000
}

SECTIONS
{
   .text : { *(.text*) } > rom
   .bss  : { *(.bss*) } > ram
}
Makefile:
.SUFFIXES:
.SUFFIXES: .elf .o .bin .s .c .h
.PRECIOUS: %.o
DEBUG=-g
ARMGNU?=arm-none-eabi

#COPS = -Wall  -Os -nostdlib -nostartfiles -ffreestanding -save-temps
AOPS = --warn --fatal-warnings

all : simple.elf #myfirst.elf

.s.o :
        $(ARMGNU)-as $(DEBUG) -al -o $@ $< >$*.lst

.o.elf:
        $(ARMGNU)-ld  $(DEBUG)  -e Reset -o $@ -T memmap $<
#       $(ARMGNU)-ld  $(DEBUG)  -e Reset -o $@ $<
        $(ARMGNU)-objdump -D $@ > $*.list
        $(ARMGNU)-objcopy $@ $*.bin -O binary
        $(ARMGNU)-nm $@
        cp $@ firmware.elf

clean:
        rm -f *.bin *.o *.list *.lst *.elf

```

UweBonnes, 2020-12-26:
I already tried to answer on mikrocontroller.net. Forgot to send :-(
You map at 0, where there is only alias memory. "info mem" in gdb  shows no flash memory mapped at 0. Flash your program to 0x08000000 with e.g. "blackmagic -a 0x08000000 simple.bin" and will things look much better

chriskuku, 2020-12-26:
But I want to use symbols and .elf, does .bin get me gdb debugging facility?

When I map rom to 0x08000000 I don't get what I expect at address 0x00000000, the vectors.
```

$ ./GDB_bmp 
GNU gdb (GNU Tools for Arm Embedded Processors 8-2018-q4-major) 8.2.50.20181213-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-apple-darwin10 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
Available Targets:
No. Att Driver
 1      STM32F40x M4
0x5b37b93e in ?? ()
(gdb) load
Loading section .text, size 0x450 lma 0x8000000
Start address 0x8000440, load size 1104
Transfer rate: 2 KB/sec, 552 bytes/write.
(gdb) x /10x 0
0x0:	0x0cfec4ba	0x45d7ebea	0x98904420	0x5b37b93e
0x10:	0xb14eaf54	0x77f928bd	0x2f0328a9	0x5e65553d
0x20:	0xa95cee38	0xbfdbba7c
(gdb) 

```
this is garbage.

chriskuku, 2020-12-26:
A question to this forum: is editing a post not possible? When I click on EDIT in the upper right corner drop down, nothing happens.

chriskuku, 2020-12-26:
A question about technique of this forum: is editing a post not possible? When I click on EDIT in the upper right corner drop down, nothing happens.

chriskuku, 2020-12-26:
And more quirks:

I don't want to see my code relocated to 0x08000000. And my program still writes to location 0x000004.

```
$ ./GDB_bmp 
GNU gdb (GNU Tools for Arm Embedded Processors 8-2018-q4-major) 8.2.50.20181213-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-apple-darwin10 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
Available Targets:
No. Att Driver
 1      STM32F40x M4
0x5b37b93e in ?? ()
(gdb) load
Loading section .text, size 0x450 lma 0x8000000
Start address 0x8000440, load size 1104
Transfer rate: 2 KB/sec, 552 bytes/write.
(gdb) x /10x 0
0x0:	0x0cfec4ba	0x45d7ebea	0x98904420	0x5b37b93e
0x10:	0xb14eaf54	0x77f928bd	0x2f0328a9	0x5e65553d
0x20:	0xa95cee38	0xbfdbba7c
(gdb) s
Reset () at simple.s:36
36	      ldr r0,=ram+4
(gdb) x /10x 0
0x0:	0x0cfec4ba	0x45d7ebea	0x98904420	0x5b37b93e
0x10:	0xb14eaf54	0x77f928bd	0x2f0328a9	0x5e65553d
0x20:	0xa95cee38	0xbfdbba7c
(gdb) i mem
Using memory regions provided by the target.
Num Enb Low Addr   High Addr  Attrs 
0   y  	0x08000000 0x08010000 flash blocksize 0x4000 nocache 
1   y  	0x08010000 0x08020000 flash blocksize 0x10000 nocache 
2   y  	0x08020000 0x08100000 flash blocksize 0x20000 nocache 
3   y  	0x10000000 0x10010000 rw nocache 
4   y  	0x20000000 0x20050000 rw nocache 
(gdb) x /10x 0x08000000
0x8000000 <Bottom>:	0x40000000	0x08000440	0x0800044c	0x0800044c
0x8000010 <Bottom+16>:	0x0800044c	0x0800044c	0x0800044c	0x00000000
0x8000020 <Bottom+32>:	0x00000000	0x00000000
(gdb) s
37	      str r1, [r0]
(gdb) s
Loop () at simple.s:38
38	Loop:    b Loop
(gdb) x /10x 0
0x0:	0x0cfec4ba	0x00000188	0x98904420	0x5b37b93e
0x10:	0xb14eaf54	0x77f928bd	0x2f0328a9	0x5e65553d
0x20:	0xa95cee38	0xbfdbba7c
(gdb) 

```

stoyan-shopov, 2020-12-26:
@chriskuku Just some notes. It is strange that when you write to address `0x20000000`, you are also seing the changes at address `0`, have you, by some change, remapped the SRAM memory to address `0`?

Another thing, it shouldn't be a problem, because you are not using the stack, but why do you set your stack pointer (R13) to `0x40000000`? I am looking at the `STM32F40x` reference manual, and I don't see any memory at this address.

UweBonnes, 2021-04-03:
Thread seems stale

closed 2021-04-03 by UweBonnes