blackmagic-debug/blackmagic #823: libftdi1 not found

chriskuku, 2020-12-26:
WHen trying to compile blackmagic with PROBE_HOST=hosted Im getting:

$ make PROBE_HOST=hosted
 GIT include/version.h
Package libftdi1 was not found in the pkg-config search path.
Perhaps you should add the directory containing `libftdi1.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libftdi1' found
Package hidapi-libusb was not found in the pkg-config search path.
Perhaps you should add the directory containing `hidapi-libusb.pc'
to the PKG_CONFIG_PATH environment variable
No package 'hidapi-libusb' found

chriskuku, 2020-12-26:
Forgot to mention that my host is macOS (Catalina 10.15.7). I'm trying to install the required packages through macports right now and see how it works out.

eclig, 2020-12-26:
You need something like "sudo apt install libftdi1-dev" or whatever it's
spelled ob a Mac.

chriskuku, 2020-12-26:
I solved it after some weird rounds through port install. I got libftdi1 installed in /opt/local and included 
-I/opt/local/include in the Makefile and still got an undefined 
```

Package hidapi-libusb was not found in the pkg-config search path.
Perhaps you should add the directory containing `hidapi-libusb.pc'
to the PKG_CONFIG_PATH environment variable
No package 'hidapi-libusb' found
```

until I found that all resolves so to say "blackmagically" when you supply the following arguments to your make command line:

`make PROBE_HOST=hosted HOSTED_BMP_ONLY=1`

With that all the 'hidapi-libusb' and libftdi1 stuff will not be required.

UweBonnes, 2020-12-27:
HOSTED_BMP_ONLY=1 on MACOS is not really what you want, as remote protocoll on MACOS needs some love. Normal "hosted" offers the BMP gdb server and command line for other debugers.

UweBonnes, 2021-04-03:
Seems resolved

closed 2021-04-03 by UweBonnes