blackmagic-debug/blackmagic #824: I already tried to answer on mikrocontroller.net. Forgot to send :-(

chriskuku, 2020-12-26:
I already tried to answer on mikrocontroller.net. Forgot to send :-(
You map at 0, where there is only alias memory. "info mem" in gdb  shows no flash memory mapped at 0. Flash your program to 0x08000000 with e.g. "blackmagic -a 0x08000000 simple.bin" and will things look much better

_Originally posted by @UweBonnes in https://github.com/blacksphere/blackmagic/issues/822#issuecomment-751301598_

The command doesn't look very successful. I silently runs very quick without saying anything:

`../Downloads/blackmagic/src/blackmagic -v 1 -a 0x08000000 simple.bin`

chriskuku, 2020-12-26:
To add more data (why can't I edit messages here?):

```
$ ../Downloads/blackmagic/src/blackmagic -v 3 -d /dev/cu.usbmodem7AAF76931 -a 0x08000000 simple.bin
BMP hosted (BMP Only) v1.7.1-121-g42f590c
Failed on select
Failed on select
Failed on select
Target voltage: !GV# Volt
Failed on select
Failed on select
Failed on select
Speed set to 2980.7025 MHz for SWD
Failed on select
Failed on select
swdptap_seq_out failed, error So10ffffffff#
```

chriskuku, 2020-12-26:
Or with a bit more verbosity:

```
$ ../Downloads/blackmagic/src/blackmagic -v31 -d /dev/cu.usbmodem7AAF76931 -a 0x08000000 simple.bin
BMP hosted (BMP Only) v1.7.1-121-g42f590c
+#!GA#
Failed on select
Remote is #!GA#
!GZ0#
Failed on select
 !GV#
Failed on select
Target voltage: !GV# Volt
!GF003d0900#
Failed on select
!Gf#
Failed on select
!Gf#
Failed on select
Speed set to 2980.7025 MHz for SWD
remote_swdptap_init
!SS#
Failed on select
swdptap_seq_out        16 ticks: ffffffff
!So10ffffffff#
Failed on select
swdptap_seq_out failed, error So10ffffffff#
```

UweBonnes, 2020-12-27:
You run on MACOS. hosted seems to have problems on MACOS. Somebody with MACOS needs to take care.

esden, 2021-12-16:
@chriskuku is this issue still present? Should we keep this issue open and work on improving functionality on MacOS?

dragonmux, 2023-06-08:
Given the lack of comments and replies to this and as a lot of work has been done to improve BMDA since this was opened, we are going to close this as complete, but if the issue persists, please comment and we will re-open this issue.

closed 2023-06-08 by dragonmux

chriskuku, 2023-06-08:
Could someone please give me the "make" command to build the hosted version of blackmagic under macOS. Just would like to quickly check whether the problem is gone with the latest version of blackmagic.

Sorry for the late reply.

dragonmux, 2023-06-08:
No worries, it's `make PROBE_HOST=hosted HOSTED_BMP_ONLY=0` for macOS for now

chriskuku, 2023-06-08:
Success:

```
$ ./blackmagic -v 1 -a 0x08000000  simple.bin
Black Magic Debug App v1.9.0-714-g3bc59eb7
 for Black Magic Probe, ST-Link v2 and v3, CMSIS-DAP, JLink and libftdi/MPSSE
Using 1d50:6018 799EC0D3 Black Sphere Technologies
 Black Magic Probe (STLINK), (Firmware v1.7.1-69-gbf548e9)
Probe firmware does not support the newer JTAG commands, please update it.
Target voltage: 3.29V Volt
Please update probe firmware to enable SWD/JTAG frequency selection
Device has fixed frequency for SWD
Please update probe firmware to enable high impedance clock feature
SWD access has invalid ack 6
Trying old JTAG to SWD sequence
scan_multidrop: false
Please update your probe's firmware for improved error handling
DP DPIDR 0x1ba01477 (v1 rev0) designer 0x43b partno 0xba
RESET_SEQ failed
AP   0: IDR=14770011 CFG=00000000 BASE=e00ff003 CSW=a3000040 (AHB-AP var1 rev1)
Halt via DHCSR(00030003): success after 4ms
ROM: Table BASE=0xe00ff000 SYSMEM=0x00000001, Manufacturer  20 Partno 410
0 0xe000e000: Generic IP component - Cortex-M3 SCS (System Control Space) (PIDR = 0x00000004001bb000  DEVTYPE = 0x00 ARCHID = 0x0000)
-> cortexm_probe
CPUID 0x411fc231 (M3 var 1 rev 1)
Calling stm32f1_probe
1 0xe0001000: Generic IP component - Cortex-M3 DWT (Data Watchpoint and Trace) (PIDR = 0x00000004001bb002  DEVTYPE = 0x00 ARCHID = 0x0000)
2 0xe0002000: Generic IP component - Cortex-M3 FBP (Flash Patch and Breakpoint) (PIDR = 0x00000004000bb003  DEVTYPE = 0x00 ARCHID = 0x0000)
3 0xe0000000: Generic IP component - Cortex-M3 ITM (Instrumentation Trace Module) (PIDR = 0x00000004001bb001  DEVTYPE = 0x00 ARCHID = 0x0000)
4 0xe0040000: Debug component - Cortex-M3 TPIU (Trace Port Interface Unit) (PIDR = 0x00000004001bb923  DEVTYPE = 0x11 ARCHID = 0x0000)
5 Entry 0xfff42002 -> Not present
ROM: Table END
***  1   STM32F1 medium density M3
Please update probe firmware to enable high impedance clock feature
RAM   Start: 0x20000000 length = 0x5000
Flash Start: 0x08000000 length = 0x20000 blocksize 0x400
Erasing 1280 bytes at 0x08000000
Flashing 1280 bytes at 0x08000000
Success!
Flash Write succeeded for 1280 bytes,    7.529kiB/s
Please update probe firmware to enable high impedance clock feature
```