blackmagic-debug/blackmagic #825: Why add pc by 2 in cortexm_pc_write

mttbx, 2021-01-02:
I noticed that there's cortexm_pc_write in cortexm_halt_resume function. And 0xBE00 is a bkpt instruction. Why should we add pc by 2?

stoyan-shopov, 2021-01-02:
This just skips breakpoint instructions (2 bytes long) when resuming the target. There is no point in resuming the target if you know that you have hit a breakpoint, as you will hit the breakpoint again and again. Adding 2 to the PC skips the breakpoint instruction, and only then resumes the target.  I believe such instructions are explicitly used with semihosting.

mttbx, 2021-01-03:
Yes, I think so too, but is there any doc tells us that we need to add pc by two?

stoyan-shopov, 2021-01-03:
The ARM documents, and reading the source code, I don't think there is anything better than this

mttbx, 2021-01-03:
Can you be more specific? Just a reference would be good enough.

stoyan-shopov, 2021-01-03:
You can refer to the ARM architecture reference manual for ARMv7M devices, document `ddi0403`, from ARM: https://developer.arm.com/documentation/ddi0403/latest

But, in short, when you resume a target, and there is a breakpoint on the address where the target will continue running - the target will immediately stop again on this address. If you do not skip this instruction, by adding 2 (bytes), which is the length of the `BKPT` instruction - your program will get stuck hitting the same breakpoint over and over again, and will never advance. As I said, I believe this use case is with semihosting, but I have no experience with semihosting, so cannot be 100 % sure. At least this is my understanding

mttbx, 2021-01-03:
Thank you for your help and explain! I'll check it out.

closed 2021-01-03 by mttbx