blackmagic-debug/blackmagic #826: JTAG issues

UweBonnes, 2021-01-03:
Playing with JTAG on none-Cortex targets show some JTAG problems.  https://github.com/UweBonnes/blackmagic/tree/avr  makes JTAG work in more situations. If you intend work on other targets, try that branch.

brainstorm, 2021-01-16:
@UweBonnes Perfect timing! I was playing with JTAG on non-Cortex targets too in https://github.com/espressif/openocd-esp32/issues/135#issuecomment-761525207 with another probe arrangement (Glasgow + OpenOCD remote bitbang interface)... Would JTAG work with the ESP32S2?

Bonus question: Would it work with a reprogrammed STLinkV2.1 dongle acting as BMP on that branch you are pointing out?

UweBonnes, 2021-01-16:
In principleJTAG should work.A reprogrammed STLink does not have the JTAG restrictions that a ST-Firmware STLink has. But first test that JTAG works at all with the reprogrammed STLINK. I have not reprogrammed my STLink yet and have to get heared reports,So consider the pin settings in platform.h  untested regarding TDI and TDO . And there is no work on an ESP32 JTAG server done yet to my knowledge.

brainstorm, 2021-01-17:
Ok, here's a field report for a STLinkv2 (not baite)... Compiling the stlink-tool on the `stlink21` branch:

```diff
diff --git a/src/stlink.c b/src/stlink.c
index 07eee94..07bbff4 100644
--- a/src/stlink.c
+++ b/src/stlink.c
@@ -410,7 +410,7 @@ int stlink_flash(struct STLinkInfo *info, const char *filename) {
         cur_chunk_size = chunk_size;
     }
     int wdl = 2;
-    if ((info->stinfo_bl_type == STLINK_BL_V3)) {
+    if (info->stinfo_bl_type == STLINK_BL_V3) {
        if (((base_offset + flashed_bytes) & ((1 << 14) - 1)) == 0) {
            uint32_t sector_start[8] = {0x08000000, 0x08004000, 0x08008000, 0x0800C000,
                                        0x08010000, 0x08020000, 0x08040000, 0x08060000};
```

Otherwise `stlink-tool` does not compile:

```C
$ make
cc -Wall -Wextra -Werror -I/usr/local/Cellar/libusb/1.0.23/include/libusb-1.0 -g -Og -o src/stlink.o -c src/stlink.c
src/stlink.c:413:31: error: equality comparison with extraneous parentheses [-Werror,-Wparentheses-equality]
    if ((info->stinfo_bl_type == STLINK_BL_V3)) {
         ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
src/stlink.c:413:31: note: remove extraneous parentheses around the comparison to silence this warning
    if ((info->stinfo_bl_type == STLINK_BL_V3)) {
        ~                     ^              ~
src/stlink.c:413:31: note: use '=' to turn this equality comparison into an assignment
    if ((info->stinfo_bl_type == STLINK_BL_V3)) {
                              ^~
                              =
1 error generated.
make: *** [src/stlink.o] Error 1
```

Then compiling and flashing under the `avr` branch seems to go fine:

```shell
$ make -j2 PROBE_HOST=stlink
(all good, ~91kiB `blackmagic.bin`)

$ stlink-tool blackmagic.bin
StlinkV21 Bootloader found
Firmware version : V2J17S4
Loader version : 14152
ST-Link ID : 0671FF575648854987215342
Firmware encryption key : 1D8F728E179842EBFA341D5493CC77DA
Current mode : 1
Type V2
set address to 0x08004000 done
Download at 0x08004000 done
.set address to 0x08004400 done
Download at 0x08004400 done
.set address to 0x08004800 done
Download at 0x08004800 done
(...)
Download at 0x0801a400 done
.set address to 0x0801a800 done
Download at 0x0801a800 done
.set address to 0x0801ac00 done
Download at 0x0801ac00 done
$
```

But unfortunately I cannot see any `/dev/cu.*` interface as I used to... I'll investigate.

UweBonnes, 2021-01-17:
what compiler, what version?

UweBonnes, 2021-01-17:
What "stlink21 branch"?

brainstorm, 2021-01-18:
Oh, sorry, should have indeed been more specific, here you go:

> what compiler, what version?

```
$ cc --version
Apple clang version 12.0.0 (clang-1200.0.32.28)
Target: x86_64-apple-darwin19.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

> What "stlink21 branch"?

Your stlink-tool `stlink21` branch here:

https://github.com/UweBonnes/stlink-tool/tree/stlinkv21

UweBonnes, 2021-01-18:
Ah, I did not notice the keyword "stlink-tool"!

Did you carefully follow the Readme.md given on "https://github.com/blacksphere/blackmagic/tree/master/src/platforms/stlink"

brainstorm, 2021-01-18:
Ahh, good one, sorry, I was missing:

`For ST-LINK/V2, after each stlink replug, call (...) stlink-tool without arguments`

Good one, now I can see the serial ports:

```
cu.usbmodemE3DBD7E21
cu.usbmodemE3DBD7E23
```

Now, what would be the 4 JTAG wires (TMS, TDO, TDI, TCK) mapped to on the STLinkv2? I can only seem to find SWD pinouts?:

https://github.com/blacksphere/blackmagic/blob/master/src/platforms/stlink/Connector

UweBonnes, 2021-01-18:
The JTAG wires should be the same as on the original ST STlink. From what I understand, you have an original white ST-Link-V2 with the 20 pin Jtag connector and the SWIM connector. Turn to the ST documentation to find and verify the layout.

brainstorm, 2021-01-18:
No, I have the chinese probes, NOT the original STLink white box... but I
can definitely open my chinese USB dongle and solder out some of the pins
if possible?

brainstorm, 2021-01-18:
These chinese probes do indeed seem to have plenty of spare I/O to accomodate 4 pins (either disposing of one GND, 3V3 and 5V pins or just breaking out on another connector:

https://hackaday.io/project/162597/logs

I'll investigate.

UweBonnes, 2021-01-19:
Try to use the TDI/TDO pins PA6/7 as defined in stlink/platform.h

brainstorm, 2021-01-30:
For now I went for https://github.com/jeanthom/openocd-dirtyjtag/issues/3 (substitutes SWD w/ JTAG pins, but seems to work)... but I have a modded stlinkv2 queued and ready to go if that doesn't pan out:

[image]

[image]

@UweBonnes have you ever worked with DJTAG2 or using SPI for better performance? `remote bitbang` makes those stlinkv2 probes a non-starter :-S

UweBonnes, 2021-01-30:
You write to tense, I get no sense out of what you write. Google (DJtag2) does not help and what are your performance problems or proposals?

esden, 2021-12-16:
Is this issue still on topic? Are we still talking about JTAG issues or getting stlinkv2 working?

brainstorm, 2021-12-16:
Reading through I might have derailed the original intent of this issue, sorry about that :-S

esden, 2021-12-16:
@brainstorm No worries. :) Is the original issue still unresolved? Are there any actionable items on this? Or should we close this issue. If it is still unresolved can we improve the topic to be more specific? "JTAG issues" is bit vague. This would help us to focus on a problem and try to resolve it.

brainstorm, 2021-12-16:
To @UweBonnes point:

> You write too tense, I get no sense out of what you write. Google (DJtag2) does not help and what are your performance problems or proposals?

https://github.com/jeanthom/DirtyJTAG/tree/dirtyjtag2

My performance problems were related with JTAG bitbanging on the Glasgow which are not at all related to blackmagic (back then I was learning about platforms and JTAG in general, now I know a bit better, I think).

So I guess that the subject for this issue should be changed to "JTAG support for non-cortex targets (should/could?) be merged back to master"?

esden, 2021-12-16:
@brainstorm sounds good. I have updated the topic.

esden, 2022-08-02:
After further investigation I don't think this issue is actually needed any more. Considering that @dragonmux was able to implement JTAG PDI without changing the JTAG code. Thus I am closing this issue, if there are some concrete deficiencies that need to be addressed please feel free to open a dedicated issue for that.

closed 2022-08-02 by esden