blackmagic-debug/blackmagic #827: RP2040 Support

dexterfoxy, 2021-01-26:
Any idea? Is it even possible?

UweBonnes, 2021-01-26:
Debugging should work out of the box.  Detecting needs somebody to run hosted "blackmagic -t" and post the output. Flashing needs implementation of the QSPI algo, probably doable but needs real hardware and some time.

wasv, 2021-01-26:
Came here to ask exactly this question. I have some pico's and an stlink clone with black magic probe firmware. I'll poke around at debugging and detecting later this week. @UweBonnes do you know what a good starting point in the code base would be for flashing the QSPI algo?

UweBonnes, 2021-01-26:
The Rp2040 has no internal flash. Flash for user code is connected via (Q)SPI. So if BMP should be able to program this flash this must be implemented. At the moment, no BMP device supports externl flash. So there is no good starting point yet.

wasv, 2021-01-26:
Looks like the RP2040 requires multidrop SWD support. It doesn't seem to be widely supported, the Raspberry Pi devs had to patch it into openocd, and JLinks only support it after hardware revision 10. I can't find any reference to multidrop in the blackmagic repo, so I assume this project doesn't support that either? For reference, I have found that there may be a way to debug a RP2040 using openocd and an FTDI chip (see https://github.com/raspberrypi/openocd/issues/11).

UweBonnes, 2021-01-27:
I had a short look at the multidrop doc, but to implement it, real hardware is needed. If there is already support using FTDI, porting is easier, as you can switch on the command line between the already supported program and BMP hosted.

UweBonnes, 2021-01-27:
The Pico board has no dedicated debug connector and no reset buttom. Not very friendly for bring up.

wasv, 2021-01-27:
There is a debug connector at the bottom, but I was also unable to find a reset button.
[image]

UweBonnes, 2021-01-27:
It seems, Pin30 "Run" is also "Nrst", To quote: "RUN is the RP2040 enable pin, and has an internal (on-chip) pull-up resistor to 3.3V of about ~50K Ohms. To reset RP2040, short this pin low."

UweBonnes, 2021-02-15:
Have a look at the https://github.com/UweBonnes/blackmagic/tree/rp branch. No flashing yet, but debugging and memory read.

UweBonnes, 2021-05-07:
No feedback and probably fixed by the commit.

closed 2021-05-07 by UweBonnes

Riffer, 2023-02-27:
Sorry ask in this old issue: is the pico supported now officially?