blackmagic-debug/blackmagic #829: STM32F405RG Support

justinmbrock, 2021-02-09:
Adafruit makes a Feather board with an STM32F405RG. I'd love to try to turn it into a Black Magic Probe similar to how others have converted STM32 Blue Pills, but I am not familliar with where you have your pin and hardware definitions for your firmware targets. I've tried comparing the source code for different targets hoping to find a common location where I can map the exposed pins to the different functions, but there are a lot of differences between just the native, stlink, and swlink folders. Do you have any guidance on how to proceed?

UweBonnes, 2021-02-09:
All in all, the F4 series has too few USB EP to build a full features BMP.  The f4discovery is some relict for bootstraping STlinks  by not beeing really BMP equivalent, but  modifying  should get you started.

justinmbrock, 2021-02-09:
Thank you for your quick reply! One more question before I decide if this is worth trying. How many endpoints are needed to support all BMP features?

UweBonnes, 2021-02-09:
One more. Look at how Sid price solved this with his wireless BMP.

closed 2021-06-14 by justinmbrock