blackmagic-debug/blackmagic #831: Flashing Adafruit Feather Sense

jimakos96, 2021-02-15:
I am trying flash the adafruit feather sense using the BMP ,because the board does not have an swd connector , I took the SWCLK ,SWDIO with jumper cables and connected them to the swd cable  .My problem is that when i connect both boards  (BMP and Feather ) to power and try to connect to the  GDB Serve i get the following error .

[image]

UweBonnes, 2021-02-15:
First thing fishy here is "Target Voltage 1.1V" Is the feather supply really that low? BMP native can do level translation in the range 1.8 to 5 Volt, but not 1.1 Volt. Most other BMP hardware is fixed at 3.3 V IO voltage
Otherwise when reporting problems, be more verbose:
- What exact BMP Hardware
- What firmware version
- What target board, what MCU, perhaps with a link . Do not assume that everybody knows "Adafruit Feather Sense"
- Compile hosted and run with hosted as it is much more verbose.

jimakos96, 2021-03-01:
I fixed it by upgrading the firmware in BMP and  by enabling the power from the gdb.

UweBonnes, 2021-03-01:
Good. Next time please close if resolved!

closed 2021-03-01 by UweBonnes