blackmagic-debug/blackmagic #838: Adding Platform Support for STM32F0

geraldnilles, 2021-03-06:
I am interested in porting BMP to the STM32F0 family.  Specifically, I am considering the stm32f078cb.  My goal is to make the smallest possible USB to SWD/UART bridge possible.  This particular chip is 3mmx3mm and allows crystal-less USB operation.  

Before I start investing the resources, are there any major issues that would immediately make this project DOA?  I know its a lower performance CPU (Cortex M0), but it has quite a bit of flash and the appropriate PHYs.  Is the current BMP hardware close to CPU limited? Or is there headroom to slow down the CPU performance? 

Thanks for you advice.

stoyan-shopov, 2021-03-06:
Just a note, I had ported the BMP to an STM32F070RB device (128 KB flash, 16 KB RAM, 48 MHz), maybe 5 years ago, and it was working fine back then. Unfortunately I don't seem to keep the sources anymore, but as far as I remember, it was relatively straightforward to port the code.

UweBonnes, 2021-03-06:
https://github.com/UweBonnes/blackmagic/tree/f3_if has support for local F3 and F072 variants. At least both platforms compile with changes after rebasing to master, but I do not have hardware to test at hand. CPU power help when reading flash, writing is mostly devices limited. RAM helps for tracing.  With builtin DFU bootloader we have 4 kiB more flash, so that is more than available on native.  B.t.w, what design rules do you intend to use on the WLCSP49? 0.4 mm pitch is tough!

geraldnilles, 2021-03-07:
That's amazing!  I'll start with this fork and let you know how it works after we build.

My goal is to embed this circuit into a device during development and then de-pop this debug MCU in production. We work with 0.35mm pitch parts so the 0.4mm wont be a problem. For the sake of fast turn around, i might spin a quick board based on the QFP part for development purposes. Assuming this works well, i'll publish a PR with the MCU's schematic and any FW changes.

Another goal of this project is to have multiple UART ports in addition to the SWD.  This particular MCU has 4 USARTs and i'd like to support all 4.  Based on your prior experiences, how much loading does the USB-CDC contribute to CPU/RAM usage?  We typically run at 115200 baud without any flow control. On paper, it shouldn't be too bad, but i'm not sure how heavy this CDC drivers are. I also dont expect to be maxing-out all 4 UARTs at the same time... it is mostly for monitoring boot logs.

UweBonnes, 2021-03-07:
You did consider PogoPins and such and then you decided to prefer an extra MCU? It will also to mor BOM hassles. Look with findchips for stock for the stm32f072cbu.

To support more UARTs, more USB Endpoints are needed. The st_usbfs has just enough for the BMP ( 2 * CDCACM(*3), DFU, Trace). usb_fx07 USB has to few EPs. Only USB HS, as on F723 Hs port has more endpoints.

It is good to put the files online, but I will only accept them for master when they show something new, not if they are "yap = yet another platform". The StlinkV3 e.g. is something new, high speed, much flash, more EP, etc. Otherwise I will probably tag the PR "$3 Board support" as more platform variants.

geraldnilles, 2021-03-07:
The ST reference manual says it supports "16 mono-directional or 8 bidirectional endpoints".  Are each of the three CDCACM endpoints bidirectional?

stoyan-shopov, 2021-03-07:
A usb endpoint is always unidirectional. A usb 'pipe', whatever that means, can be a collection of two usb endpoints, and therefore 'bidirectional'.

For a usb CDCACM interface, you need 2 usb endpoints (one IN, one OUT), and one IN endpoint for status reporting, as mandated by the standard. This status IN endpoint is a very unpleasant thing, as you need to have it, it does nothing, and it decreases your total number of usable hardware endpoints

I hope I am not too wrong in my understanding

UweBonnes, 2021-03-07:
Under features, there is "Configurable number of endpoints from 1 to 8" If this translates to 16 monodirectional EPs , we will be more than glad to find out. Maybe https://community.st.com/s/question/0D50X0000A7WRvG/which-stm32-supports-16-usb-endpoints-and-4-uartsusarts sheds some light on that issue.

esden, 2021-12-16:
Just to make sure. Are you still working on this @geraldnilles, if not I would close this issue.

geraldnilles, 2021-12-16:
i am not working on it.  will close.  thanks for everyone's help

closed 2021-12-16 by geraldnilles