blackmagic-debug/blackmagic #845: Hosted flash write resets the target

fabalthazar, 2021-03-24:
Hosted blackmagic resets the target after a flash write.
cl_utils.c, line 479:
`target_reset(t);`

In case of an applicative firmware writing to Flash right after a reset (e.g. initializing a ROM area), a subsequent verify with "blackmagic -V" fails.
One could expect the hosted BMP to leave the CPU halted after a flash write, leaving the choice to verify right after and then reset with "blackmagic -R".
That would be in line with the GDB behaviour: "load", "compare_sections" then "r" commands.

Unless there is a particular need for a reset there?

UweBonnes, 2021-03-25:
I have no problem with e.g. "blackmagic -S 0x1000 -r bla; blackmagic bla; blackmagic -V bla". So what is the target that you have problems. Otherwise I do not like typing, so I'd like to make a pure " blackmagic bla" make the CPU halt, flash the file and reset. If you want to keep the CPU under reset, consider adding options. Perhaps "blackmagic -W bla" for halt, flash, verify and reset e.g.

fabalthazar, 2021-03-25:
The sequence I use is `blackmagic bla; blackmagic -V bla;`.

Let me clarify and give an example.

By "applicative firmware writing Flash (ROM area)", I mean an area (>= 1 page(s)) in the programmable code area used as, for instance, a configuration area. No EEPROM nor OTP memory involved.
My problem appeared on an STM32G0 chip but can easily apply to STM32 range or other manufacturers.

Example:
I try to flash a firmware that is designed to use a page as configuration (let's say at 0x8001000).
0x8000000: code
0x8001000: 0x00 or 0xFF reserved for config
0x8001800: rest of code
The code is designed to initialize the configuration area (at 0x8001000) at startup.

Steps to reproduce my problem:

1. Flash binary firmware at 0x8000000 with blackmagic;
2. The chip resets -> the freshly programmed code starts to write at 0x8001000;
3. Verify with "blackmagic -V" fails with a mismatch at 0x8001000.

The example would just require a halt held between steps 1 and 3.

On reflection, it makes sense to reset the chip after a hosted write since by design, the subsequent blackmagic calls from the shell make process exit at each step, so detach (and resume!) the target. And resuming the target with a Program Counter at a location where a new code has just been written is awkward.
The comparison with a GDB sequence is inappropriate.

Your idea of an option is interesting.

Do your share my understanding?

UweBonnes, 2021-03-25:
Probably your program changes something starting at 0x08001000 quite after start. This happens between 2 and 3 . And steps 1 and 3 do not halt the CPU, they read out concurrent to the running CPU without  changing the CPU state beside stealing CPU cycles( If I understand things right)

fabalthazar, 2021-03-25:
That's it, my program changes after 0x8001000 between 2 and 3.
In that case, I haven't found a way to verify the program written with hosted bmp only.
We are limited to a not infinitely short spacing out between two process calls.

I trust you if you say the CPU is not halted, you wrote the code and you're surely more expert in Cortex-M than me.
Isn't it a bit weird to potentially change the code the CPU is going to fetch? Can it lead to an unpredictable state? Just a thought.

UweBonnes, 2021-03-25:
When we flash (write), we halt the CPU and we reset after flashing( writing), but when we only read we do that concurrent to the running CPU. Where do you see the problems?

fabalthazar, 2021-03-26:
Sorry to be meticulous. I try to detail things (sometimes simple) to be sure to understand well.
In your previous message you said "steps 1 and 3 do not halt the CPU". Step 1 is flashing/writing in my example. Now you say that "when we flash, we halt the CPU". I am a bit confused.

If I look at your code, in function `cl_execute()`, I see that the halt seems to be done via the `target_attach_n()` function call. Then the BMP_MODE_FLASH_WRITE, BMP_MODE_FLASH_READ or BMP_MODE_FLASH_VERIFY operations are dealt and finally `target_detach()` is called (I assume it resumes the target).

Should I understand that WRITE / READ / VERIFY are all done in halt state?
In such case there is no risk of unpredictable behaviour if the program is changed while the CPU is running.

fabalthazar, 2021-03-26:
Let's come back to the primary issue.

What do you thing of this proposal?

"blackmagic program.bin" unchanged: flash, reset. That's simple and what 90% of people want to do.
"blackmagic -w program.bin" is the same. Added "-w" option for explicit writing.
"blackmagic -w -V program.bin": Combined -w and -V options for writing and verifying immediately after, in the same blackmagic process execution. In other words: flash, verify, reset. It keeps things simple and does not break compatibility.

UweBonnes, 2021-03-26:
Sorry , I am talking nonsense. Read and verify happen halted, as reading on a simple LED blinking program shows.

UweBonnes, 2021-03-26:
The proposal sounds good. Will you make a PR?

Otherwise for the moment, verify  0x08000000 to 0x08001000 and above 0x08002000 in two steps.

fabalthazar, 2021-03-26:
I can try to make a PR if the architecture is friendly for this kind of combined operations. And if I feel I will not break everything!

UweBonnes, 2021-03-28:
I work on flashing the RP Pi. Here a similar same problem applies. If you flash non-valid firmware to (XIP) flash at 0x10000000 and reboot, boot  does not find a valid image and selects to boot to USB Mass Storage. Mass storage does not map XIP, and so verify  does not see the flash.

fabalthazar, 2021-03-31:
Perhaps #847 can fix this similar RP Pico issue?

UweBonnes, 2021-04-02:
> blackmagic -v1 -C -E gpio.bin
..
***  1      STM32F74x M7
> blackmagic -C -V gpio.bin
Verify failed at flash region 0x00200000
> blackmagic -C -w -V gpio.bin
...
Flash Write succeeded for 28564 bytes,   43.609 kiB/s
Verify failed at flash region 0x00200000
> blackmagic -C -V gpio.bin
...
Read/Verify succeeded for 28564 bytes,   44.912 kiB/s

It seems the combination -w -V combination does not clip the region to compare/

fabalthazar, 2021-04-02:
I am not good at F7 and particularly ITCM and AXIM flash interfaces but here is my analysis.

> blackmagic -v1 -C -E gpio.bin

Without start address given, the erase operates at 0x00200000, through the ITCM interface. This interface does not support write access, so I suppose the erase operation has no effect (?).

> ..
> ***  1      STM32F74x M7
> blackmagic -C -V gpio.bin
> Verify failed at flash region 0x00200000

gpio.bin does not exist in flash through ITCM. That's OK.

> blackmagic -C -w -V gpio.bin
> ...
> Flash Write succeeded for 28564 bytes,   43.609 kiB/s
> Verify failed at flash region 0x00200000

It is surprising that flash write emits no error when done through ITCM interface. But OK.
I'm not surprised the binary has not landed in the flash.

> blackmagic -C -V gpio.bin
> ...
> Read/Verify succeeded for 28564 bytes,   44.912 kiB/s

But here I am surprised that after a reset the program is there!

> 
> It seems the combination -w -V combination does not clip the region to compare.

Does `blackmagic -C -w -V -a 0x08000000 gpio.bin` succeeds? (after having done a mass erase or not)
B.t.w., does gpio.bin tries to modify the flash itself? I suppose not, looking at your results, but just to be sure.

(I certainly have to dive deeper in the manual to understand well the write operations and ITCM/AXIM interfaces)

UweBonnes, 2021-04-02:
Argh, I forgot about ITCM  and its special handling. F7 redirects write to ITCM to normal flash. 
>blackmagic -C -a 0x08000000 -w -V gpio.bin
works as expected!

PR #847 (2021-04-02): Bugfix 845 - hosted verify right after programming
Fixes #845 .

Allows verifying the written firmware right after programming, before the target is reset.
Typical usage: blackmagic -w -V firmware.bin

Particularly useful with a non cooperative firmware that changes the flash upon reset.

Tested with these combinations: -w -V, -wV, -V -w, -Vw.
Tested with BMP only (not STLink, CMSIS...).

To discuss and rework :)
Then squash.
closed 2021-04-02 by UweBonnes