blackmagic-debug/blackmagic #849: Cortex-M33 (Armv8-M) has more registers

UweBonnes, 2021-04-03:
As some core registers in V8 have a secure and non-secure incarnation, the announced register set needs to be updated.

esden, 2021-12-16:
What core registers? Can you add more references here so that we can track progress on this addition?

dragonmux, 2024-08-31:
On TrustZone enabled parts, there are banked secure and non-secure versions of all core registers, however.. because the secure ones are not accessible in non-secure mode and as a result GDB makes no provisions for extracting both, and only makes provisions for extracting the current MSP and PSP only (feature block `org.gnu.gdb.arm.m-system`), there is no change required for the core registers.

Support is required at some point in the future for the system block and MVE block (all of which are also secure/non-secure banked, but only one of the two reportable at any one time), however these are not ARMv8-specific as such and are general holes in the Cortex-M support.

This issue will get closed with the associated branch that we're working on to sort out the system block and fix ADIv6 AP based core register read-out and write-back.

closed 2024-09-04 by esden