blackmagic-debug/blackmagic #853: DFU-UTIL 1.0 needs more information in blackmagic_dfu.bin

kunosee, 2021-04-14:
Hello,

as I reported in another issue (https://github.com/blacksphere/blackmagic/issues/308#issuecomment-818842194). dfu-utils in version 1.0 require additional information about VID/PID in the dfu-binary, otherwise you can't upload via VID/PID as target address.
```
dfu-util.exe  -d 1d50:6018,:6017,0000:0000 -s 0x08002000 -D blackmagic.bin
dfu-util 0.10

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Warning: Invalid DFU suffix signature
A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1d50:6017
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 1024
Error: File ID 0000:0000 does not match device (1d50:6018 or 1d50:6017)
```
The current workaround is, to address the device with its serial (-S XXXXX) and to ommit the VID/PID address. When addressing this issue the problem about the invalid DFU suffix (see above) should also be investigated.

UweBonnes, 2021-04-14:
The preferred tool is scripts/stm32mem.py. Is there really a need for dfu-utils?

kunosee, 2021-04-14:
I have never worked with stm32mem.py, only with stm32flash.exe, maybe because most tutorials are using dfu-util. I will give it a try for sure. I see issues like this also as an kind of help to others, searching for error messages.  Even when such an issue gets closed, there is an explanation and workaround in search engines available.  I tried to figure out, where dfu-util looks for those magic bytes, but haven't found a solution right now.  If I have more time, I might dig into this deeper.  For now, I am happy as my tool-chain works, so its your decision what to do with this issue.

UweBonnes, 2021-04-14:
That is the pity with external tutorials. They get written once and never updated. They are not under the control of the project. Often the authors do not react any more on hints or remarks given to them.The only things under project control are the Wiki and file in the project like readme.md . Please try to follow what is there and if things miss or are wrong, point out here.

esden, 2021-12-16:
@kunosee to address your issue:
I think the mistake in your command is the additional `0000:0000` vid pid at the end of the `-d` parameter.

The way I usually run dfu-util is like this: `dfu-util -d 1d50:6018,:6017 -s 0x08002000:leave -D blackmagic-native.bin`. I think it is described this way in the [wiki](https://github.com/blacksphere/blackmagic/wiki/Upgrading-Firmware#dfu-util). 

Let me know if that solves the issue or not. It would be good to update the documentation for this. :)

esden, 2022-08-02:
As there was no further activity here in >6 months I am closing this issue.

closed 2022-08-02 by esden