blackmagic-debug/blackmagic #858: BMP remote protocoll and exception/timeout handling

UweBonnes, 2021-04-18:
Exception and timeout handling with BMP remote protocoll  needs to be improved.This is seen e.g. when scanning the ROMTABLE of a STM32WLE5. It is the same chip as the STM32WL55 with an M0 at AP1. AP1 is somehow disabled on theSTM32WLE5, but AP1 registers are available. Reading CIDR on the STM32WLE5 gives a timeout, the remote protocoll probably aborts early for its read timeout and the exception is not set.

esden, 2021-12-16:
Is that an enhancement or a bug? This issue is a little bit vague. I would like to be able to categorize it properly.

dragonmux, 2023-02-12:
This seems like we might have inadvertently fixed this with all the improvements to how we talk ADIv5, from implementing proper protocol recovery to fixing the error recovery logic and how we spin on `WAIT` replies from the device. We'll mark this as a v1.10 item that needs testing, with a view to optimistically closing this issue if we can prove it's been fixed.

closed 2023-04-03 by esden