blackmagic-debug/blackmagic #872: Mass erase on L433 fails

avrhack, 2021-05-03:
When I try to do a mon erase_mass on an STM32L443 the prompt returns immediately and nothing happens - the chip is not erased. The options bytes are set to locked, so my understanding is that a mass erase is the ONLY way to recover those to the default.

I'm running V1.7.1 of BMP.

Any thoughts as the only other alternative I have is to skip the MCU and replace with a new one which I obviously want to avoid.

UweBonnes, 2021-05-05:
I tested on a STM32L412 and could erase and reprogram. Try again, best with hosted, Start with "blackmagic -t" and then erase with "blackmagic -E -S 0x40000"

esden, 2021-12-16:
@avrhack there is another option too. It is possible the MCU is in fully locked mode, which can't be recovered from. Which one of the locked modes is set?

avrhack, 2021-12-27:
Sorry for the very delayed reply gents. As it was so cheap, to save any more wasted time and bought a genuine ST programmer, went to the dark side (ie Windows LOL) and use that to erase the STM. From there, the black magic probe worked perfectly so my suspicion is as Estin proposed which is there was a lot of mould bit somewhere which the probe was unable to clear for some reason

esden, 2021-12-27:
@avrhack thank you for the reply. This is good to know that there is some improvement potential regarding unlocking on our end. I will keep it in mind.

But as you resolved your immediate issue I will close this issue. Cheers! :)

closed 2021-12-27 by esden

avrhack, 2021-12-28:
> @avrhack thank you for the reply. This is good to know that there is some improvement potential regarding unlocking on our end. I will keep it in mind.
> 
> But as you resolved your immediate issue I will close this issue. Cheers! :)

Make sense - sorry I should have done so earlier. If it happens again presumably we can reopen for the audit trail and I'll try and do more investigation with your help? I'm thinking at the very least I should have used the STM official programmer to determine which lock bit were set......