blackmagic-debug/blackmagic #873: Blackmagic PC application error when accepting connection: 10022 Windows 10

pshri, 2021-05-05:
Please refer to #754
I am no longer able to use the PC application. Probably due to some windows update. Not sure exactly when it broke. Currently on Windows 10 Build 21370Rel 210424-1611. Did a rebuild of latest code. Same error.
Also disabled firewall and restarted computer. No go.

Getting the following error:

Using 0483:3752 0226AA525436666547143849 STMicroelectronics STM32 STLink
error when accepting connection: 10022

Did some research on error code. It means 
Invalid argument.
Some invalid argument was supplied (for example, specifying an invalid level to the setsockopt function). In some instances, it also refers to the current state of the socket—for instance, calling accept on a socket that is not listening.

https://docs.microsoft.com/en-us/windows/win32/winsock/windows-sockets-error-codes-2

pshri, 2021-05-08:
Did some more digging. Added return codes to  [gdb_if.c](https://github.com/blacksphere/blackmagic/files/6445449/gdb_if.c.txt). Bind was failing error [100013](https://docs.microsoft.com/en-us/windows/win32/winsock/windows-sockets-error-codes-2) (permission denied)

Hyper-V was reserving the ports. Explanation and solution in [What is Administered port exclusions in windows 10?](https://stackoverflow.com/questions/58216537/what-is-administered-port-exclusions-in-windows-10)

closed 2021-05-08 by pshri

UweBonnes, 2021-05-08:
If you could test #874 that would be fine.