blackmagic-debug/blackmagic #875: Flashing RP2040 via gdb gets stuck after erasing

9ary, 2021-05-09:
As per the title, upload gets stuck. `blackmagic <file>` works just fine, but no luck through gdb.
```gdb
$ arm-none-eabi-gdb target/thumbv6m-none-eabi/debug/examples/blinky
GNU gdb (GDB) 10.1
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from target/thumbv6m-none-eabi/debug/examples/blinky...
(gdb) target extended-remote :2000
Remote debugging using :2000
(gdb) mon swdp_scan
Target voltage: 3.29V
Available Targets:
No. Att Driver
 1      Raspberry RP2040 M0+
 2      Raspberry RP2040 M0+
 3      Raspberry RP2040 Rescue(Attach to reset!) 
(gdb) att 1
Attaching to program: /home/novenary/source/rp-hal/target/thumbv6m-none-eabi/debug/examples/blinky, Remote target
0x000020f0 in ?? ()
(gdb) load
Loading section .boot2, size 0x100 lma 0x10000000
(stuck)
```
<details><summary>blackmagic log (very long)</summary>
```
INFO: Open USB 8087:0029 class e0 failed
BMP hosted v1.7.1-195-gcfbe9f1-dirty
 for ST-Link V2/3, CMSIS_DAP, JLINK and LIBFTDI/MPSSE
Using 1d50:6018 BECBB1F7 Black Sphere Technologies
 Black Magic Probe (STLINK/V2) v1.7.1-195-gcfbe9f1
+#!GA#
       K(STLINK/V2) v1.7.1-195-gcfbe9f1
Remote is (STLINK/V2) v1.7.1-195-gcfbe9f1
Listening on TCP: 2000
Got connection
gdb_getpacket : qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+
gdb_putpacket : PacketSize=400;qXfer:memory-map:read+;qXfer:features:read+
gdb_getpacket : vMustReplyEmpty
*** Unsupported packet: vMustReplyEmpty
gdb_putpacket : 
gdb_getpacket : !
gdb_putpacket : OK
gdb_getpacket : Hg0
*** Unsupported packet: Hg0
gdb_putpacket : 
gdb_getpacket : qXfer:features:read:target.xml:0,3fb
gdb_putpacket : E01
gdb_getpacket : qTStatus
*** Unsupported packet: qTStatus
gdb_putpacket : 
gdb_getpacket : ?
gdb_putpacket : W00
gdb_getpacket : qRcmd,737764705f7363616e
 !GV#
       K2.93V
 !GV#
       K3.29V
gdb_putpacket : O54617267657420766f6c746167653a20332e3239560a
[... 4007 lines cut ...]
Write AP 0x01 DB1   : 0x00000011
!Hd00ff0118#
       K00200420
dp_read addr 20042000: 00000004
Read  AP 0x01 DB2   : 0x20042000
!HL0000011400000012#
       K00000000
Write AP 0x01 DB1   : 0x00000012
!Hd00ff0118#
       K00000000
dp_read addr 0000: 00000004
Read  AP 0x01 DB2   : 0x00000000
!HL0000011400000014#
       K00000000
Write AP 0x01 DB1   : 0x00000014
!Hd00ff0118#
       K00000000
dp_read addr 0000: 00000004
Read  AP 0x01 DB2   : 0x00000000

Erase done!
gdb_putpacket : OK
```

Attaching gdb directly to the probe rather than using the hosted gdb server yields me the same result.

9ary, 2021-05-11:
Your fix sounds reasonable. I'll give it a try later today, thanks!

9ary, 2021-05-11:
Works here, thanks!

```gdb
(gdb) load
Loading section .boot2, size 0x100 lma 0x10000000
Loading section .vector_table, size 0xc0 lma 0x10000100
Loading section .text, size 0xe34 lma 0x100001c0
Loading section .rodata, size 0x44c lma 0x10000ff4
Start address 0x100001cc, load size 5184
Transfer rate: 13 KB/sec, 648 bytes/write.
```

> Any reason to change the spinner? If there is a problem with the spinner, it should be fixed in other places too. And best one commit for the algo changes and one for the spinner.

Yep, agreed.

jamesturton, 2021-05-11:
When I was testing I noticed the following behaviour:
- Calling `tc_printf` with the spinner char from the command `monitor erase_mass` seemed to always work.
- Calling `tc_printf` from `load` command caused this hanging behaviour. I don't know so much on the inner-workings of GDB but it seemed that calling `tc_printf` during the load command caused the `vFlashWrite` packet to be ignored from gdb_main.

The addition of _flash_flush_cache improves stability when flashing and loading without resetting the RP2040.

> And best one commit for the algo changes and one for the spinner.

I will split the commit into two.

jamesturton, 2021-05-11:
> I'm not sure what the new spinnercounter variable is for though. Increasing the timeout value from 400 to 500 seems sufficient on my end.

Why is the spinner character being sent at all? My understanding is that it is to suppress a timeout warning from GDB not receiving any traffic for during long operations - is this correct?

If so, there is absolutely no need to be sending that much traffic to the host. The `spinnercounter` variable reduces the traffic by 500 times.

UweBonnes, 2021-05-11:
You are right. tc_printf may not be used in gdb. And the high rate of the spinner is also not good.

jamesturton, 2021-05-11:
I have updated my branch at:
https://github.com/jamesturton/blackmagic/tree/rp-updates

@UweBonnes Is there anything else I should do before submitting a PR?
Also, is there any more testing you would like me to do for the RP2040 target?

UweBonnes, 2021-05-14:
Please push the PR and do not take away maintainer write access. I have a small patch pending that disables the spinner when not called from a monitor command

PR #879 (2021-05-16): RP2040 fixes
Various fixes for the rp2040 target:
- Firstly fixing some casting errors in debug calls causing build errors for the native probe.
- Added `rp_flash_resume` method for added stability after erase or load commands.
- Added more debug logging for RP2040 target.
- Increased timeout for `flash_range_program` command when XOSC has not been enabled.
- Increased spinner timeout to 500ms (fixes #875)
closed 2021-05-16 by UweBonnes