blackmagic-debug/blackmagic #876: Is there a handbook or step by step tutorial?

Reneg973, 2021-05-10:
Hi,

I bought a BMP 2.1, connected it to PC, configured USB drivers. But first there is this red LED always on - what does it mean? I don`t find any descriptions about the LED codes.

UweBonnes, 2021-05-10:
(gdb) tar ext /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) mon ver
Black Magic Probe v1.7.1-120-g3457b235-dirty, Hardware Version 3
Copyright (C) 2015  Black Sphere Technologies Ltd.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

You mean really the read LED and not the orange? I see the read LED2 only
use in the DFU bootloader as "chaser light" as DFU indicator.

Reneg973, 2021-05-10:
when I run dfu-util like on wiki side `dfu-util.exe -d 1d50:6018,:6017 -s 0x08002000:leave -D blackmagic.bin`  I see the following:
Opening DFU capable USB device...
ID 1d50:6018
Run-time device DFU version 011a
Claiming USB DFU Runtime Interface...
Determining device status: state = appIDLE, status = 0
Device really in Runtime Mode, send DFU detach request...
Device will detach and reattach...
Opening DFU USB Device...
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 1024
Error: File ID 0000:0000 does not match device (1d50:6018 or 1d50:6017)

UweBonnes, 2021-05-10:
An older dfu-util will do. The preferred way however is "cd src'; make; ../scripts/stm32mem blackmagic.bin"

Reneg973, 2021-05-10:
> (gdb) tar ext /dev/ttyACM0
> Remote debugging using /dev/ttyACM0
> (gdb) mon ver
> Black Magic Probe v1.7.1-120-g3457b235-dirty, Hardware Version 3
> Copyright (C) 2015 Black Sphere Technologies Ltd. [...]

you are right, after seeing the "red-orange-green" ample blinking, I can confirm it is orange

Reneg973, 2021-05-10:
NICE, with older dfu-util it works

UweBonnes, 2021-05-10:
A patch to build for recent dfu-utils is welcome...
Otherweise I updated the Wiki to prefer stm32-mem

Reneg973, 2021-05-10:
Thanks for help, now I finally can use tar ext .

UweBonnes, 2021-05-10:
Please also have a look at blackmagic/hosted.

UweBonnes, 2021-05-10:
> ~/devel/dfu-util/src/dfu-util -s 0x08002000:leave -D blackmagic.bin 
dfu-util 0.10-dev
...
Downloading element to address = 0x08002000, size = 98436
Erase           [=========================] 100%        98436 bytes
Erase    done.
Download        [=========================] 100%        98436 bytes
Download done.
File downloaded successfully
Transitioning to dfuMANIFEST state

This still seems to work!

Reneg973, 2021-05-10:
yes, works now - thanks again

closed 2021-05-10 by Reneg973

UweBonnes, 2021-05-10:
Please be explicit: Did you update dfu-utils and it works now?

Reneg973, 2021-05-10:
running on windows, no, it just works with v0.9, but it is not dfu-util 0.10-dev.

Reneg973, 2021-05-10:
and calling make in src folder creates an erroneous header file version.h:
`"#define FIRMWARE_VERSION \"v1.7.1-195-gcfbe9f1\"" `

Means there are quotes around the define

UweBonnes, 2021-05-10:
It works the right way on Linux. Can you exchange in Makefile to
	$(Q)echo '#define FIRMWARE_VERSION "$(shell git describe --always --dirty --tags)"' > $@
and test on windows? Surrounding double quote is substituted to single quote and escaped double quote gets simple double quite.

Reneg973, 2021-05-10:
better now, but next issue:

"  CC      platforms/native/platform.c"
In file included from ../libopencm3/include/libopencm3/cm3/nvic.h:147,
                 from platforms/native/platform.c:33:
../libopencm3/include/libopencm3/dispatch/nvic.h:8:11: fatal error: libopencm3/stm32/f1/nvic.h: No such file or directory
    8 | # include <libopencm3/stm32/f1/nvic.h>
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

Reneg973, 2021-05-10:
at least I tried :)
>>
! was unexpected at this time.
make: *** [Makefile:15: all] Error 255

UweBonnes, 2021-05-10:
Probably libopencm3 compilation without mingw will not work on windows. Should I tar up and zip my compiled libopencm3 and send to you (32 MB)? Otherwise use mingw