blackmagic-debug/blackmagic #877: no rule to make target stdint.h needed by 'adiv5.o'

chriskuku, 2021-05-14:
I'm trying to do a 

make PROBE_HOST=hosted 

under macOS (11.3.1 - BigSur) and I'm getting:

```
$ make PROBE_HOST=hosted
 GIT include/version.h
make[1]: *** No rule to make target `/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.0/include/stdint.h', needed by `adiv5.o'.  Stop.
make: *** [all] Error 2
$ 

```

UweBonnes, 2021-05-14:
I suppose you started with  "make PROBE_HOST=hosted clean" to clean out junk from other platforms. I do not know or have MacOS, so some investigation on your side is welcome!

esden, 2021-12-16:
@chriskuku did you manage to get the build to work? The error is bit confusing as it seems to be using the xcode stdint header. What arm cross compile toolchain are you using? I have not tried using the xcode cross compiler just the official bare metal official ARM toolchain from here: https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

Let me know if it works for you. We might need to update the documentation how to build the firmware on MacOS. :)

esden, 2021-12-16:
Ohh I only now realize that this is for hosted. Considering that you got much further in the next issue that you opened #878 I assume you got this to work. So I am closing this issue.

closed 2021-12-16 by esden