blackmagic-debug/blackmagic #878: build errors under macOS with PROBE_HOST=hosted

chriskuku, 2021-05-14:
I'm getting: (with a fresh download as of today, BTW:where is the current version buried?):

```
$ cd blackmagic-master
$ ls
COPYING		Makefile	UsingSWO	libopencm3	src
HACKING		README.md	driver		scripts		upgrade
$ make PROBE_HOST=hosted
fatal: bad revision 'HEAD'
 GIT include/version.h
Package hidapi-libusb was not found in the pkg-config search path.
Perhaps you should add the directory containing `hidapi-libusb.pc'
to the PKG_CONFIG_PATH environment variable
No package 'hidapi-libusb' found
  CC      target/adiv5.c
Package hidapi-libusb was not found in the pkg-config search path.
Perhaps you should add the directory containing `hidapi-libusb.pc'
to the PKG_CONFIG_PATH environment variable
No package 'hidapi-libusb' found
  CC      target/adiv5_jtagdp.c
Package hidapi-libusb was not found in the pkg-config search path.
Perhaps you should add the directory containing `hidapi-libusb.pc'
to the PKG_CONFIG_PATH environment variable
No package 'hidapi-libusb' found
  CC      target/adiv5_swdp.c
Package hidapi-libusb was not found in the pkg-config search path.
Perhaps you should add the directory containing `hidapi-libusb.pc'
to the PKG_CONFIG_PATH environment variable
No package 'hidapi-libusb' found
  CC      command.c
command.c:41:16: error: redefinition of typedef 'cmd_handler' is a C11 feature [-Werror,-Wtypedef-redefinition]
typedef bool (*cmd_handler)(target *t, int argc, const char **argv);
               ^
./target/target_internal.h:53:16: note: previous definition is here
typedef bool (*cmd_handler)(target *t, int argc, const char **argv);
               ^
1 error generated.
make[1]: *** [command.o] Error 1
make: *** [all] Error 2
```

UweBonnes, 2021-05-14:
Building assumes to happen in the git directory "git clone https://github.com/blacksphere/blackmagic.git".  To have less dependancies, start with "make PROBE_HOST=hosted HOSTED_BMP_ONLY=1".  Care for the other platforms later.

What compiler is used? It seems to be more picky.

UweBonnes, 2021-05-14:
Test also with #880

chriskuku, 2021-05-14:
Thanks. It seems that I had some mess . In the old directory - where do I find the version? - of December 2020
I now was able to run a working build:

`$ make clean
  CLNHDR  include/libopencm3/stm32/f0/irq.json
  CLNHDR  include/libopencm3/stm32/f1/irq.json
  CLNHDR  include/libopencm3/stm32/f2/irq.json
  CLNHDR  include/libopencm3/stm32/f3/irq.json
  CLNHDR  include/libopencm3/stm32/f4/irq.json
  CLNHDR  include/libopencm3/stm32/f7/irq.json
  CLNHDR  include/libopencm3/stm32/l0/irq.json
  CLNHDR  include/libopencm3/stm32/l1/irq.json
  CLNHDR  include/libopencm3/stm32/l4/irq.json
  CLNHDR  include/libopencm3/stm32/g0/irq.json
  CLNHDR  include/libopencm3/stm32/g4/irq.json
  CLNHDR  include/libopencm3/stm32/h7/irq.json
  CLNHDR  include/libopencm3/gd32/f1x0/irq.json
  CLNHDR  include/libopencm3/lpc13xx/irq.json
  CLNHDR  include/libopencm3/lpc17xx/irq.json
  CLNHDR  include/libopencm3/lpc43xx/m4/irq.json
  CLNHDR  include/libopencm3/lpc43xx/m0/irq.json
  CLNHDR  include/libopencm3/lm3s/irq.json
  CLNHDR  include/libopencm3/msp432/e4/irq.json
  CLNHDR  include/libopencm3/efm32/tg/irq.json
  CLNHDR  include/libopencm3/efm32/g/irq.json
  CLNHDR  include/libopencm3/efm32/lg/irq.json
  CLNHDR  include/libopencm3/efm32/gg/irq.json
  CLNHDR  include/libopencm3/efm32/hg/irq.json
  CLNHDR  include/libopencm3/efm32/wg/irq.json
  CLNHDR  include/libopencm3/efm32/ezr32wg/irq.json
  CLNHDR  include/libopencm3/sam/3a/irq.json
  CLNHDR  include/libopencm3/sam/3n/irq.json
  CLNHDR  include/libopencm3/sam/3s/irq.json
  CLNHDR  include/libopencm3/sam/3u/irq.json
  CLNHDR  include/libopencm3/sam/3x/irq.json
  CLNHDR  include/libopencm3/sam/4l/irq.json
  CLNHDR  include/libopencm3/sam/d/irq.json
  CLNHDR  include/libopencm3/vf6xx/irq.json
  CLNHDR  include/libopencm3/swm050/irq.json
  CLNHDR  include/libopencm3/pac55xx/irq.json
  CLEAN   lib/stm32/f0
  CLEAN   lib/stm32/f1
  CLEAN   lib/stm32/f2
  CLEAN   lib/stm32/f3
  CLEAN   lib/stm32/f4
  CLEAN   lib/stm32/f7
  CLEAN   lib/stm32/l0
  CLEAN   lib/stm32/l1
  CLEAN   lib/stm32/l4
  CLEAN   lib/stm32/g0
  CLEAN   lib/stm32/g4
  CLEAN   lib/stm32/h7
  CLEAN   lib/gd32/f1x0
  CLEAN   lib/lpc13xx
  CLEAN   lib/lpc17xx
  CLEAN   lib/lpc43xx/m4
  CLEAN   lib/lpc43xx/m0
  CLEAN   lib/lm3s
  CLEAN   lib/lm4f
  CLEAN   lib/msp432/e4
  CLEAN   lib/efm32/tg
  CLEAN   lib/efm32/g
  CLEAN   lib/efm32/lg
  CLEAN   lib/efm32/gg
  CLEAN   lib/efm32/hg
  CLEAN   lib/efm32/wg
  CLEAN   lib/efm32/ezr32wg
  CLEAN   lib/sam/3a
  CLEAN   lib/sam/3n
  CLEAN   lib/sam/3s
  CLEAN   lib/sam/3u
  CLEAN   lib/sam/3x
  CLEAN   lib/sam/4l
  CLEAN   lib/sam/d
  CLEAN   lib/vf6xx
  CLEAN   lib/swm050
  CLEAN   lib/pac55xx
  CLEAN   doc
  CLEAN
$ make PROBE_HOST=hosted clean
  CLEAN
$ make PROBE_HOST=hosted HOSTED_BMP_ONLY=1
 GIT include/version.h
  CC  target/adiv5.c
  CC  target/adiv5_jtagdp.c
  CC  target/adiv5_swdp.c
  CC  command.c
  CC  target/cortexa.c
  CC  target/cortexm.c
  CC  crc32.c
  CC  target/efm32.c
  CC  exception.c
  CC  platforms/pc/gdb_if.c
  CC  gdb_main.c
  CC  gdb_hostio.c
  CC  gdb_packet.c
  CC  hex_utils.c
  CC  target/jtag_devs.c
  CC  target/jtag_scan.c
  CC  target/lmi.c
  CC  target/lpc_common.c
  CC  target/lpc11xx.c
  CC  target/lpc17xx.c
  CC  target/lpc15xx.c
  CC  target/lpc43xx.c
  CC  target/lpc546xx.c
  CC  target/kinetis.c
  CC  main.c
  CC  morse.c
  CC  target/msp432.c
  CC  target/nrf51.c
  CC  target/nxpke04.c
  CC  platforms/hosted/platform.c
  CC  remote.c
  CC  target/sam3x.c
  CC  target/sam4l.c
  CC  target/samd.c
  CC  target/samx5x.c
  CC  target/stm32f1.c
  CC  target/stm32f4.c
  CC  target/stm32h7.c
  CC  target/stm32l0.c
  CC  target/stm32l4.c
  CC  target/target.c
  CC  platforms/pc/serial_unix.c
  CC  timing.c
  CC  platforms/pc/cl_utils.c
  CC  platforms/pc/utils.c
  CC  platforms/hosted/bmp_remote.c
  CC  platforms/hosted/remote_swdptap.c
  CC  platforms/hosted/remote_jtagtap.c
  CC  platforms/hosted/bmp_serial.c
  LD      blackmagic
$ 
`

chriskuku, 2021-05-14:
> Building assumes to happen in the git directory "git clone https://github.com/blacksphere/blackmagic.git". To have less dependancies, start with "make PROBE_HOST=hosted HOSTED_BMP_ONLY=1". Care for the other platforms later.
> 
> What compiler is used? It seems to be more picky.

I downloaded the .zip and unpacked. blackmagic-master was created. I'm now git cloning and a directory blackmagic is created.

I'm wondering why "make clean" says: 
$ make PROBE_HOST=hosted HOSTED_BMP_ONLY=1 clean
  CLEAN
$ make clean
make[1]: *** No rule to make target `clean'.  Stop.
make: *** [clean] Error 2

And I'm getting now:

```
$ make PROBE_HOST=hosted HOSTED_BMP_ONLY=1
 GIT include/version.h
  CC      target/adiv5.c
  CC      target/adiv5_jtagdp.c
  CC      target/adiv5_swdp.c
  CC      command.c
  CC      target/cortexa.c
  CC      target/cortexm.c
  CC      crc32.c
  CC      target/efm32.c
  CC      exception.c
  CC      platforms/pc/gdb_if.c
  CC      gdb_main.c
  CC      gdb_hostio.c
  CC      gdb_packet.c
  CC      hex_utils.c
  CC      target/jtag_devs.c
  CC      target/jtag_scan.c
  CC      target/lmi.c
  CC      target/lpc_common.c
  CC      target/lpc11xx.c
  CC      target/lpc17xx.c
  CC      target/lpc15xx.c
  CC      target/lpc43xx.c
  CC      target/lpc546xx.c
  CC      target/kinetis.c
  CC      main.c
  CC      morse.c
  CC      target/msp432.c
  CC      target/nrf51.c
  CC      target/nxpke04.c
  CC      platforms/hosted/platform.c
In file included from platforms/hosted/platform.c:38:
platforms/hosted/cmsis_dap.h:47:22: error: unknown type name 'swd_proc_t'
int dap_swdptap_init(swd_proc_t *swd_proc) {return -1;}
                     ^
platforms/hosted/cmsis_dap.h:50:1: error: expected identifier or '('
## pragma GCC diagnostic pop
^
2 errors generated.
make[1]: *** [platform.o] Error 1
make: *** [all] Error 2
$ 
```

chriskuku, 2021-05-14:
FWIW:
```

$ cc -v
Apple clang version 12.0.5 (clang-1205.0.22.9)
Target: x86_64-apple-darwin20.4.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

chriskuku, 2021-05-14:
I find that there are some files in platforms are 0 bytes. 
The file cmsis_dap.h has an error in line 50:

# pragma GCC diagnostic pop

it should read (one # only)

chriskuku, 2021-05-15:
Datapoint: I tried to compile a freshly git cloned repo under Ubuntu 20.04.1 (x86_64) and the issue with swd_proc_t is persistent.
Not a macOS issue.

UweBonnes, 2021-05-15:
The double typedef should be gone with #881

chriskuku, 2021-05-24:
I did a
 `make PROBE_HOSTED=hosted HOSTED_BMP_ONLY=1 `
today and compilation went smooth until the end, when make spit out :

```
  CC      timing.c
  CC      platforms/stm32/timing_stm32.c
  CC      platforms/common/swdptap.c
  CC      platforms/common/jtagtap.c
  LD      blackmagic.elf
adiv5.o: file not recognized: file format not recognized
collect2: error: ld returned 1 exit status
make[1]: *** [blackmagic.elf] Error 1
make: *** [all] Error 2
$ 
```

After a 
`make PROBE_HOSTED=hosted clean`

compilation went smooth.

Thanks for fixing.

chriskuku, 2021-05-25:
When building in the top-level directory 

`make PROBE_HOST=hosted`

I'm getting 
```
$ cd src
$ make PROBE_HOST=hosted clean
  CLEAN
$ make PROBE_HOST=hosted
 GIT include/version.h
Package hidapi-libusb was not found in the pkg-config search path.
Perhaps you should add the directory containing `hidapi-libusb.pc'
to the PKG_CONFIG_PATH environment variable
No package 'hidapi-libusb' found
  CC      target/adiv5.c
```

UweBonnes, 2021-05-25:
How did you try to fix the problem "No package 'hidapi-libusb' found"? 
Compiling hosted in the base directory skips compiling libopencm3 needed for firmware  BMP, so no real difference.

chriskuku, 2021-05-25:
> How did you try to fix the problem "No package 'hidapi-libusb' found"?
> Compiling hosted in the base directory skips compiling libopencm3 needed for firmware BMP, so no real difference.

By ignoring :) A binary (blackmagic) got being produced anyway in the end. I was just wondering whether there was a way to get rid of these messages or if it would add some functionality when that lib is present.

rbsexton, 2021-05-31:
I found the same issue when trying to build with make PROBE_HOST=hosted and the hidapi library from home-brew on my Mac.  

The issue seems to be that the hidapi library from homebrew is called "hidapi" and not "hidapi-libusb"

I tested this by making this change to Makefile.inc:

```--- a/src/platforms/hosted/Makefile.inc
+++ b/src/platforms/hosted/Makefile.inc
@@ -51,8 +51,8 @@ ifneq ($(HOSTED_BMP_ONLY), 1)
  ifneq (, $(findstring mingw, $(SYS)))
   SRC += hid.c
  else
-  CFLAGS += $(shell pkg-config --cflags hidapi-libusb)
-  LDFLAGS += $(shell pkg-config --libs hidapi-libusb)
+  CFLAGS += $(shell pkg-config --cflags hidapi)
+  LDFLAGS += $(shell pkg-config --libs hidapi)
  endif
 endif
```

Now I'm getting clean compiles and black magic attempts (unsuccessfully) to connect to my ST-Link probe.

UweBonnes, 2021-05-31:
Please put that in some if endif so that Linux is not touched.Make a PR and if it compiles for me, I will apply.

esden, 2021-12-16:
Just to make sure. Is the only thing needed here to add a macOS specific switch to the Makefile.inc?

esden, 2022-03-30:
Hey everyone. So I took a look into this issue. It turns out that it was already fixed, quite a while ago, with this patch 1b49823f646919c80aa67057e37486fc69164268 but the issue did not get closed for some reason... :/

Thank you for reporting the issue @rbsexton I have tested bmp pc hosted build on Mac OS with homebrew as library source and it seems to be working for me. If there some other issues on Mac OS please feel free to open a new issue for that.

closed 2022-03-30 by esden

werediver, 2022-08-12:
To add to the pool of solutions: "hidapi" installed via Homebrew may not get found without specifying search paths.

```
CFLAGS="-I${HOMEBREW_PREFIX}/include" LDFLAGS="-L${HOMEBREW_PREFIX}/lib" make PROBE_HOST=hosted
```