blackmagic-debug/blackmagic #884: unable to load a file on LPC802 board over SWD

0x72D0, 2021-05-17:
I tried the following on the LPCXpresso 802 OM40000 Rev A dev board from NXP. For the BlackMagicProbe, I use a Black Magic Probe Mini V2.1 with the native binary from the latest automated build:

[image]

Has you can see, when I try to load the program, the loading failed. The board use a LPC802M001JDH20 and I know that SWD flashing on LPC is a little bit tricky ( https://github.com/blacksphere/blackmagic/wiki/Adding-target-drivers ) but I can see that this chip is supposed to be supported in lpc11xx.c. Here's my setup:

[image]

I also have the jumper to enable SWD:

[image]

the schematics for the board can be found here: https://www.nxp.com/products/processors-and-microcontrollers/arm-microcontrollers/general-purpose-mcus/lpc800-cortex-m0-plus-/lpcxpresso802-for-the-lpc802-family-of-mcus:OM40000

UweBonnes, 2021-05-17:
Untils some point in time, wild probing for devices was done, with the danger of hangup. Now a more educated guess is made and the guess did not know about the LPC8(0)? I just added  probing for  lpc11 when a M0+ is detected. I just pushed a fix as https://github.com/UweBonnes/blackmagic/tree/fixes and will soon make it #885. Can you test?

0x72D0, 2021-05-17:
Thanks for your quick response, I tried the fix you sent. It did succeed to correctly detect the chip this time, but unfortunately the chip still fail to load new code:

[image]

0x72D0, 2021-05-17:
Thanks for your quick response, I tried the fix you sent. The fix seem to succeed, but now I have a new error message while trying to load the code:

[image]

UweBonnes, 2021-05-17:
You must first attach before flashing "load"

0x72D0, 2021-05-17:
yeah I forgot to load the first time, but in when I attach to my device I still receive the following error:

[image]

UweBonnes, 2021-05-17:
Does the LPC80x have flash protection? Is it set? Can try to flash in some other way?

0x72D0, 2021-05-17:
I succeed to upload it by removing the ext dbg jumper and by uploading my code in USB with MCUxpresso.

UweBonnes, 2021-05-18:
If it is a OM4000 board, it has a built-in CMSIS-DAP. BMP/hosted should recognize the CMSIS-DAP and it should be useable with pyocd and OpenOcd too.

0x72D0, 2021-05-18:
To test with OpenOcd I use the same bmp with the same configuration from when I tried to flash using the SWD interface. I've mounted my bmp in an Ubuntu VM as ttyACM0.

It's my first time using OpenOcd, so I don't know if I did it wrong, but it give me the following results:

[image]

UweBonnes, 2021-05-18:
OpenOCD does not handle the BMP dongle.. Unplug the BMP dongle and switch to the built-in CMSIS DAP adapter. Try again with the OpenOCD test above, Try also PyOCD. And BMP/hosted "make PROBE_HOST=hosted", can handle CMSIS DAP too. Compile with "make PROBE_HOST=hosted"

0x72D0, 2021-05-18:
I still have the error with OpenOCD while the hosted blackmagic is running (I've disconnected the dongle and I've removed the ext dbg jumper this time):

[image]

I'm pretty sure the board is detected as ttyACM0, dmesg give me the following output:

[image]

I'm gonna try to test with pyOCD soon.

UweBonnes, 2021-05-18:
blackmagic -t should find the device and blackmagic -h to learn abouy the verbosity levels.

0x72D0, 2021-05-18:
blackmagic -t -v 31 give me the following output:

```INFO: Open USB 80ee:0021 class  0 failed
BMP hosted v1.6.1-790-gb1325be
 for ST-Link V2/3, CMSIS_DAP, JLINK and LIBFTDI/MPSSE
Using 1fc9:0132 09028035 NXP Semiconductors
 LPC11U3x CMSIS-DAP v1.0.4
Blacklist
cmd :   00.03.
cmd res:03.00.
cmd :   00.00.04.
cmd res:00.05.31.2e.31.30.00.00.00.00.00.00.00.00.00.00.
Ver 1.10, cmd :   00.00.f0.
cmd res:00.01.11.00.00.00.00.00.00.00.00.00.00.00.00.00.
Cap (0x11): SWD, Atomic Cmds
Running in Test Mode
Target voltage: (null) Volt
cmd :   00.11.00.09.3d.00.
cmd res:11.00.00.00.00.00.
Speed set to  4.0000 MHz for SWD
cmd :   00.04.02.80.00.80.00.
cmd res:04.00.00.00.00.00.00.
cmd :   00.13.00.
cmd res:13.00.00.
cmd :   00.02.01.
cmd res:02.01.00.
cmd :   00.01.00.01.
cmd res:01.00.00.00.
cmd :   00.12.88.ff.ff.ff.ff.ff.ff.ff.9e.e7.ff.ff.ff.ff.ff.ff.ff.00.
cmd res:12.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.
cmd :   00.05.00.01.02.
cmd res:05.01.01.77.14.c1.0b.00.00.00.00.00.00.00.00.00.
cmd :   00.12.20.ff.ff.ff.ff.
cmd res:12.00.
cmd :   00.12.20.ff.ff.ff.ff.
cmd res:12.00.
cmd :   00.12.20.92.f3.09.62.
cmd res:12.00.
cmd :   00.12.20.95.2d.85.86.
cmd res:12.00.
cmd :   00.12.20.e9.af.dd.e3.
cmd res:12.00.
cmd :   00.12.20.a2.0e.bc.19.
cmd res:12.00.
cmd :   00.12.0c.a0.01.00.00.
cmd res:12.00.
cmd :   00.12.20.ff.ff.ff.ff.
cmd res:12.00.
cmd :   00.12.20.ff.ff.ff.0f.
cmd res:12.00.
cmd :   00.05.00.01.02.
cmd res:05.01.01.77.14.c1.0b.00.00.
	dap_read_reg 00 0bc11477
DPIDR 0x0bc11477 (v1 MINDP rev0)
cmd :   00.05.00.01.06.
cmd res:05.01.01.40.00.00.f0.00.00.
	dap_read_reg 04 f0000040
dp_read 0004 f0000040
Read  CTRL/STAT: 0xf0000040
Write CTRL/STAT: 0xf0000040
	dap_write_reg 04 f0000040
cmd :   00.05.00.01.04.40.00.00.f0.
cmd res:05.01.01.00.00.00.00.00.00.
cmd :   00.05.00.01.06.
cmd res:05.01.01.40.00.00.f0.00.00.
	dap_read_reg 04 f0000040
dp_read 0004 f0000040
Read  CTRL/STAT: 0xf0000040
Write CTRL/STAT: 0xf4000040
	dap_write_reg 04 f4000040
[... 695 lines cut ...]
cmd :   00.05.00.04.08.00.00.00.00.01.52.00.00.03.05.28.10.00.e0.0d.00.00.00.00.
cmd res:05.04.01.00.00.00.00.00.00.00.00.00.00.00.00.00.
ap_mem_write_sized @ e0001038 len 4, align 4: 00 00 00 00
memwrite @ e0001038 len 4, align 2 , 00000000 start: 
cmd :   00.05.00.04.08.00.00.00.00.01.52.00.00.03.05.38.10.00.e0.0d.00.00.00.00.
cmd res:05.04.01.00.00.00.00.00.00.00.00.00.00.00.00.00.
ap_mem_write_sized @ e0001048 len 4, align 4: 00 00 00 00
memwrite @ e0001048 len 4, align 2 , 00000000 start: 
cmd :   00.05.00.04.08.00.00.00.00.01.52.00.00.03.05.48.10.00.e0.0d.00.00.00.00.
cmd res:05.04.01.00.00.00.00.00.00.00.00.00.00.00.00.00.
ap_mem_write_sized @ e0001058 len 4, align 4: 00 00 00 00
memwrite @ e0001058 len 4, align 2 , 00000000 start: 
cmd :   00.05.00.04.08.00.00.00.00.01.52.00.00.03.05.58.10.00.e0.0d.00.00.00.00.
cmd res:05.04.01.00.00.00.00.00.00.00.00.00.00.00.00.00.
ap_mem_write_sized @ e000edfc len 4, align 4: 00 00 00 01
memwrite @ e000edfc len 4, align 2 , 01000000 start: 
cmd :   00.05.00.04.08.00.00.00.00.01.52.00.00.03.05.fc.ed.00.e0.0d.00.00.00.01.
cmd res:05.04.01.00.00.00.00.00.00.00.00.00.00.00.00.00.
ap_mem_write_sized @ e000edf0 len 4, align 4: 00 00 5f a0
memwrite @ e000edf0 len 4, align 2 , a05f0000 start: 
cmd :   00.05.00.04.08.00.00.00.00.01.52.00.00.03.05.f0.ed.00.e0.0d.00.00.5f.a0.
cmd res:05.04.01.00.00.00.00.00.00.00.00.00.00.00.00.00.
cmd :   00.03.
cmd res:03.00.
```

UweBonnes, 2021-05-18:
-v31 is very noisy. "-t" impies "-v3" and should detect the device. "blachmagic <file.bin>" would try to flash the binary at 0. Or start the gdb server only "blackmagic" and cionnect gdb to "tar ext :2000"

UweBonnes, 2021-05-18:
I will get a OM40000UL soon...

0x72D0, 2021-05-18:
Ok I think I understand better, so I ran blackmagic -v 3 and try to connect to it with gdb-multiarch, it succeed to detect the LPC802 correctly, but the loading still fail with the same error:

[image]

the hosted blackmagic output is the following:

[image]

I'm gonna try with pyOCD soon

0x72D0, 2021-05-18:
Same problem on pyOCD:

[image]

0x72D0, 2021-05-18:
with pyOCD I still have a failure on the loading:

[image]

UweBonnes, 2021-05-18:
So there is something more to understand that neither pyocd and bmp knows yet. What gdb server does MCUxpresso use?

0x72D0, 2021-05-18:
The one I use to successfully upload my code on the OM40000 dosen't seem to be a gdb server, it's called the LinkServer Debugger.

0x72D0, 2021-05-18:
Did someone succeed to upload on a LPC802 chip before( without being on the OM40000 )? I'm more trying to see if I can upload on a similar chip before the real board with an onboard Jtag interface arrive.

UweBonnes, 2021-05-18:
#835

0x72D0, 2021-05-19:
I just succeed flashing the chip via SWD using the LPC-LINK 2 and the LinkServer on MCUxpresso.

UweBonnes, 2021-05-19:
I can read the flash content of the OM40000 board to a bin file. Trying to reprogram (erase, flash) with BMP/CMSIS only erases and than steps into error. Flash is erase. BMP/BMP_HOSTED hangs in the last chunk. However repowering bring up the board and compare succeeds.

esden, 2021-12-16:
Considering that we have a report that it was possible to program that host board and the target. I am assuming this issue is resolved, so I am closing it. Thank you everyone. :) 

If there is still a problem here that needs to be addressed please feel free to reopen this issue.

closed 2021-12-16 by esden