blackmagic-debug/blackmagic #888: Working with blackmagic_dfu source

Manfred-O, 2021-05-24:
Hi, i need to make some changes to the blackmagic_dfu.bin to suit better my needs. One thing is a like to have a possibility to switch to dfu mode from inside my application i am not able to switch the boot pin unfortunatley to get there continuously. I could do that by just software reseting and i am back in dfu mode, leds starts blinking again, but that did not last long and it's back in application mode. I cloned the git repo and i can not see a specific folder of the blackmagic_dfu project. Can anyone point me to the relevant folders and files?  

Thanks in advance

UweBonnes, 2021-05-24:
blackmagic_dfu  builds together with the firmware. "dfu-utils -e " or some usb-connection to the dfu endpoint (see scripts/*) brings you to the dfu loader. DFU / blinking LED stays active until something activly switches back to application..

Manfred-O, 2021-05-25:
Hi, with firmware you mean all 3 binarys will be created? But i just need the blackmagic_dfu.bin and need to adapt it a little bit. I just flashed the blackmagic_dfu.bin to my board. My application does not implemment a dfu it will just communicate with USB. And with in my app i just like to software reset the board and it will come up in dfu mode which starts at 0x08000000 and stays there until update is done. 

Does the relevant files are inside folder /src/platforms/stm32 and /src/platforms/swlink?   
The starting point "main()" is inside "usbdfu.c"?

Thanks

UweBonnes, 2021-05-25:
DFU needs a trigger from the main program to restart the main programm with a "start dfu" flag set  and in the main program evaluation if the flag is set and so to continue to run  the DFU programm or otherwise to switch  to the normal program. BMP sets some magic in some unused GPIO and does a core reset. On startup DFU looks for that magic. A better way is to write some magic to uninitialized RAM and do a real reset. ST/StlinkV21 evaluates the Reset reason as another way.

Manfred-O, 2021-05-25:
Yes i had something similar in my mind like placing some magic numbers inside backup register of RTC or flash does not need to be updated that frequently, hope so. Where is the starting point of blackmagic_dfu.bin so i can walk through and look where i can place my additional lines. Can i let *_dfu.bin be in dfu mode for longer via a specified timeout and just increase the timeout that would be the easiest way?

UweBonnes, 2021-11-11:
Is your problem still pending? Look how the different platforms trigger DFU entry. For native, the  GPIO Register methode is done for legacy reasons. Some magic number and a clean reset is better. Please revisit the issue and discuss more or close.

closed 2021-11-11 by Manfred-O