blackmagic-debug/blackmagic #889: Implicit function declaration

JDuchniewicz, 2021-05-25:
I am trying to build bmp with target `PROBE_HOSH=swlink` and I am getting these errors:

```
target/efm32.c: In function ‘efm32_probe’:
target/efm32.c:645:2: error: implicit declaration of function ‘sniprintf’ [-Werror=implicit-function-declaration]
  snprintf(priv_storage->efm32_variant_string,
  ^
target/efm32.c: In function ‘efm32_aap_probe’:
target/efm32.c:1018:2: error: implicit declaration of function ‘siprintf’ [-Werror=implicit-function-declaration]
  sprintf(priv_storage->aap_driver_string,
  ^
cc1: all warnings being treated as errors
make[1]: *** [Makefile:99: efm32.o] Error 1
make: *** [Makefile:17: all] Error 2
```

I installed `arm-none-eabi-{gcc, newlib}`.

UweBonnes, 2021-05-25:
Check your installation of newlib/nanolib! Who does provide newlib on your system? I use https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

JDuchniewicz, 2021-05-27:
I installed a package from AUR (Arch Linux, usually it should just work). Maybe the newlib differs (is newer probably) from the ones tested usually on.

UweBonnes, 2021-05-27:
Try to compile with the whole block of sniprintf and friends defines removed. For me, a test resulted in a binary  20 byte larger.

esden, 2021-12-16:
@JDuchniewicz were you able to compile the firmware? If not I can check how my setup looks like on arch and update the documentation.

JDuchniewicz, 2021-12-16:
I settled for the pre-built binary if I correctly remember.

esden, 2021-12-16:
Ok, that sounds good. I will close this issue then. :)

closed 2021-12-16 by esden

JDuchniewicz, 2021-12-16:
Hopefully it won't bite back in the future (maybe it was just due to toolchain misconfiguration or exotic hardware setup 🤷🏽)