blackmagic-debug/blackmagic #890: Request SAMC21 support

ikcalB, 2021-05-28:
I did buy the BMP to support this awesome project, just realising that my required platform is not yet supported (SAMC21).

Has there been any work on this so far?

If not, I might try to add support.
Skimming through the sources I suppose there is more, than just replacing 's/samd/samc/'.
Is there a starting point for porting?

Best regards from Austria,
Florian

UweBonnes, 2021-05-28:
I have not much experience with SAM, STM32 is much better known to me. For a new platform, find out if there is familiarity with already existing platforms. The entry page of the NVMCTRL section of the SAMD2 and SAMC2 family datasheets seems very similar. Work out the differences and test. Eventually spy at OpenOCD and pyOcd code for those devices.

When implementing target algorithms, I perfer using BMP hosted as no constant firmware upload is needed. 

Consider chatting in the discord channel

franzflasch, 2023-08-02:
Any news on this? From what I see openocd does not seem to treat the SAMC and SAMD devices differently. I guess it should be only some minor adaption to target/samd.c to make it work.

perigoso, 2023-08-02:
*(I'm not an official maintainer)* From what I looked at the time adding support for the SAMC should be trivial, but without personal interest or a device to test on I didn't Implement it myself, I don't think anyone else did and looking at the source seems to confirm this.

dragonmux, 2023-08-02:
There is support in development - we need to encourage the user in question to contribute it upstream to this repo with a PR. SAMC has subtly a different Flash controller and different identification information so the support is actually a fork of the SAMD support right now.

Failing that, we'll hopefully get to implementing support in the v2.0 cycle

franzflasch, 2023-08-02:
Is that code available somewhere? I have a samc21 here and i could test it.

dragonmux, 2023-08-02:
Unfortunately not yet publicly - we'll make sure this issue ticket gets roped into whatever eventual PR when it is made public so you'll be able to know about it though.

franzflasch, 2023-08-02:
In case someone is interested: As a quick and dirty hack I changed the SAMD_DID_CONST_VALUE in target/samd.c:
```
-#define SAMD_DID_CONST_VALUE   0x10000000U
+#define SAMD_DID_CONST_VALUE   0x11000000U
```

Not sure what the SAMD_DID_CONST_VALUE is supposed to be anyway. 

Now I am able to flash the SAMC21.

PR #1659 (2023-10-30): Feature: samd: Implemented support for the SAMC21 series

## Detailed description

This PR adds a new target: Atmel SAMC21.

- The PR was tested on a  Atmel SAMC21N18A (rev E) M0+, with parameters:

	- did: 0x11011420
	- family: 2
	- series: 1
	- devsel: 32

- SAMD_DID_MASK was changed from 0xff380000U to 0xfe380000U.
This allows SAMC21 parts where DID begins with 0x11 to be accepted.

- SAMD_DID_FAMILY_MASK was changed from 0x3fU to 0x1fU.
According to the datasheet, the field "family" in the DID Device Identification Register is a 5 bit field, so the mask ought to be 0x1f, not 0x3f.

- sampl [... cut]
closed 2023-10-30 by dragonmux