blackmagic-debug/blackmagic #891: Cannot unlock cks32f103c8t6

turtiustrek, 2021-05-29:
BMP flashed on a bluepill (host) connected to cks32f103c8t6(target) cannot unlock read/write protection

Probe information:
```
(gdb) monitor
Black Magic Probe(STLINK/V2) v1.7.1-212-g6454d0b, Hardware Version 0
Copyright (C) 2015  Black Sphere Technologies Ltd.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

(gdb) 

```
SWD scan:
```
(gdb) monitor s
Target voltage: 2.95V
Available Targets:
No. Att Driver
 1      STM32F1 (clone) medium density M3
(gdb) 
```
Trying to erase & flash blackmagic on target:
```
(gdb) monitor option erase
0x1FFFF800: 0x5AA5
0x1FFFF802: 0xFFFF
0x1FFFF804: 0xFFFF
0x1FFFF806: 0xFFFF
0x1FFFF808: 0xFFFF
0x1FFFF80A: 0xFFFF
0x1FFFF80C: 0xFFFF
0x1FFFF80E: 0xFFFF
(gdb) load /home/apple/Programming/blackmagic/src/blackmagic.elf 
Loading section .text, size 0x195b8 lma 0x8002000
Error writing data to flash
(gdb) 
```

UweBonnes, 2021-05-29:
Reset "r" or un-and replug. Option setting need to be reloaded.

turtiustrek, 2021-05-29:
Thanks for the  response, i tried both methods and still cannot write to the flash.
Here is the log.
The first attach("att 1") is showing a  replug and  then from there I preformed the flash which was a fail so then using the reset "r" also had similar results. 

```
(gdb) monitor s
Target voltage: 2.96V
Available Targets:
No. Att Driver
 1      STM32F1 (clone) medium density M3
(gdb) att 1   
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0xf0013096 in ?? ()
(gdb) load /home/apple/Programming/blackmagic/src/blackmagic.elf 
Loading section .text, size 0x195b8 lma 0x8002000
Error writing data to flash
(gdb) monitor option erase
0x1FFFF800: 0x5AA5
0x1FFFF802: 0xFFFF
0x1FFFF804: 0xFFFF
0x1FFFF806: 0xFFFF
0x1FFFF808: 0xFFFF
0x1FFFF80A: 0xFFFF
0x1FFFF80C: 0xFFFF
0x1FFFF80E: 0xFFFF
(gdb) r
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program:  
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.

Program received signal SIGSEGV, Segmentation fault.
0x00000000 in ?? ()
(gdb) monitor s
Target voltage: 3.25V
Available Targets:
No. Att Driver
 1      STM32F1 (clone) medium density M3
(gdb) att 1
A program is being debugged already.  Kill it? (y or n) y
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.

Thread 1 stopped.
0xf0013096 in ?? ()
(gdb) load /home/apple/Programming/blackmagic/src/blackmagic.elf
Loading section .text, size 0x195b8 lma 0x8002000
Error writing data to flash
(gdb) 
```

UweBonnes, 2021-05-29:
Why does the program load at 0x08002000? It seems you want to flash blackmagic. You missed to flash `!the` bootloader!

turtiustrek, 2021-05-29:
Yes, trying to flash blackmagic. same issue with flashing the bootloader
```
(gdb) load /home/apple/Programming/blackmagic/src/blackmagic_dfu.elf
Loading section .text, size 0x1e48 lma 0x8000000
Error writing data to flash
(gdb) load /home/apple/Programming/blackmagic/src/blackmagic_dfu.hex 0x8000000
Loading section .sec1, size 0x1ea0 lma 0x10000000
Load failed
(gdb) restore /home/apple/Programming/blackmagic/src/blackmagic_dfu.hex 0x8000000
Restoring section .sec1 (0x8000000 to 0x8001ea0) into memory (0x10000000 to 0x10001ea0)
warning: restore: memory write failed (Unknown error -1).
(gdb) 
```

UweBonnes, 2021-05-29:
You upset the core when you booted with no bootloader. In some cases only POR reset getst thing back. First try should be connect iunder reset, so that the core does not run into a trap.

turtiustrek, 2021-05-29:
If i understood correctly, you're saying to pull NRST pin down **and** then connect from the host?

UweBonnes, 2021-05-29:
If reset is connected, use "mon(itor) con(nect_srst) ena(ble). Otherwise try first to boot chip internal bootloader (BOOT0 = 1). As last resort, fiddle with reste manual.

turtiustrek, 2021-05-29:
Yes, that was indeed the key. However right after flashing the DFU, the actual binary does not flash.
```
(gdb) load /home/apple/Programming/blackmagic/src/blackmagic_dfu.elf 
Loading section .text, size 0x1e48 lma 0x8000000
Loading section .data, size 0x58 lma 0x8001e48
Start address 0x80015a4, load size 7840
Transfer rate: 24 KB/sec, 871 bytes/write.
(gdb) load /home/apple/Programming/blackmagic/src/blackmagic.elf 
Loading section .text, size 0x195b8 lma 0x8002000
Error writing data to flash
(gdb) monitor option 
usage: monitor option erase
usage: monitor option <addr> <value>
0x1FFFF800: 0x5AA5
0x1FFFF802: 0xFFFF
0x1FFFF804: 0xFFFF
0x1FFFF806: 0xFFFF
0x1FFFF808: 0xFFFF
0x1FFFF80A: 0xFFFF
0x1FFFF80C: 0xFFFF
0x1FFFF80E: 0xFFFF
(gdb) 

```

UweBonnes, 2021-05-29:
Opps,I missed that you have a F103 clone. I think while all F103x8 so long have 128 kByte flash, clones with xxx103x8 only have  64 kB. So flashing will fail above 0x08010000 in your case.

turtiustrek, 2021-05-29:
Ah, so this chip is actually capped at 64Kb? if so then this makes sense.
Thank you.

closed 2021-05-29 by turtiustrek

netroy, 2021-06-03:
@turtiustrek Just FYI: I have a bunch of these clones as well, and I just [trimmed down](https://github.com/netroy/blackmagic/commit/711559feef6520beea9f3f9348d4d4611f33f9fb) the binary to fit the smaller flash.