blackmagic-debug/blackmagic #899: SWD Speed

Frischmidt, 2021-07-06:
Hi,

i want to set down the speed of the SWD connection form the BMP to the target. I didn't find a command for GDB or a variable in the source code. Could you show me, how to manipulate the speed of the connection  or where i could change the default Baud Value for SWD?

Thank you for your help!

UweBonnes, 2021-07-06:
Normally, if you have problems, it is not the speed, but insufficient wiring  and reflectionas and such. Oftne more ground wires help. In case youreally need it, speed controll has been added to git head around end of 2020 "mon frequency ..." or for hosted "blackmagic -f ..."

Frischmidt, 2021-07-06:
thanks for the respond!

"blackmagic -f ..." which command is this? I couldn't use it with gdb...
for which dialogue i s this command?

UweBonnes, 2021-07-06:
Blackmagic compiled with "make PROBE_HOST=hosted" compiles a PC executable, able to talk to BMP firmware, stlink, jlink, ftdi an cmsis-dap. Look in the wiki.

Frischmidt, 2021-07-06:
when i use the ./blackmagic -f 115200

i got a error that -f is an invalid Option:

...:~/Dokumente/SHK/blackmagic-1.7.1/src$ sudo ./blackmagic -f 115200
./blackmagic: invalid option -- 'f'
Using 1d50:6018 E3EECAE1 Black Sphere Technologies Black Magic Probe
Remote is Black Magic Probe v1.7.1
1 0xe0001000: 0x00000000 <- does not match preamble (0xB105000D)
3 0xe0000000: 0xb1b1b1b1 <- does not match preamble (0xB105000D)
4 0xe0040000: 0x00000000 <- does not match preamble (0xB105000D)
Open file 115200 failed: No such file or directory
Can not map file: No such file or directory. Aborting!
...:~/Dokumente/SHK/blackmagic-1.7.1/src$ sudo ./blackmagic -h
Blackmagic PC-Hosted for: 
	BMP Firmware, ST-Link V2/3, CMSIS_DAP, JLINK and LIBFTDI/MPSSE 

this command doesn't set the default speed to like 115200 Hz, or am I wrong ?
is there a possibility to set the default speed value to a specific rate?

Where can i use you're second possibility with  "mon frequency ..."?

Thanks for your help!

Frischmidt, 2021-07-13:
Hi,
i'am not sure what you mean by using "git head"...
I could execute the makefile (after i installed some libs) and speak with my BMP. 
I got this output when i scan for a nordic dongle:

sudo ./blackmagic -f 115200 -t
BMP hosted v1.7.1-230-g41719c5
 for ST-Link V2/3, CMSIS_DAP, JLINK and LIBFTDI/MPSSE
Using 1d50:6018 E3EECAE1 Black Sphere Technologies
 Black Magic Probe
Running in Test Mode
Target voltage: 3.3V Volt
Update Firmware to allow to set max SWJ frequency
Device has fixed frequency for SWD
DPIDR 0x2ba01477 (v1 rev2)
Please update BMP firmware for substantial speed increase!
RESET_SEQ failed
AP   0: IDR=24770011 CFG=00000000 BASE=e00ff003 CSW=23000040 (AHB-AP var1 rev2
Halt via DHCSR: success 01030003 after 3ms
ROM: Table BASE=0xe00ff000 SYSMEM=0x00000001, designer 244 Partno   8
0 0xe000e000: Generic IP component - Cortex-M4 SCS (System Control Space) (PIDR = 0x04000bb00c  DEVTYPE = 0x00 ARCHID = 0x0000)-> cortexm_probe
CPUID 0x410fc241 (M4 var 0 rev 1)
1 0xe0001000: Generic IP component - Cortex-M3 DWT (Data Watchpoint and Trace) (PIDR = 0x04003bb002  DEVTYPE = 0x00 ARCHID = 0x0000)
2 0xe0002000: Generic IP component - Cortex-M3 FBP (Flash Patch and Breakpoint) (PIDR = 0x04002bb003  DEVTYPE = 0x00 ARCHID = 0x0000)
3 0xe0000000: Generic IP component - Cortex-M3 ITM (Instrumentation Trace Module) (PIDR = 0x04003bb001  DEVTYPE = 0x00 ARCHID = 0x0000)
4 0xe0040000: Debug component - Cortex-M4 TPIU (Trace Port Interface Unit) (PIDR = 0x04000bb9a1  DEVTYPE = 0x11 ARCHID = 0x0000)
5 0xe0041000: Debug component - Cortex-M4 ETM (Embedded Trace) (PIDR = 0x04000bb925  DEVTYPE = 0x13 ARCHID = 0x0000)
ROM: Table END
AP   1: IDR=02880000 CFG=00000000 BASE=00000000 CSW=00000000 (AHB-AP var0 rev0
***  1      Nordic nRF52 M4
***  2      Nordic nRF52 Access Port 
RAM   Start: 0x20000000 length = 0x40000
Flash Start: 0x00000000 length = 0x100000 blocksize 0x1000
Flash Start: 0x10001000 length = 0x1000 blocksize 0x1000

The Output says that the device has a fixed frequency for the SWD communication.
Update Firmware to allow to set max SWJ frequency
Device has fixed frequency for SWD

Should the command "blackmagic -f 115200" set the default value of the SWD connection? So that when i use the ARM gdb Interface, the probe has a frequency of 115200?

UweBonnes, 2021-07-13:
As explained before, SWD speed setting is only needed in very rare occasions. Don't bother if you are not sure you need it!  To allow this setting, both firmware and hosted if used need to be recent. SWD speed setting does not affect serial communication. Gdb serial has no real uart involved and is as fast as possible. Uart serial communication uses the speed setting apply that you use to open the device.

esden, 2021-12-16:
@Frischmidt Welcome to black magic. :)

Just to clarify. The `-f` option sets the baud rate for the UART speed that blackmagic is using to talk to the debugger hardware. This is not changing the SWD speed. The SWD speed is fixed can not be adjusted. SWD is what the debugger hardware uses to talk to your target that you are trying to debug.

In most cases the best approach to solve SWD communication issues is to improve the cabling or connections.

As far as I can tell the blackmagic hosted software is able to detect your target fine. So we need more information to understand your issue so that we can help you the best way possible. :)

UweBonnes, 2021-12-19:
Just to reply Esden, the -f option really changes SWJ speed. Recent firmware is needed to handle. "Please update BMP firmware for substantial speed increase!" points to an old firmware not able to handle. So if Frischmidt could retest with updated firmware, things could be clarified up.

dragonmux, 2023-02-02:
This looks like it was resolved, if SWD speed is still an issue, you may re-open this issue. However, we are going to close this as solved now.

closed 2023-02-02 by dragonmux