blackmagic-debug/blackmagic #900: Current master is broken for stm32f1 target

Azq2, 2021-07-22:
Target: stm32f103
```
Version      : 0x22
Option 1     : 0x00
Option 2     : 0x00
Device ID    : 0x0410 (STM32F10xxx Medium-density)
- RAM        : Up to 20KiB  (512b reserved by bootloader)
- Flash      : Up to 128KiB (size first sector: 4x1024)
- Option RAM : 16b
- System RAM : 2KiB
```

Command for flash:
```
monitor swdp_scan
attach 1
bt
load
kill
```

BMP hardware: stm32f103 (standart bluepill)

When i flash firmware with:
```
[ 3240.594580] usb 2-7: Product: Black Magic Probe (STLINK/V2) v1.7.1-232-g0c63903-dirty
```

I have random hard faults on any code on any chip.

But when i flash firmware with:
```
[42955.138587] usb 2-8: Product: Black Magic Probe (STLINK/V2) v1.7.1-154-g8067875
```

All works fine.

Azq2, 2021-07-23:
If you don't reproduce errors, i will try send .elf with problem and dumps from flash after flashing with both versions.

UweBonnes, 2021-07-23:
Can you try to reproduce with hosted? Compiling with "PROBE_HOST=hosted HOSTED_BMP_ONLY=1" has the least external dependancies. If it is visible with hosted, bisecting is easy. Otherwise I can only try to reproduce next week.

Azq2, 2021-07-25:
@UweBonnes in hosted mode works fine, but in native mode (on stm32f103) i still reproduce bug
[example.zip](https://github.com/blacksphere/blackmagic/files/6873575/example.zip)

bad_flash.bin: flashed with v1.7.1-232-g0c63903-dirty
good_flash.bin: flashed with v1.7.1-154-g8067875
flash-dumper.elf: original elf file

Binary diff:
```diff
--- bad		2021-07-25 11:33:34.828479773 +0300
+++ good	2021-07-25 11:33:39.348220008 +0300
@@ -447,9 +447,54 @@
 00001cd0  21 d4 98 07 49 bf 5b 08  9b 08 01 20 13 60 5c bf  |!...I.[.... .`\.|
 00001ce0  02 20 13 60 70 47 99 b2  09 b9 10 20 1b 0c d9 b2  |. .`pG..... ....|
 00001cf0  09 b9 08 30 1b 0a 19 07  04 bf 1b 09 04 30 99 07  |...0.........0..|
-00001d00  04 bf 9b 08 02 30 ff ff  ff ff ff ff ff ff ff ff  |.....0..........|
-00001d10  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
-*
+00001d00  04 bf 9b 08 02 30 d9 07  03 d4 5b 08 00 f1 01 00  |.....0....[.....|
+00001d10  03 d0 13 60 70 47 00 20  70 47 20 20 70 47 00 00  |...`pG. pG  pG..|
+00001d20  10 b5 0c 46 01 21 ff f7  05 ff 02 46 28 b9 4f f4  |...F.!.....F(.O.|
+00001d30  a0 71 04 4b 04 48 00 f0  3f fb 01 23 44 61 03 61  |.q.K.H..?..#Da.a|
+00001d40  10 bd 00 bf 89 3b 00 08  9a 3b 00 08 2d e9 f0 4f  |.....;...;..-..O|
+00001d50  91 46 0a 69 d9 f8 10 30  0c 46 9a 42 be bf 0b 46  |.F.i...0.F.B...F|
+00001d60  4c 46 99 46 27 69 d9 f8  10 a0 a3 68 61 68 07 eb  |LF.F'i.....hah..|
+00001d70  0a 06 b3 42 85 b0 b8 bf  01 31 ff f7 db fe 30 b9  |...B.....1....0.|
+00001d80  02 46 40 f2 5d 11 43 4b  43 48 00 f0 15 fb 00 f1  |.F@.].CKCH......|
+00001d90  14 05 2b 46 00 22 05 eb  86 08 43 45 21 d3 04 f1  |..+F."....CE!...|
+00001da0  14 03 03 eb 87 07 09 f1  14 03 03 eb 8a 02 02 92  |................|
+00001db0  3a 1b 15 3a 22 f0 03 02  04 32 04 f1 15 01 8f 42  |:..:"....2.....B|
+00001dc0  38 bf 04 22 01 92 02 9a  03 93 9a 42 0c d8 00 2e  |8..".......B....|
+00001dd0  03 dd 58 f8 04 3d 00 2b  59 d0 06 61 05 b0 bd e8  |..X..=.+Y..a....|
+00001de0  f0 8f 43 f8 04 2b d8 e7  b3 f8 00 a0 ba f1 00 0f  |..C..+..........|
+00001df0  23 d0 a9 46 4f f0 00 0c  04 f1 14 0e 5e f8 04 2b  |#..FO.......^..+|
+00001e00  d9 f8 00 10 1f fa 82 fb  89 b2 0a fb 0b 11 61 44  |..............aD|
+00001e10  d9 f8 00 c0 12 0c 4f ea  1c 4c 0a fb 02 c2 02 eb  |......O..L......|
+00001e20  11 42 89 b2 41 ea 02 41  77 45 4f ea 12 4c 49 f8  |.B..A..AwEO..LI.|
+00001e30  04 1b e3 d8 01 9a 45 f8  02 c0 03 9a 04 33 b2 f8  |......E......3..|
+00001e40  02 90 b9 f1 00 0f 20 d0  ae 46 4f f0 00 0a 29 68  |...... ..FO...)h|
+00001e50  04 f1 14 0c bc f8 00 b0  be f8 02 20 89 b2 09 fb  |........... ....|
+00001e60  0b 22 92 44 41 ea 0a 41  4e f8 04 1b 5c f8 04 2b  |.".DA..AN...\..+|
+00001e70  be f8 00 10 12 0c 09 fb  02 11 67 45 01 eb 1a 41  |..........gE...A|
+00001e80  4f ea 11 4a e6 d8 01 9a  a9 50 04 35 9b e7 01 3e  |O..J.....P.5...>|
+00001e90  9d e7 00 bf 89 3b 00 08  9a 3b 00 08 2d e9 f8 43  |.....;...;..-..C|
+00001ea0  15 46 12 f0 03 02 06 46  0f 46 07 d0 25 4c 01 3a  |.F.....F.F..%L.:|
+00001eb0  00 23 54 f8 22 20 ff f7  9f fe 07 46 ad 10 3d d0  |.#T." .....F..=.|
+00001ec0  74 6a 7c b9 10 20 ff f7  11 fe 02 46 70 62 28 b9  |tj|.. .....Fpb(.|
+00001ed0  4f f4 d7 71 1c 4b 1d 48  00 f0 6e fa c0 e9 01 44  |O..q.K.H..n....D|
+00001ee0  04 60 c4 60 d6 f8 24 80  d8 f8 08 40 4c b9 40 f2  |.`.`..$....@L.@.|
+00001ef0  71 21 30 46 ff f7 14 ff  00 23 04 46 c8 f8 08 00  |q!0F.....#.F....|
+00001f00  03 60 4f f0 00 09 eb 07  0a d5 39 46 22 46 30 46  |.`O.......9F"F0F|
+00001f10  ff f7 1c ff 80 46 39 46  30 46 ff f7 4b fe 47 46  |.....F9F0F..K.GF|
+00001f20  6d 10 0b d0 20 68 38 b9  22 46 21 46 30 46 ff f7  |m... h8."F!F0F..|
+00001f30  0d ff 20 60 c0 f8 00 90  04 46 e4 e7 38 46 bd e8  |.. `.....F..8F..|
+00001f40  f8 83 00 bf 08 3d 00 08  f5 3a 00 08 9a 3b 00 08  |.....=...:...;..|
+00001f50  2d e9 f0 47 0c 46 07 46  91 46 23 69 49 68 03 eb  |-..G.F.F.F#iIh..|
+00001f60  62 18 a3 68 4f ea 62 1a  08 f1 01 06 b3 42 0b db  |b..hO.b......B..|
+00001f70  38 46 ff f7 df fd 05 46  48 b9 02 46 40 f2 d9 11  |8F.....FH..F@...|
+00001f80  29 4b 2a 48 00 f0 18 fa  01 31 5b 00 ee e7 00 23  |)K*H.....1[....#|
+00001f90  00 f1 14 01 00 f1 10 02  18 46 53 45 37 db 20 69  |.........FSE7. i|
+00001fa0  2a ea ea 7a 04 f1 14 03  19 f0 1f 09 01 eb 8a 01  |*..z............|
+00001fb0  03 eb 80 00 2f d0 8a 46  4f f0 00 0c c9 f1 20 0e  |..../..FO..... .|
+00001fc0  1a 68 02 fa 09 f2 42 ea  0c 02 4a f8 04 2b 53 f8  |.h....B...J..+S.|
+00001fd0  04 2b 98 42 22 fa 0e fc  f2 d8 03 1b 15 3b 23 f0  |.+.B"........;#.|
+00001fe0  03 03 04 33 04 f1 15 02  90 42 38 bf 04 23 41 f8  |...3.....B8..#A.|
+00001ff0  03 c0 bc f1 00 0f 01 d0  08 f1 02 06 01 3e 38 46  |.............>8F|
 00002000  21 46 2e 61 ff f7 d6 fd  28 46 bd e8 f0 87 42 f8  |!F.a....(F....B.|
 00002010  04 0f 01 33 c1 e7 04 39  53 f8 04 2b 98 42 41 f8  |...3...9S..+.BA.|
 00002020  04 2f f9 d8 ea e7 00 bf  89 3b 00 08 9a 3b 00 08  |./.......;...;..|
@@ -658,9 +703,54 @@
[... 80 lines cut ...]
+00003dd0  42 42 42 02 02 02 02 02  02 02 02 02 02 02 02 02  |BBB.............|
+00003de0  02 02 02 02 02 02 02 10  10 10 10 20 00 00 00 00  |........... ....|
+00003df0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
+*
+00003ec0  00 00 00 00 00 00 00 00  00 00 00 00 04 00 00 20  |............... |
+00003ed0  00 00 00 00 8c 3e 00 08  ac 3e 00 08 6c 3e 00 08  |.....>...>..l>..|
+00003ee0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
 *
[... cut ...]
+00003fc0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
+00003fd0  43 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |C...............|
+00003fe0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
+00003ff0  43 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |C...............|
 00004000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
 00004010  b9 27 00 08 19 24 00 08  00 00 00 00 6b 3d 00 08  |.'...$......k=..|
 00004020  58 3d 00 08 4f 3d 00 08  4f 3d 00 08 4f 3d 00 08  |X=..O=..O=..O=..|
```

UweBonnes, 2021-07-25:
Commit 0c6390307115b48 (Recover from bad AP access )sets the low_access timeout to 20 ms. Firmware only ticks at 10 hertz, so the sequence low_access set timeout, send out 8 bit command, read 3 bit result as "wait" and timer increment tick already hits the timeout even so only miminal time has elapsed. Increment timeout so that at least for one full timer tick the wait happens. 

Please check #902 and report back

Azq2, 2021-07-26:
@UweBonnes works fine.
```
25b94290f84e929153e1c01c97c400a0  new.bin
25b94290f84e929153e1c01c97c400a0  old.bin
```

UweBonnes, 2021-07-27:
Fixes in recent GIT head

closed 2021-07-27 by UweBonnes