blackmagic-debug/blackmagic #903: GDB/probe connection occasionally gets stuck

bobtwinkles, 2021-07-25:
Something is causing GDB to hang after sending a qL request packet to the blackmkagic host.

GDB log:
<details>
Sending packet: $qRcmd,737764705f7363616e#3c...Ack
Packet received: O54617267657420766f6c746167653a20332e33560a
Target voltage: 3.3V
Packet received: O417661696c61626c6520546172676574733a0a
Available Targets:
Packet received: O4e6f2e20417474204472697665720a
No. Att Driver
Packet received: O203120202020202053544d33324c3178204d330a
 1      STM32L1x M3
Packet received: OK
Sending packet: $vAttach;1#37...Ack
Packet received: T05
Packet vAttach (attach) is supported
Sending packet: $qC#b4...Ack
Packet received: 
Sending packet: $Hg0#df...Ack
Packet received: 
Sending packet: $qXfer:features:read:target.xml:0,3fb#46...Ack
Packet received: m<?xml version="1.0"?><!DOCTYPE target SYSTEM "gdb-target.dtd"><target>  <architecture>arm</architecture>  <feature name="org.gnu.gdb.arm.m-profile">    <reg name="r0" bitsize="32"/>    <reg name="r1" bitsize="32"/>    <reg name="r2" bitsize="32"/>    <reg name="r3" bitsize="32"/>    <reg name="r4" bitsize="32"/>    <reg name="r5" bitsize="32"/>    <reg name="r6" bitsize="32"/>    <reg name="r7" bitsize="32"/>    <reg name="r8" bitsize="32"/>    <reg name="r9" bitsize="32"/>    <reg name="r10" bitsize="32"/[508 bytes omitted]
Sending packet: $qXfer:features:read:target.xml:3fb,3fb#11...Ack
Packet received: mme="faultmask" bitsize="8" save-restore="no"/>    <reg name="control" bitsize="8" save-restore="no"/>  </feature></target>
Sending packet: $qXfer:features:read:target.xml:475,3fb#b6...Ack
Packet received: l
Sending packet: $g#67...Ack
Packet received: c0260020a42700200000000000000000d0430008a4270020c0260020a02400206c25002000000240405c004078250020a0270020842400208d0c00083c1d000800000061842400200000000000000000
Sending packet: $qOffsets#4b...Ack
Packet received: 
Sending packet: $qSymbol::#5b...Ack
Packet received: 
Packet qSymbol (symbol-lookup) is NOT supported
Sending packet: $qfThreadInfo#bb...Ack
Packet received: 
Sending packet: $qL1200000000000000000#50...Ack
Packet received: 
</details>

Log from `./blackmagic -v 31 `
<details>

INFO: Open USB 8087:0aa7 class e0 failed
BMP hosted v1.7.1-232-g0c63903
 for ST-Link V2/3, CMSIS_DAP, JLINK and LIBFTDI/MPSSE
Using 1d50:6018 BFE7C4F0 Black Sphere Technologies
 Black Magic Probe  v1.7.1-232-g0c63903
+#!GA#
       K v1.7.1-232-g0c63903
Remote is  v1.7.1-232-g0c63903
Listening on TCP: 2000
Got connection
gdb_getpacket : qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+
gdb_putpacket : PacketSize=400;qXfer:memory-map:read+;qXfer:features:read+
gdb_getpacket : vMustReplyEmpty
*** Unsupported packet: vMustReplyEmpty
gdb_putpacket : 
gdb_getpacket : !
gdb_putpacket : OK
gdb_getpacket : Hg0
*** Unsupported packet: Hg0
gdb_putpacket : 
gdb_getpacket : qXfer:features:read:target.xml:0,3fb
gdb_putpacket : E01
gdb_getpacket : qTStatus
*** Unsupported packet: qTStatus
gdb_putpacket : 
gdb_getpacket : ?
gdb_putpacket : W00
[... 4829 lines cut ...]
*** Unsupported packet: qfThreadInfo
gdb_putpacket : 
gdb_getpacket : qL1200000000000000000
*** Unsupported packet: qL1200000000000000000
gdb_putpacket : 
</details>

And the GDB script that hangs:
```gdb
target extended-remote localhost:2000
set debug remote 1
monitor swdp_scan
attach 1                                # Appears to hang during attach, here
load
```

GDB version info:
```
$ arm-none-eabi-gdb --version
GNU gdb (GNU Arm Embedded Toolchain 10-2020-q4-major) 10.1.90.20201028-git
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

bobtwinkles, 2021-07-25:
it's possible the target's (an STM32L151C8T6)  SWD/JTAG core has suffered some unfortunate electronic accidents and thus isn't operating at 100%, but we probably shouldn't be hanging GDB regardless.

UweBonnes, 2021-07-26:
Does the STM32L1 use longer sleep versions? BMP does not enable debug in these states by default. If some transaction happens in that state, timeout will probably happen. Somenody needs to implement...

bobtwinkles, 2021-07-26:
I'm not sure I understand what you mean by "longer sleep versions". To the best of my knowledge the STML1 parts use ST's standard SWD/JTAG implementation but I haven't absorbed that particular part of the reference manual in detail.

UweBonnes, 2021-07-26:
Does your code use WFI/WFE? What is the minimal rate at interrupts happening. E.g. in Ethernut we use WFI, but also 1 kHz timer tick. That way, longest sleep is 1 mS.

bobtwinkles, 2021-07-26:
Oh, I see. My code polls the USB peripheral in a tight loop, with no interrupts at all. So the CPU shouldn't ever be going to sleep.

UweBonnes, 2021-07-26:
A simple "/blackmagic -t" log  will help more.

bobtwinkles, 2021-07-29:
Sorry, this didn't repro for awhile but I saw it again today. Unfortunately I don't have a detail capture from exactly after it happened (I'll try to grab that next time...). The non-detailed log just said that the SWD probe failed and BMP automatically fell back to JTAG, which worked for scan. After attempting to attach using JTAG it seems like the debug unit is really stuck in a bad state, and produced these detail logs: https://gist.github.com/bobtwinkles/5b81d4d597dc60ab4681c024b9815de5

bobtwinkles, 2021-07-29:
oh, another potentially important datapoint is that the attach attempts via GDB do appear to at least halt the CPU, as my heatbeat LED stops flashing.

UweBonnes, 2021-07-29:
Random problem in most cases are hardware problems, flacky cables, flaky supply, loose pins. "blackmagic -t" on the L1 Discos works repeated.

esden, 2021-12-16:
@bobtwinkles I also have the feeling that this seems like a "flaky hardware" problem. But I could always be wrong. Should we continue investigating or should we close this issue?

bobtwinkles, 2021-12-26:
I finished the project that this was attached to and didn't see too many more issues. I also agree that it was likely a hardware problem, sorry for the noise.

If something similar happens in the future I now have much better tools for diagnosing them, so hopefully I'll at least have some SIGROK traces for you =).

closed 2021-12-26 by bobtwinkles