blackmagic-debug/blackmagic #904: NXP LPC11xx "XL" variants not detected (& supported)

compuphase, 2021-07-29:
The BMP supports the LPC11xx microntroller series, for the normal and the low-power ("L") variants. But not the extended low-power ("XL") variants. The cause is that the MCU is detected by reading the "DEVICE_ID" register, but the XL variants lack this register. You can still get the Device ID through an IAP call.

This is a catch-22: before you make an IAP call, you must be certian that there is an IAP function at the fixed address. In other words, you must know that the MCU is from the LPC11xx series, before you can get the Device ID. But to establish that the MCU is from the LPC11xx series, you need the... Device ID,

I am thinking of two solutions:

1) Read the memory at the address of the IAP function and check that it matches the entry code of the IAP function. If it matches, call the IAP function.

2) Add a "monitor" command to set the MCU, so you are basically overruling autodetection.

At this moment, I prefer option 1. Your thoughts on this are welcome, before I start on implementing this.

UweBonnes, 2021-07-29:
As always, please compile hosted , run "blackmagic -t" and post the result. Let's hope the ROM Table is unique so we can make the IAP call without danger to crash.

UweBonnes, 2021-07-29:
Otherwise, LPC11 gives a lot of hits on the NXP website. Please point a link to the reference/user manual.

And option 1 sounds most reasonable to me. Make sure the LPC_PROBE in that case come last, so we do not stumble on the feets of other already supported devices.

compuphase, 2021-07-30:
Link to the user manual: https://community.nxp.com/pwmxy87654/attachments/pwmxy87654/lpc/28119/2/UM10398.pdf
This is revision 12.4, the latest that I know of.
The part identification numbers are in section 26.5.11 (starting at page 431). Part numbers that end with /103, /203 or /303 are XL variants. Chapter 3.1 (page 23) says that the DEVICE_ID register is not available in XL variants.

Below is the output of "blackmagic -t". Interestingly, when probing for LPC11xx it comes up with a rubbish IDCODE (as expected) but when probing for LPC8xx, it has the correct IDCODE 0x00020023 (for the LPC1112/103 that was attached). It looks like the DEVICE_ID register was simply moved (despite what the manual says and which is repeated in https://www.nxp.com/docs/en/supporting-information/TN00005.pdf).

INFO: Open USB 0930:6545 class  0 failed
BMP hosted v1.7.1-233-g6d6a67b
 for ST-Link V2/3, CMSIS_DAP, JLINK and LIBFTDI/MPSSE
Using 1d50:6018 7BB180B4 Black Sphere Technologies
 Black Magic Probe  v1.7.1-233-g6d6a67b-dirty
Powering up device!
Running in Test Mode
Target voltage: 3.3V Volt
Speed set to  3.2727 MHz for SWD
DPIDR 0x0bb11477 (v1 MINDP rev0)
RESET_SEQ failed
AP   0: IDR=04770021 CFG=00000000 BASE=e00ff003 CSW=03000040 (AHB-AP var2 rev0
Halt via DHCSR: success 01030003 after 2ms
ROM: Table BASE=0xe00ff000 SYSMEM=0x00000001, designer 43b Partno 471
0 0xe000e000: Generic IP component - Cortex-M0 SCS (System Control Space) (PIDR = 0x04000bb008  DEVTYPE = 0x00 ARCHID = 0x0000)-> cortexm_probe
CPUID 0x410cc200 (M0 var 0 rev 0)
LPC11xx: Unknown IDCODE 0x3022302b
LPC8xx: Unknown IDCODE 0x00020023
1 0xe0001000: Generic IP component - Cortex-M0 DWT (Data Watchpoint and Trace) (PIDR = 0x04000bb00a  DEVTYPE = 0x00 ARCHID = 0x0000)
2 0xe0002000: Generic IP component - Cortex-M0 BPU (Breakpoint Unit) (PIDR = 0x04000bb00b  DEVTYPE = 0x00 ARCHID = 0x0000)
ROM: Table END
*** 1 Unknown ARM Cortex-M Designer 43b Partno 471 M0

UweBonnes, 2021-07-31:
Solved with #906

closed 2021-07-31 by UweBonnes