blackmagic-debug/blackmagic #912: LPC1343

oleg-login, 2021-08-12:
Please report Designer   0 and Partno   0 and the probed device
Available Targets:
No. Att Driver
*** 1 Unknown ARM Cortex-M Designer   0 Partno   0 M3

UweBonnes, 2021-08-12:
It seem, nobody cared for the LPC13 series yet. Somebody with the part needs to understand the LPC13 programmin algos , understand the differences to the other LPC and write code and post a patch request.

oleg-login, 2021-08-12:
It's clear. But, in a stable release v1.7.1, everything is well determined.

UweBonnes, 2021-08-13:
You should be able to attach to the device and debiug it. can you try and report back?

oleg-login, 2021-08-18:
C:\Users\oleg>arm-none-eabi-gdb.exe

(gdb) target extended-remote \\.\COM15
Remote debugging using \\.\COM15
(gdb) monitor tpwr enable
Enabling target power
(gdb) monitor swdp_scan
Target voltage: 3.3V
Please report Designer   0 and Partno   0 and the probed device
Available Targets:
No. Att Driver
*** 1 Unknown ARM Cortex-M Designer   0 Partno   0 M3
(gdb) att 1
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
warning: while parsing target memory map (at line 1): Required element <memory> is missing
0x00000e6c in ?? ()
(gdb) set mem inaccessible-by-default off
(gdb) load d:/project/blinky.elf
Loading section .text, size 0x168 lma 0x0
Start address 0xc0, load size 360
Transfer rate: 13 KB/sec, 360 bytes/write.
(gdb) file d:/project/blinky.elf
A program is being debugged already.
Are you sure you want to change the file? (y or n) y
Reading symbols from d:/project/blinky.elf...done.
(gdb) b blinky
Breakpoint 2 at 0xc6: file main.c, line 6.
(gdb)

UweBonnes, 2021-08-19:
As you see, debugging is possible. The fake success for the load command is caused by no flash regions defined. So gdb thinks it can upload to RAM. Obvious a "compare-section" will fail.

esden, 2021-12-16:
Adding this hardware should be fairly easy. The flash programming algorithm can probably be derived from other already supported LPC parts. Then we need to add the hardware ID to the database.

@oleg-login would you be interested in taking a stab at this? If so a pull request would be very much appreciated. :)

perigoso, 2022-07-25:
this doesnt quite add up, as there is a PR/commit that supposedly added support for it
28f0ced97e391f60d282edbee58bf9b714c3b155 in #551 

this was included in v1.7.1 too, but by the fact they got the `Please report Designer` prompt, they were running the code after the id switch case probe logic, so the support is probably broken, lpc11xx is not probed on `Cortex-M3 ROM` like the log for this device (Designer 0 Partno 0 M3) suggests it should

PR #1132 (2022-07-26): target: fix probing for lpc1343
I have no means to test this, but looking at the symptoms, and the git history, I believe this should fix it

fix #912
closed 2022-07-26 by dragonmux