blackmagic-debug/blackmagic #914: Request support for ATSAMR34

vslinuxdotnet, 2021-08-18:
Hello, thank you for your great project.
But is possible to support the ATSAMR34 ?
Thank You

UweBonnes, 2021-08-19:
As the ATSAMR34 is a CortexM, debugging is already possible, even as the chip is not recognized. However nobody has written the flash algorithm yet, so for flashing support somebody must write the code. This is often no big deal, as the flashing procedure may be similar to other SAM flashing procedures already implemented. I do not have a ATSAMR34, so  I can not test and so I will not do it.

esden, 2021-12-16:
Adding this hardware should be fairly easy. The flash programming algorithm can probably be derived from other already supported ATSAM parts. Then we need to add the hardware ID to the database.

@vslinuxdotnet would you be interested in taking a stab at this? If so a pull request would be very much appreciated. :)

vslinuxdotnet, 2021-12-29:
> Adding this hardware should be fairly easy. The flash programming algorithm can probably be derived from other already supported ATSAM parts. Then we need to add the hardware ID to the database. [...]

@esden at moment I don't have many free time, but I test it, if you need.
I buy a programmer for that device at two months :)
Thank you

UweBonnes, 2021-12-29:
Do you have any debugger that is supported by BMP/hosted(Stlinkv2|3, cmsis-dap, jlink and mpsse)? For a start, compile hosted and post the output of "blackmagic -t". That way, we can see if the device is detectable from the ROM Table.

vslinuxdotnet, 2021-12-30:
> 
Yes I have JLINK and STLINK V2.
I compile it then run ./blackmagic -t
Result:

BMP hosted (BMP Only) v1.7.1-331-gc832cb0
Using:
 Black Magic Probe (Unknown)  Firmware
Running in Test Mode
Target voltage: unknown Volt
Update Firmware to allow to set max SWJ frequency
Device has fixed frequency for SWD
DPIDR 0x0bc11477 (v1 MINDP rev0)
Please update BMP firmware for substantial speed increase!
RESET_SEQ failed
AP   0: IDR=04770031 CFG=00000000 BASE=41003003 CSW=03000040 (AHB-AP var3 rev0
Halt via DHCSR: success 00030003 after 14ms
ROM: Table BASE=0x41003000 SYSMEM=0x00000001, designer  1f Partno cd0
ROM: Table BASE=0xe00ff000 SYSMEM=0x00000001, designer 43b Partno 4c0
 0 0xe000e000: Generic IP component - Cortex-M0 SCS (System Control Space) (PIDR = 0x04000bb008  DEVTYPE = 0x00 ARCHID = 0x0000) -> cortexm_probe
CPUID 0x410cc601 (M0+ var 0 rev 1)
 1 0xe0001000: Generic IP component - Cortex-M0 DWT (Data Watchpoint and Trace) (PIDR = 0x04000bb00a  DEVTYPE = 0x00 ARCHID = 0x0000)
 2 0xe0002000: Generic IP component - Cortex-M0 BPU (Breakpoint Unit) (PIDR = 0x04000bb00b  DEVTYPE = 0x00 ARCHID = 0x0000)
 ROM: Table END
1 0x41006000: Debug component - CoreSight MTB-M0+ (Simple Execution Trace) (PIDR = 0x04001bb932  DEVTYPE = 0x31 ARCHID = 0x0a31)
ROM: Table END
***  1      Atmel SAML21 M0+
RAM   Start: 0x20000000 length = 0x8000
Flash Start: 0x00000000 length = 0x40000 blocksize 0x100

UweBonnes, 2021-12-30:
From the SAM R34/R35" datasheet: "The SAM R34/R35 contains an ATSAML21 Die Revision C ARM Cortex-M0+ processor. " So that seem fine!

vslinuxdotnet, 2021-12-30:
> From the SAM R34/R35" datasheet: "The SAM R34/R35 contains an ATSAML21 Die Revision C ARM Cortex-M0+ processor. " So that seem fine!
OK so next steeps are?

UweBonnes, 2021-12-30:
Debug and flashing should work now. Test that.  Please check the wiki for general usage. Switch over to the discord channel for general discussion. And if you find errors, reopen or open a new issue if that is a different subject.

closed 2021-12-30 by UweBonnes