blackmagic-debug/blackmagic #917: Add support for the Texas Instruments CC2538

huegelcesar, 2021-08-31:
First of all, thank you for your amazing project!!

I would like to have the [blacksphere/blackmagic](https://github.com/blacksphere/blackmagic) supporting the Texas Instruments CC2538 debugging and flashing.

Is that possible?
I'm willing to collaborate with you. I have the necessary time and the target device, but I'm not clear where to start

Thank you!

UweBonnes, 2021-08-31:
The CC2538 is a CortexM, so debugging is possible in principle. However is uses cJTAG by default and must be active switched to JTAG. BMP does not talk cJTAG (yet), so this switch sequence is needed. My advice: get BMP hosted going with any supported adapter. Start to add the switching sequence. Then at least debugging should be possible. Then add proper target detection and the programming algos. This all can be done w/o reprogramming the firmware. cJTAG support would need firmware changes. Final integration in  master is possible, if the new code is not too flash hungry.

huegelcesar, 2021-08-31:
What do you mean by "get BMP hosted going with any supported adapter"?

UweBonnes, 2021-08-31:
Read the wiki, read the code, scan old discussions, ask specific questions.

esden, 2021-12-16:
Hi @huegelcesar,

Welcome to black magic! :)

Adding this hardware should be fairly easy. The flash programming algorithm can probably be derived from other already supported TI parts. Then we need to add the hardware ID to the database.

The best place to look is in the `src/target` directory, each group of similar MCU cores has their own .c file with the necessary hooks implemented. If you are lucky all you need to do is to add your particular chip to one of them.