blackmagic-debug/blackmagic #918: STM32L0  connection and flash upload fail

silicongarage, 2021-09-03:
sw ver: 1.7.1
hw ver: 2.1e
device: STM32L031F6

With SRST disabled (mon con d), the device is correctly identified (mon s) as STM32L0x M0+ but am unable to attach (mon s)
With SRST enabled (mon con e), the device is identified as ARM Cortex-M M0+ as can be attached (mon s) but fails when uploading flash (load)

UweBonnes, 2021-09-03:
What version exact? Can you run hosted "make PROBE_HOST=hosted clean; make PROBE_HOST=hosted; ./blackmagic -t" and post the console output?

silicongarage, 2021-09-03:
I just tried building hosted with both arm-gcc 9.3.1 & 8.3.1 and got the following message:
platforms/hosted/ftdi_bmp.c:344:8: error: 'ftdi_usb_purge_buffers' is deprecated [-Werror,-Wdeprecated-declarations]

UweBonnes, 2021-09-03:
On git-head, hosted should build w/o ftdi by default. Are you using git head?

silicongarage, 2021-09-03:
okay, was using source from 1.7.1

Running in Test Mode
Target voltage: 3.3V Volt
Update Firmware to allow to set max SWJ frequency
Device has fixed frequency for SWD
DPIDR 0x0bc11477 (v1 MINDP rev0)
Please update BMP firmware for substantial speed increase!
RESET_SEQ failed
AP   0: IDR=04770031 CFG=00000000 BASE=f0000003 CSW=03000040 (AHB-AP var3 rev0
Halt via DHCSR: success 00030003 after 11ms
ROM: Table BASE=0xf0000000 SYSMEM=0x00000001, designer  20 Partno 425
ROM: Table BASE=0xe00ff000 SYSMEM=0x00000001, designer 43b Partno 4c0
 0 0xe000e000: Generic IP component - Cortex-M0 SCS (System Control Space) (PIDR = 0x04000bb008  DEVTYPE = 0x00 ARCHID = 0x0000) -> cortexm_probe
CPUID 0x410cc601 (M0+ var 0 rev 1)
 1 0xe0001000: Generic IP component - Cortex-M0 DWT (Data Watchpoint and Trace) (PIDR = 0x04000bb00a  DEVTYPE = 0x00 ARCHID = 0x0000)
 2 0xe0002000: Generic IP component - Cortex-M0 BPU (Breakpoint Unit) (PIDR = 0x04000bb00b  DEVTYPE = 0x00 ARCHID = 0x0000)
 ROM: Table END
1 Entry 0x200002 -> Not present
ROM: Table END
***  1      STM32L0x M0+
RAM   Start: 0x20000000 length = 0x5000
Flash Start: 0x08000000 length = 0x10000 blocksize 0x80
Flash Start: 0x08010000 length = 0x10000 blocksize 0x80
Flash Start: 0x08020000 length = 0x10000 blocksize 0x80
Flash Start: 0x08080000 length = 0x1800 blocksize 0x4

silicongarage, 2021-09-03:
I just unplugged bmp from usb and back in and now get this:

BMP hosted (BMP Only) 
Using:
   
Running in Test Mode
Target voltage: 3.3V Volt
Update Firmware to allow to set max SWJ frequency
Device has fixed frequency for SWD
Exception: SWDP invalid ACK
Trying old JTAG to SWD sequence
Exception: SWDP invalid ACK
No usable DP found
Scan SWD failed, trying JTAG!
Update Firmware to allow to set max SWJ frequency
Device has fixed frequency for JTAG
No target found

silicongarage, 2021-09-03:
just observed the following behavior:
(1) go into arm-gdb and "mon con e", "mon s", "quit"
(2) first run of ./blackmagic -t  works
(3) second run of ./blackmagic -t fails

UweBonnes, 2021-09-03:
If you need to connect under reset, "blackmagic -t -C". "blackmagic -h" is your friend.

UweBonnes, 2021-09-03:
Please provide the  "blackmagic -t -C" log.

silicongarage, 2021-09-03:
Connecting under reset
Running in Test Mode
Target voltage: 3.3V Volt
Update Firmware to allow to set max SWJ frequency
Device has fixed frequency for SWD
DPIDR 0x0bc11477 (v1 MINDP rev0)
Please update BMP firmware for substantial speed increase!
RESET_SEQ failed
AP   0: IDR=04770031 CFG=00000000 BASE=f0000003 CSW=03000040 (AHB-AP var3 rev0
ROM: Table BASE=0xf0000000 SYSMEM=0x00000001, designer  20 Partno 425
ROM: Table BASE=0xe00ff000 SYSMEM=0x00000001, designer 43b Partno 4c0
 0 0xe000e000: Generic IP component - Cortex-M0 SCS (System Control Space) (PIDR = 0x04000bb008  DEVTYPE = 0x00 ARCHID = 0x0000) -> cortexm_probe
CPUID 0x410cc601 (M0+ var 0 rev 1)
 1 0xe0001000: Generic IP component - Cortex-M0 DWT (Data Watchpoint and Trace) (PIDR = 0x04000bb00a  DEVTYPE = 0x00 ARCHID = 0x0000)
 2 0xe0002000: Generic IP component - Cortex-M0 BPU (Breakpoint Unit) (PIDR = 0x04000bb00b  DEVTYPE = 0x00 ARCHID = 0x0000)
 ROM: Table END
1 Entry 0x200002 -> Not present
ROM: Table END
***  1      STM32L0x M0+
RAM   Start: 0x20000000 length = 0x5000
Flash Start: 0x08000000 length = 0x10000 blocksize 0x80
Flash Start: 0x08010000 length = 0x10000 blocksize 0x80
Flash Start: 0x08020000 length = 0x10000 blocksize 0x80
Flash Start: 0x08080000 length = 0x1800 blocksize 0x4

UweBonnes, 2021-09-03:
Does  "blackmagic -t -C" work repeated and only bmp/firmware fail on the second attempt. Then update your firmware and try again. Otherweise a log of a failing attempt is what is needed.

silicongarage, 2021-09-03:
blackmagic -t -C works repeatedly
arm-gdb without SRST will not attach, but reports STM32L0x M0+
arm-gdb with SRST will attach, reports ARM Cortex-M M0+ , but fails uploading of flash
I have re-loaded 1.7.1 and still have the problem

silicongarage, 2021-09-03:
okay, I just pulled the git head and build blackmagic.bin
this appears to fix flash upload problem if I do a "mon con e"

UweBonnes, 2021-09-04:
Probably
```Author: Uwe Bonnes <bon@elektron.ikp.physik.tu-darmstadt.de>
Date:   Thu Nov 28 14:37:44 2019 +0100

    Erase: Fix endless erase when erase-area did not end on (page|block) boarder.

```
helped. Plesae close.

closed 2021-09-05 by silicongarage