blackmagic-debug/blackmagic #920: stm32h7: "Reply contains invalid hex digit 105" during compare-sections

arbrauns, 2021-09-06:
When trying to use `compare-sections` to verify the flash of my STM32H743XIH6, I get the following error:

```
$ arm-none-eabi-gdb -nx --batch -ex 'target extended-remote :2000' -ex 'monitor swdp_scan' -ex 'attach 1' -ex 'compare-sections' /tmp/binary.elf
Target voltage: 1.8V
Available Targets:
No. Att Driver
 1      STM32H7 M7
prvCheckTasksWaitingTermination () at /[redacted]/tasks.c:3589
3589	/[redacted]/tasks.c: No such file or directory.
Section .virtualboot, range 0x8000000 -- 0x8000008: matched.
Section .isr_vector, range 0x8020000 -- 0x8020298: matched.
Ignoring packet error, continuing...
Reply contains invalid hex digit 105
Ignoring packet error, continuing...
Can't detach process.
```

`blackmagic -v31` log (too big to paste): [blackmagic.log](https://github.com/blacksphere/blackmagic/files/7115264/blackmagic.log)

Firmware and hosted are both on the latest master (dd28fa57435fcea107f23621b4d2392e6f656520) with this hack to make it compile on gcc 11:

```diff
diff --git a/src/Makefile b/src/Makefile
index 188ecfa..ef56f82 100644
--- a/src/Makefile
+++ b/src/Makefile
@@ -8,7 +8,7 @@ MAKEFLAGS += --no-print-dir
 Q := @
 endif
 
-CFLAGS += -Wall -Wextra -Werror -Wno-char-subscripts \
+CFLAGS += -Wall -Wextra -Werror -Wno-char-subscripts -Wno-error=stringop-overread -Wno-error=format-truncation \
 	-std=gnu99 -g3 -MD -I./target \
 	-I. -Iinclude -I$(PLATFORM_DIR)
 
```

#820 contains the same error message, but I can't figure out if it's actually related.

UweBonnes, 2021-09-06:
What version is your H743, old or castrated H745? Can you make the elf and bin file available or even the project? Do you expect problems when flashed to a Nucleo-H743 (old chip revision)?

UweBonnes, 2021-09-06:
Several issues: 
- default gdb timeout to wait for an answer is 2 seconds. error happens after some timeouts.
- Reading in hosted is at best half speed of reading in firmware, but only with large chunks
- crc32 only use 128 byte chunks.
- reading 2 MByte of the H743xI takes about 14 second in firmware, 28 s with hosted@1024 byte chunks and  80 sec@128 byte chunks as in crc32
- It seems, there is no way to send some "wait" note for the qCRC packet.
- Memory faults are not propagated by BMP for CRC

So at least you need to increment the timeout substantial  with "set remotetimeout xx". Using the serial debugger will also speed things up.

UweBonnes, 2021-09-06:
In https://github.com/UweBonnes/blackmagic/tree/fixes now keep-alive data (0)  is sent around every second, so no more need  to increase remotetimeout. Can you try and report back?

arbrauns, 2021-09-07:
I built 57d0f48cefcb4b5b1b9ba6ed0bdcd0f10600109a and it works great both in firmware and hosted (without `set remotetimeout`)! Thanks a lot for the quick investigation and fix!

closed 2021-09-07 by arbrauns

arbrauns, 2021-09-13:
For future reference: fixed by #922.