blackmagic-debug/blackmagic #921: GD32E230 Option byte write needs to be 32bit sized

walmis, 2021-09-06:
I've stumbled across an issue that prevented me from resetting the flash protection option bytes. I've traced the issue, and it appears that target_mem_write16 does not work on GD32E230 to write option bytes.
Quick fix below, but not sure how it would affect other targets.

```
static bool stm32f1_option_write_erased(target *t, uint32_t addr, uint16_t value)
{
	if (value == 0xffff)
		return true;
	/* Erase option bytes instruction */
	target_mem_write32(t, FLASH_CR, FLASH_CR_OPTPG | FLASH_CR_OPTWRE);
	target_mem_write32(t, addr, 0xFFFF0000 | value); // <-- change here
	/* Read FLASH_SR to poll for BSY bit */
	while (target_mem_read32(t, FLASH_SR) & FLASH_SR_BSY)
		if(target_check_error(t))
			return false;

	return true;
}
```

UweBonnes, 2021-09-06:
PM0075 for the F103 in section 2.3.5 Option byte programming clearly tells: "Then the user has to set the OPTPG bit in the FLASH_CR register and perform a half-word write operation at the desired Flash address"
Same in RM0316 for the F3. So you need to seperate the cases!

walmis, 2021-09-06:
Yeah, whats strange from GD32E230 manual:
GD32E23x_User_Manual_EN_v1.2-1.pdf
It would appear that 16bit write should work here.

>2.3.9.  Option byte programming
The FMC provides a 32-bit word/32-bit double word programming function which is used for
modifying the option byte block contents. The following steps show the programming
operation sequence.
 Unlock the FMC_CTL register if necessary.
 Unlock the OBWEN bit in FMC_CTL register if necessary.
 Check the BUSY bit in FMC_STAT register to confirm that no flash memory operation is
in progress (BUSY equal to 0). Otherwise, wait until the operation has been finished.
 Write the program command into the OBPG bit in FMC_CTL register.
 A 32-bit word/16-bit half word write at desired address by DBUS.

esden, 2021-12-16:
I think the best way here would be to use 16 bit (half word) writes for backwards compatibility with the ST parts. If that is not possible then we will have to find a way to handle the GD parts separately from ST parts which can be tricky as they are meant to be drop in replacements.

What are your thoughts @walmis?

perigoso, 2022-06-06:
what about a read back verify? I'm not familiar with the targets, but assuming we can read that memory, and that "target_mem_write16 does not work" means the value in memory didn't actually change, we could readback and on fail attempt to do a 32 bit wide write, this of course if we really can't distinguish what target device we're attached to some other way

kind of necro bumping here, was just looking through the bug label

perigoso, 2022-06-27:
some clarification, does this affect all GD parts, or only the GD32E230 family?

PR #1081 (2022-06-27): STM32F1 general cleanup and GD32E230 option write special case
trying to do something about #921 
I am not able to physically test this as I don't have any GD32E230 devices

@walmis do you think you could do some testing for me?

fixes #921
closed 2022-06-27 by dragonmux