blackmagic-debug/blackmagic #925: Fails to scan/halt a Cortex M0+ (MINDP) with STLink V2

fabalthazar, 2021-09-15:
```shell
$ ./blackmagic -t
INFO: Open USB 80ee:0021 class  0 failed
BMP hosted v1.7.1-256-gdd28fa5
 for ST-Link V2/3, CMSIS_DAP, JLINK and LIBFTDI/MPSSE
Using 0483:3748 S?ixeITF8g STMicroelectronics
 STM32 STLink
STLink firmware version: V2J37S7
Leaving DEBUG Mode
Running in Test Mode
Target voltage: 2.93V Volt
Divisor for 4.0000 MHz is 0
Speed set to  3.6000 MHz for SWD
Leaving MASS Mode
Read Core ID: 0x0bc11477
DPIDR 0x0bc11477 (v1 MINDP rev0)
RESET_SEQ failed
AP   0: IDR=04770031 CFG=00000000 BASE=f0000003 CSW=03000040 (AHB-AP var3 rev0
read_retry failed. STLINK_SWD_AP_FAULT
stlink_readmem from  e000edf0 to 806e9914, len 4failed
Halt via DHCSR: Failure DHCSR ffffffff after  2001ms
...
```

Target : STM32G070
Probe : STLink V2

Break introduced in dd28fa5.

fabalthazar, 2021-11-14:
Can it be closed?

UweBonnes, 2021-11-14:
If the problems is solved for you, please close it.

closed 2021-11-14 by fabalthazar