blackmagic-debug/blackmagic #929: GDB 11 compatibility

arbrauns, 2021-09-17:
With GDB 11.1, attaching to an STM32H7 target no longer works:

```
$ arm-none-eabi-gdb -batch -nx -ex 'target extended-remote :2000' -ex 'monitor swdp_scan' -ex 'attach 1'
Target voltage: 1.8V
Available Targets:
No. Att Driver
 1      STM32H7 M7
../../gdb/remote.c:7979: internal-error: ptid_t remote_target::select_thread_for_ambiguous_stop_reply(const target_waitstatus*): Assertion `first_resumed_thread != nullptr' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Quit this debugging session? (y or n) [answered Y; input not from terminal]

This is a bug, please report it.  For instructions, see:
<https://www.gnu.org/software/gdb/bugs/>.

../../gdb/remote.c:7979: internal-error: ptid_t remote_target::select_thread_for_ambiguous_stop_reply(const target_waitstatus*): Assertion `first_resumed_thread != nullptr' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Create a core file of GDB? (y or n) [answered Y; input not from terminal]
fish: Job 1, 'arm-none-eabi-gdb -batch -nx -e…' terminated by signal SIGABRT (Abort)
```

<details>
<summary>hosted log</summary>

```
$ ./src/blackmagic -v31
BMP hosted (BMP Only) v1.7.1-263-ga9854e9-dirty
Using:
 Black Magic Probe (Unknown) v1.7.1-263-ga9854e9-dirt@ D6C88BE7
+#!GA#
       K v1.7.1-263-ga9854e9-dirty
Remote is  v1.7.1-263-ga9854e9-dirty
Listening on TCP: 2000
Got connection
gdb_getpacket : qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+;memory-tagging+
gdb_putpacket : PacketSize=400;qXfer:memory-map:read+;qXfer:features:read+
gdb_getpacket : vMustReplyEmpty
*** Unsupported packet: vMustReplyEmpty
gdb_putpacket :
gdb_getpacket : !
gdb_putpacket : OK
gdb_getpacket : Hg0
*** Unsupported packet: Hg0
gdb_putpacket :
gdb_getpacket : qXfer:features:read:target.xml:0,3fb
gdb_putpacket : E01
gdb_getpacket : qTStatus
*** Unsupported packet: qTStatus
gdb_putpacket :
gdb_getpacket : ?
gdb_putpacket : W00
gdb_getpacket : qRcmd,737764705f7363616e
 !GV#
       K1.8V
 !GV#
       K1.8V
gdb_putpacket : O54617267657420766f6c746167653a20312e38560a
!GF003d0900#
       K0
!Gf#
       K17f03100
!Gf#
       K17f03100
Speed set to  3.2727 MHz for SWD
remote_swdptap_init
!SS#
       K0
swdptap_seq_out        32 ticks: ffffffff
[... 3925 lines cut ...]
       Kf0aac60b
ap_memread @ 52002020 len 4: f0 aa c6 0b
gdb_putpacket : T05
gdb_getpacket : qC
*** Unsupported packet: qC
gdb_putpacket :
gdb_getpacket : Hg0
*** Unsupported packet: Hg0
gdb_putpacket :
gdb_getpacket : qXfer:features:read:target.xml:0,3fb
gdb_putpacket : m<?xml version="1.0"?><!DOCTYPE target SYSTEM "gdb-target.dtd"><target>  <architecture>arm</architecture>  <feature name="org.gnu.gdb.arm.m-profile">    <reg name="r0" bitsize="32"/>    <reg name="r1" bitsize="32"/>    <reg name="r2" bitsize="32"/>    <reg name="r3" bitsize="32"/>    <reg name="r4" bitsize="32"/>    <reg name="r5" bitsize="32"/>    <reg name="r6" bitsize="32"/>    <reg name="r7" bitsize="32"/>    <reg name="r8" bitsize="32"/>    <reg name="r9" bitsize="32"/>    <reg name="r10" bitsize="32"/>    <reg name="r11" bitsize="32"/>    <reg name="r12" bitsize="32"/>    <reg name="sp" bitsize="32" type="data_ptr"/>    <reg name="lr" bitsize="32" type="code_ptr"/>    <reg name="pc" bitsize="32" type="code_ptr"/>    <reg name="xpsr" bitsize="32"/>    <reg name="msp" bitsize="32" save-restore="no" type="data_ptr"/>    <reg name="psp" bitsize="32" save-restore="no" type="data_ptr"/>    <reg name="primask" bitsize="8" save-restore="no"/>    <reg name="basepri" bitsize="8" save-restore="no"/>    <reg na
gdb_getpacket : qXfer:features:read:target.xml:3fb,3fb
gdb_putpacket : mme="faultmask" bitsize="8" save-restore="no"/>    <reg name="control" bitsize="8" save-restore="no"/>  </feature>  <feature name="org.gnu.gdb.arm.vfp">    <reg name="fpscr" bitsize="32"/>    <reg name="d0" bitsize="64" type="float"/>    <reg name="d1" bitsize="64" type="float"/>    <reg name="d2" bitsize="64" type="float"/>    <reg name="d3" bitsize="64" type="float"/>    <reg name="d4" bitsize="64" type="float"/>    <reg name="d5" bitsize="64" type="float"/>    <reg name="d6" bitsize="64" type="float"/>    <reg name="d7" bitsize="64" type="float"/>    <reg name="d8" bitsize="64" type="float"/>    <reg name="d9" bitsize="64" type="float"/>    <reg name="d10" bitsize="64" type="float"/>    <reg name="d11" bitsize="64" type="float"/>    <reg name="d12" bitsize="64" type="float"/>    <reg name="d13" bitsize="64" type="float"/>    <reg name="d14" bitsize="64" type="float"/>    <reg name="d15" bitsize="64" type="float"/>  </feature></target>
gdb_getpacket : qXfer:features:read:target.xml:7b1,3fb
gdb_putpacket : l
Dropped broken connection: Resource temporarily unavailable
```
</details>

I bisected this down to gdb commit [`8f66807`](https://github.com/bminor/binutils-gdb/commit/8f66807b98f7634c43149ea62e454ea8f877691d) ("gdb: better handling of 'S' packets").

Environment:
- hosted: a9854e9b371587bab4a1ce242dcf626ba4e4885c
- firmware (on Mini v2.1): a9854e9b371587bab4a1ce242dcf626ba4e4885c
- GDB 11.1

UweBonnes, 2021-09-17:
Do you think this is a problem of BMP? Did you report to GDB too?

arbrauns, 2021-09-17:
I don't know what and where exactly the problem is, but I can say that a plain ST-Link with [`st-util`](https://github.com/stlink-org/stlink) works fine. Obviously gdb changed something so BMP no longer works, but maybe BMP has always been doing something funky that has only now become apparent.

I haven't reported to GDB yet, will probably do that next week.

UweBonnes, 2021-09-17:
GDB does not report some invalid package but core dumps. Even when BMP sends something wrong, gdb should not core dump but report the problem and handle it gratefull. There is at least a problem with gdb, So report to gdb!.

tuna-f1sh, 2021-09-30:
@arbrauns did you report this? I had a quick look but couldn't find anything. I've encountered this too, attaching to a STM32F4. Wanted to check before I do any more digging.

For others on arch, one can roll back for now with `sudo pacman -U /var/cache/pacman/pkg/arm-none-eabi-gdb-10.2-2-x86_64.pkg.tar.zst` or whatever the last package version was installed.

arbrauns, 2021-09-30:
No, I haven't found time yet. I also don't know how to make GDB spit out a trace of the remote protocol, which would probably be necessary to debug this further.

UweBonnes, 2021-09-30:
set debug remote 1
Have a look at the wiki

tuna-f1sh, 2021-10-01:
I've created a bug report for gdb: https://sourceware.org/bugzilla/show_bug.cgi?id=28405

ofauchon, 2021-11-02:
Hi. 
Same problem here:

- Archlinux (arm-none-eabi-gcc-11.2.0-1)
- blackmagic swlink (master branch checkout Sat Oct 30 21:01:06 2021 +1000)
- programming target : Bluepill STM32F103

Please find the log attached 
```
arm-none-eabi-gdb -ex 'target extended-remote /dev/ttyACM1' -ex 'set debug remote 1' -ex 'monitor swdp_scan' -ex 'attach 1' -ex load test.elf  2>&1 | tee  log_core_gdb11.log
```

[log_core_gdb11.log.gz](https://github.com/blacksphere/blackmagic/files/7464121/log_core_gdb11.log.gz)

thanks @tuna-f1sh  for the archlinux fix

```
wget https://archive.archlinux.org/packages/a/arm-none-eabi-gdb/arm-none-eabi-gdb-10.2-2-x86_64.pkg.tar.zst
pacman -U arm-none-eabi-gdb-10.2-2-x86_64.pkg.tar.zst 
```

UweBonnes, 2021-11-02:
Is there a way to deflate  arm-none-eabi-gcc-11.2.0-1-x86_64.pkg.tar.zst?

tuna-f1sh, 2021-11-03:
Yes, the patch works so if one is super keen to use GDB11 one can compile with the patch in that bug report. As you say, was waiting until 11 is out to close this, just to ensure it captures others having the issue.

esden, 2021-12-16:
According to the [official website](https://www.sourceware.org/gdb/) GDB V11.1 was released in September 2021. Are we waiting for V11.2? Just want to make sure when we can close this issue. :)

arbrauns, 2021-12-16:
> Are we waiting for V11.2?

That's correct, here's the latest mail for the patch in question, discussing if it should be included in 11.2: https://sourceware.org/pipermail/gdb-patches/2021-December/184171.html

esden, 2021-12-16:
Thank you @arbrauns for clarification. I hope we will get that release soon and the patch is included. 🤞 :)

tuna-f1sh, 2021-12-23:
Merged to master and 11-branch so should be in 11.2 release: https://sourceware.org/git/?p=binutils-gdb.git;a=commit;h=b622494ee378fd0a490c934c509364b4c7735273

DX-MON, 2021-12-26:
Most excellent! We'll close this issue when 11.2 is released and we confirm that the patch is included and didn't get reverted, btu this is good news.

Windfisch, 2022-01-25:
Sorry to dampen your enthusiasm :/.

I just tried a fresh build of gdb 11.2 on ArchLinux (built [the pkgbuild](https://github.com/archlinux/svntogit-community/tree/packages/arm-none-eabi-gdb/trunk) by hand) and it didn't solve the issue for me. 

However, the error message changed to

```
GNU gdb (GDB) 11.2
...
(gdb) attach 1
Attaching to program: /home/flo/synopsys/usbhost/target/thumbv7em-none-eabihf/release/blinky, Remote target
../../gdb/thread.c:72: internal-error: thread_info* inferior_thread(): Assertion `current_thread_ != nullptr' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
```

This still seems related to the issue, just failing at a different location now.

chmeeedalf, 2022-02-02:
I'm seeing the same thing, using gdb 11.1 on void linux, trying to debug an Arduino Due with a "Jeff Probe".  I applied the patch referenced above, and I see the same assertion as @Windfisch now.

tuna-f1sh, 2022-02-02:
Weird, I did check the patch in the ticket thread I created fixed the issue I was having and worked with a target - now I'm wondering how this slipped through. Maybe something else in the merge? Haven't had time to check myself yet but otherwise, I guess we'll need to make another ticket.

tuna-f1sh, 2022-02-09:
I had chance to check that the patched 11.1 does not have this issue (to confirm I wasn't going mad) and also capture the issue myself to create a new gdb bug for this problem: https://sourceware.org/bugzilla/show_bug.cgi?id=28874

DX-MON, 2022-02-09:
Thank you all for all the extra info and testing - we'll keep the issue open till GDB sorts things out or we can find a workable way around it.

tuna-f1sh, 2022-03-11:
Just to update this thread, I found what seems to be the issue that the 11.2 merge caused over the 11.1 patched. I've put a patch in that bugzilla thread (https://sourceware.org/bugzilla/show_bug.cgi?id=28874) and awaiting feedback. Hopefully this fix can be merged into 11.3...

I've got a Github fork with the patched 11.2 here too: https://github.com/tuna-f1sh/binutils-gdb/tree/gdb-11.2-patch

DX-MON, 2022-03-11:
Let's hope they merge it soon and don't break it again this time

forkbomb9, 2022-05-14:
This is still an issue in GDB 12.1...

```
arm-none-eabi-gdb -nx --batch -ex 'target extended-remote /dev/ttyACM0' -ex 'monitor swdp_scan' -ex 'attach 1'
Available Targets:
No. Att Driver
 1      STM32F1 medium density M3
../../gdb/thread.c:85: internal-error: inferior_thread: Assertion `current_thread_ != nullptr' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
----- Backtrace -----
0x55b4cfaab1cb ???
0x55b4cfda9644 ???
0x55b4cfe80c43 ???
0x55b4cfd6c531 ???
0x55b4cfc5af45 ???
0x55b4cfc161ba ???
0x55b4cfd6ff14 ???
0x55b4cfcd3713 ???
0x55b4cfcdb677 ???
0x55b4cfbfafcb ???
0x55b4cfae1654 ???
0x55b4cfd77152 ???
0x55b4cfc376a5 ???
0x55b4cfc377d1 ???
0x55b4cfee3c4a ???
0x55b4cfa16faf ???
0x7f59c93e228f ???
0x7f59c93e2349 ???
0x55b4cfa1d414 ???
0xffffffffffffffff ???
---------------------
```

dragonmux, 2022-05-14:
How is it they keep breaking the same damn thing despite multiple bug reports 😔 ?! Ok, thank you for reporting in with that - one of the team will chase it up on their bug tracker and try.. again.. to get it fixed

tuna-f1sh, 2022-05-15:
The commenters in my bug report (https://sourceware.org/bugzilla/show_bug.cgi?id=28874) agreed that it fixed it but it is hacky - essentially it reverts a function call back to how it worked in 11.1 for the (single threaded) embedded target use case. Apparently the whole thing came about to support multi-threaded embedded targets so I can see it’s not ideal from this standpoint. 

It went quiet from a merging point so I’m not surprised 12.1 still has the bug. The guy Andrew who merged the original bug and fix hasn’t chimed in. I think I’ll have to bring it up to date as a patch for 12.1 and then share it on the mailing list.

esden, 2022-06-01:
There is a patch that fixes this issue that still needs to be merged by GDB developers. Not sure why it is still pending and not merged. https://sourceware.org/pipermail/gdb-patches/2022-April/188058.html

bonidydy, 2022-07-13:
Is there any debugger that the probe works with?

dragonmux, 2022-07-13:
Hi @bonidydy, yes - you can either grab the patch from their mailing list and use GDB 11 or 12 with it applied, or you can use GDB 10 (https://developer.arm.com/downloads/-/gnu-rm for a download)

tuna-f1sh, 2022-07-13:
I'm using a patched mirror here for the lazy: https://github.com/tuna-f1sh/binutils-gdb/tree/gdb-12.1-bmppatch

Also on Arch, this is the (last working) 10.1 PKGBUILD: https://github.com/archlinux/svntogit-community/blob/22d7668d2bfa4809296bd36cfb8bfc5e239f6611/repos/community-x86_64/PKGBUILD

Might have to use `makepkg --install --skippgpcheck` because the keys are out of date now. Not sure how we can get that April patch merged? I'll try commenting on original report again.

tuna-f1sh, 2022-07-13:
That is the `arm-none-eabi` target build...it's just the old version of the main package: https://archlinux.org/packages/community/x86_64/arm-none-eabi-gdb/

dragonmux, 2022-07-20:
For those interested in testing and validating - please see the linked PR #1125 which contains our answer to the GDB folks blowing us off again. This should allow BMP to work out the box with GDB 11 and probably also 12.

esden, 2022-07-21:
Just to clarify. The #1125 is only a work around. The underlying bug in GDB is not fixed, and "Fake Threads" might interfere with "sidecar" scripting that adds support for RTOS threads. I am opening this issue again, to track progress of GDB project on this topic.

There was [finally a reaction](https://sourceware.org/pipermail/gdb-patches/2022-July/190869.html) to an attempt to submit a patch to fix the GDB bug. We can only hope that they will eventually implement a proper fix.

Qyriad, 2022-07-25:
@tuna-f1sh your bug report to GDB says
> I have re-confirmed that the patch in the original problem ticket works on the 11.1 branch (to make sure I wasn't going mad!) so something in the 11.2 merge has caused this?

Do you happen to have the commit hash you tested this on? I can't seem to find any commit where applying the fix for the original [`first_resumed_thread != nullptr`](https://sourceware.org/bugzilla/show_bug.cgi?id=28405) bug doesn't then still have the current [`current_thread_ != nullptr`](https://sourceware.org/bugzilla/show_bug.cgi?id=28874) bug.

tuna-f1sh, 2022-07-25:
@Qyriad it would have been the gdb-11.1-release tag here: https://sourceware.org/git/?p=binutils-gdb.git;a=commit;h=c599303f92b1edd6ead3947737a8ae9e1c85e08c , the patch is from here https://sourceware.org/bugzilla/show_bug.cgi?id=28405

I also had a 11.2-release patched here: https://github.com/tuna-f1sh/binutils-gdb/commit/5e714ea6775e5c77886921a3e8dbf61559f8d2b4

martinling, 2023-06-05:
FWIW, this is still broken with GDB 13.1, when using a BMP running pre-#1125 firmware (`v1.7.1-332-g8def28d`).

esden, 2023-06-05:
And it will likely stay broken forever as the GDB folks can't understand the issue and don't want to fix it. :( @martinling You will have to update your firmware sorry. :(

dragonmux, 2023-12-12:
Ok, we can't even begin to support that version and working back through commit parents from the commit ID referenced by the binary in that repo shows that to be based on v1.8.0 - please use *this* repo's code and build with `make PROBE_HOST=stlink BLUEPILL=1`. Further instructions are in the stlink platform's README.md and on the project website.

The version of the firmware you're trying to use, which has been custom patched, is old and predates the fix PR for this issue.

dragonmux, 2023-12-12:
Looks like that is enumerating, but possibly that's Windows 7 being silly about drivers - what does the Properties panel in Device Manager actually say, as that doesn't give us enough information to work from?

dragonmux, 2023-12-12:
Going at this on a slightly different tact given the info from those screenshots, is there a strong reason you're using W7 rather than any newer more supported OS? It's very difficult to know what could be going on when it's the hub or controller you're plugging the Blue Pill into that's getting mad, given Windows doesn't really give useful diagnostics short of running protocol analysers.

Given that other firmware is so old and predates the fixes for this issue, please open a new issue ticket for your current problems so we can keep this issue specifically to the issues with GDB, which is a separate problem to getting W7 to play nice with the current firmware and any potential issues with the guides you followed to make your Blue Pill a BMP.

PR #1125 (2022-07-21): Feature: fake thread support
This PR addresses the issue that the GDB folks have blown us off once again and refuse to fix GDB to support targets (and therefore GDB servers) which simply can't have threads.

In this PR we implement support for the machinery GDB uses for threads support, but only to fake there being a single thread in any given "process" (debug port on the scan chain). This has the side effect of requiring the implementation of such things as vKill otherwise GDB 11 and newer won't actually execute a process kill as they don't use the old method.

This will need merging to both  [... cut]