blackmagic-debug/blackmagic #945: Set SWCLK and SWDIO to Hi-Z on detatch

emeb, 2021-11-08:
There are many cases where the SWD interface pins are re-purposed in the target application - eg those pins are remapped for other GPIO / peripheral interfaces when the embedded firmware is running. In these cases the SWD interface is only available immediately after power-up while the NRST is asserted. After programming and subsequent reset there will be contention on the SWD pins if the BMP continues to drive these signals, so it's desirable for the BMP to set these pins to Hi-Z after completing the programming cycle.

Current native BMP hardware uses external buffers between the STM32F103 MCU running BMP firmware and the target and will not be able to completely disable the SWD lines, but non-native hardware (and much older versions of BMP) will still benefit if these pins are disabled on the STM32F103.

Note that simply disabling the SWD buffers by means of TPWR is not sufficient since it's still necessary to power the target.

UweBonnes, 2021-11-09:
STLINK V21 and V3 can switch power and due to no voltage translation buffers can tristate SWCLK and SWDIO. Both have a BMP firmware variant and so can be used for experiments and setting up the needed firmware changes

esden, 2021-12-16:
Thanks @emeb that is a good suggestion. It is bit tricky to do, but I will see if I can find an extra gpio on the mcu to allow control. No promisses but it is a good idea.

esden, 2022-03-28:
Hey @emeb,

So I think I managed to squeeze in the ability to control the SWCLK/TCK level shifter direction. (that also controls TDI, but I don't think we care about that) This way we should be able to set SWDIO and SWCLK to input, setting them to Hi-Z when idle. See highlighted parts of the attached schematic.

Let me know what you think.

[image]

emeb, 2022-03-28:
Excellent news. Looking forward to trying it out!

PR #1192 (2022-08-16): Feature: high-z TCK/SWCLK when idle

## Detailed description

Certain low pin-count parts very heavily overload the device pins with functionality. This can mean it's inappropriate to drive TCK/SWCLK before initiating a scan, after scan but before attach, and after detach.

This PR introduces tristating support for these devices when running native hardware revision 6 and newer, allowing BMP to work around this problem. This also means BMP will play nicer with other devices on the roadmap.

This has been tested working against both the SWD and JTAG modes of the Tiva-C TM4C123GH6PM.

## Your ch [... cut]
closed 2022-08-16 by esden