blackmagic-debug/blackmagic #946: Hardware enhancement: separate regulator for TPWR

compuphase, 2021-11-09:
BMP version 2.1 has a single 3.3V voltage regulator. My suggestion is (on a next version of BMP) to add a secondary regulator for TPWR only. The motivation is to tackle two scenarios.

Scenario 1: if you set TPWR and the target draws too much current, the voltage drops to below 3.3V, and this could affect the operation of the BMP itself.
Scenario 2: you have set TPWR, but the target also powers the TPWR line, and (oops) the target sets it to 5V.

When TPWR is on, the regulator output is connected to the Vref pin of the SWD connector (via Q1). But the regulator output is also the source of the reference voltage of the ADC. So when TWPR is enabled, VCC = Vref, and (here comes the twist) it is measured as 3.3V regardless of the real voltage on Vref.

That is: right now, BMP firmware cannot detect either of these two scenarios. With a second voltage regulator, the BMP circuit is better protected against these scenarios, but in addition, it can still measure the voltage on Vref if TPWR is enabled.

There are alternatives to consider:
- a polyfuse in the TPWR line would protect against drawing too much current (fix for scenario 1)
- an ideal diode in the TPWR line would block a higher voltage on Vref to affect VCC (fix for scenario 2)
- a precision zener would be a stable voltage reference for the ADC (regardless of fluctuations in VCC)

esden, 2021-12-16:
Thank you @compuphase. This is a good suggestion. I will see what I can do. To be honest if we are going down that path we should also use an adjustable voltage regulator to provide settable voltage to the target.

One issue that I still have with that is the question of what happens when the BMP side vreg is disabled and the target is sending power into the VREF pin. We should make sure that this is a high impedance input. Especially important for low power boards that might have an issue with us drawing even tiny current from the vref pin.

Ideas and thoughts are welcome. :)

compuphase, 2021-12-16:
You should either use a regulator that blocks reverse current flow (like the TPS73633), or put an ideal diode on the output of the regulator. Current from a target (when the target powers Vref) must not flow back into the BMP.

A separate voltage regulator for TPWR to protect the BMP is on top of my wish list. Having a configurable output voltage is on my list too, but near the bottom of it. My next topmost item would be: overvoltage protection on all pins of the debug and UART connectors.

UweBonnes, 2021-12-16:
These are nice and well substantiated wishes, but current part shortage e.g. expects delivery of the TPS73633   end of december 2022 from e.g.  Farnell.

DX-MON, 2021-12-16:
> These are nice and well substantiated wishes, but current part shortage e.g. expects delivery of the TPS73633 end of december 2022 from e.g. Farnell.

The quoted part number is only a suggestion to @esden of a part that fits the requirements as reference - it is not the only part or only way to implement this and it is agreed that this is something worth implementing for low power devices that may not have much power budget built into them.

esden, 2021-12-16:
> You should either use a regulator that blocks reverse current flow (like the TPS73633), or put an ideal diode on the output of the regulator. Current from a target (when the target powers Vref) must [...]

That is a good suggestion. I would vote for a "non special" part solution to have a better chance navigating the chip shortage. If we also measure and control the voltage on the target side of the diode, this can be a pretty good solution. This way we can compensate for the drop.

There might also be a more complicated solution using a mosfet as a reverse current protection. I will have to put the different solutions side by side and see what advantages and disadvantages they have taking the chip shortage issues into account. :)

I don't think I will be able to add it immediately to the next production version, but I will start prototyping some circuits to see what we can do about this.

zyp, 2021-12-16:
The trick is to put the ideal diode upstream of the LDO. That way the drop won't affect the regulated voltage.

On the [Orbtrace](https://bin.jvnv.net/file/xPod5/orbtrace_mini.pdf) (ref. page 10), I'm using a SLG46582 as a programmable LDO and a TLA2024 to monitor voltage and current draw. They're both sitting after a MAX40200 meaning they'll be powered by backfed current, but not the rest of the device.